Freescale/meta-freescale #672: imx-gpu-viv 6.4.3.p0.0 graphical issues with Qt

sapiippo, 2021-03-02:
The 6.4.3.p0.0 version of imx-gpu-viv is having problems running Qt correctly with eglfs backend (https://bugreports.qt.io/browse/QTBUG-90486). The issue are solved with the next update (6.4.3.p1.0) that has landed in master.
Would it be possible to have that and related changes be cherry-picked to gatesgarth as well?

otavio, 2021-03-02:
Yes, we can. Could you cook a backport PR for it?

sapiippo, 2021-03-02:
I can attempt one.
I'm not entirely sure if gpu update alone is enough or does it require all of release 5.4.70_2.3.0 commits.

otavio, 2021-03-02:
As people is using gatesgarth it might be good to upgrade the GPU initially.

otavio, 2021-03-03:
Fixed in https://github.com/Freescale/meta-freescale/pull/674

closed 2021-03-03 by otavio