Freescale/meta-freescale #686: Boot up failure with imx8mn-ddr4-evk with 5.4.70_2.3.0

sapiippo, 2021-03-16:
Images built from master branch fails to boot up imx8mn-ddr4-evk board.

The boot stops after:
```
U-Boot SPL 2020.04-imx_v2020.04_5.4.70_2.3.0+ge42dee801e (Dec 02 2020 - 06:37:20 +0000)
power_bd71837_init
DDRINFO: start DRAM init
DDRINFO: DRAM rate 2400MTS
DDRINFO:ddrphy calibration done
DDRINFO: ddrmix config done
Normal Boot
Trying to boot from BOOTROM
image offset 0x8000, pagesize 0x200, ivt offset 0x0
NOTICE:  BL31: v2.2(release):rel_imx_5.4.70_2.3.0-0-gf1d7187f2
NOTICE:  BL31: Built : 09:58:42, Dec  1 2020
```
Reverting commits 15cb077d7a715e92889fc8ea5dcc5a89fbec79ee "imx-mkimage: : update to 5.4.70_2.3.0" and d46f7a84f9bde371bd1ff17ed30e8542a870509b "imx-atf: upgrade to NXP release 5.4.70_2.3.0" makes it boot correctly.

zandrey, 2021-03-16:
@sapiippo 

Can you provide more details about your build setup? How do you populate your Yocto setup (repo or manual)? Which distro do you build? Which image?

Can you also post the _Build Configuration_ here, so it is clear which commits your layers are sync'ed up to?

-- andrey

sapiippo, 2021-03-16:
I have a manual setup, using latest master branch for everything and building core-image-minimal with fsl-wayland distro.

`Build Configuration:
BB_VERSION           = "1.49.2"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-fsl-linux"
MACHINE              = "imx8mn-ddr4-evk"
DISTRO               = "fsl-wayland"
DISTRO_VERSION       = "3.3-snapshot-20210316"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa53 crypto"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "master:eec64066073428e38d5b0b6947ff302943188f65"
meta-python          
meta-oe              = "master:589aa162cead42acdd7e8dbd7c0243b95e341f19"
meta-freescale       = "master:cc4b91bf3a3cf5c2034372c607783048aae44b65"
meta-freescale-distro = "master:1deb1a9032ea8ef6cf67e7c83e4e9ec3f25d1451"
`

zandrey, 2021-03-16:
I've tried the same setup, and indeed - it is not booting...

I do recall I've seen similar boot hang though...

Can you try to drop `optee` from `MACHINE_FEATURES` in [conf/machine/include/imx8mn-evk.inc](https://github.com/Freescale/meta-freescale/blob/cc4b91bf3a3cf5c2034372c607783048aae44b65/conf/machine/include/imx8mn-evk.inc#L6)?

sapiippo, 2021-03-16:
removing `optee` makes it boot correctly.

zandrey, 2021-03-16:
> 
> 
> removing `optee` makes it boot correctly.

This is what I was suspecting...

@thochstein Is it possible you take a look at this issue? It seems that the _OP-TEE_ when combined with _ATF_ in 2.3.0 release is not bootable (at least for Mini EVK).

I've had similar issues when I developed the mainline BSP support, and that is the reason why I included commit 47ae500a23984b093b3793d783553569140ff000 into my branch. I was thinking that it only affects the combination of U-Boot and OP-TEE, but it looks like there is something missing in NXP BSP as well.

If you can point out what might be wrong - that would be really helpful, then I can also drop the `optee` feature limitation commit.

-- andrey

thochstein, 2021-03-16:
OP-TEE is upgraded to 3.10 in our 5.4.70_2.3.0 release. I'll work on upgrading `meta-freescale` this morning.

zandrey, 2021-03-16:
> 
> 
> OP-TEE is upgraded to 3.10 in our 5.4.70_2.3.0 release. I'll work on upgrading `meta-freescale` this morning.

Thanks for noting that, good catch! Indeed, the _OP-TEE_ has been overlooked during the update... I'll be waiting for the update then and test it with mainline BSP afterwards, so that I can drop one commit which might not be required after all.

-- andrey

closed 2021-03-16 by otavio

zandrey, 2021-03-17:
@thochstein Thanks a lot for the upgrade in #690!

I've tried the NXP BSP for `imx8mn-ddr4-evk` and `imx8mm-lpddr4-evk` - they both boot OK after the upgrade.

@otavio 
I've also tried to enable back the `optee` feature for mainline BSP, but it is still not booting on both EVKs from above. I guess the OP-TEE support would require more digging (not only in `optee-os`, but in U-Boot as well), so I suggest that the #680 would still contain the commit 47ae500a23984b093b3793d783553569140ff000 which disables OP-TEE until I can find a root cause.

Would that be OK for you to merge #680 without OP-TEE for mainline for now?

-- andrey

otavio, 2021-03-17:
Sure @zandrey. It makes sense.

PR #690 (2021-03-16): optee-imx: Upgrade to 3.10.0
Update all recipes to pull sources from imx_5.4.70_2.3.0 branch in NXP
upstream and update SRCREV to match release tags.

Update PREFERRED_VERSION of optee for mx8 to point to 3.10.0.imx.

Following notes are additions to the version update of recipes:
--------------------------
optee-client:
- Drop CFLAGS patch as it is already applied upstream

optee-os:
- Drop gcc10 patch as it is already applied upstream
- Add dependency on python3-pycryptodomex-native
- Drop NOWERROR flag as package is reworked to disable warnings by default
- Rework mkimage load address since [... cut]
commit be6a39f2 (2021-03-16): optee-imx: Upgrade to 3.10.0

Update all recipes to pull sources from imx_5.4.70_2.3.0 branch in NXP
upstream and update SRCREV to match release tags.

Update PREFERRED_VERSION of optee for mx8 to point to 3.10.0.imx.

Following notes are additions to the version update of recipes:
--------------------------
optee-client:
- Drop CFLAGS patch as it is already applied upstream

optee-os:
- Drop gcc10 patch as it is already applied upstream
- Add dependency on python3-pycryptodomex-native
- Drop NOWERROR flag as package is reworked to disable warnings by default
- Rework mkimage load address sinc [... cut]