Freescale/meta-freescale #689: Update optee-os version

sfalbesoner, 2021-03-16:
Currently this layer still uses optee-os version 3.7.0, which was released in October 2019 (according to https://github.com/OP-TEE/optee_os/releases). That was almost one and a half years ago and another five releases have been published since, the newest one being 3.12.0. Looking at the changelog, especially versions 3.9.0 and 3.10.0 contained lots of improvements, at least in terms of number of merged pull requests (181 in 3.9.0, 122 in 3.10.0).

Is there any concrete reason the freescale community layer still uses 3.7.0?
At least version 3.10.0 is availabe at the source repository https://source.codeaurora.org/external/imx/imx-optee-os/, in the branches imx_5.4.47_2.2.0 and imx_5.4.70_2.3.0. 

I'd be available to help submitting a PR, though I can only offer testing for the i.MX7 (which in general seems to be far less in use than i.MX6 and the i.MX8).

Best regards,
Sebastian

otavio, 2021-03-16:
Sure, please prepare a PR and mark this issue as Fixed by it. :-)

sfalbesoner, 2021-03-16:
Just a few minutes later after opening this issue I found that someone is obviously already working on this: ;)
"_OP-TEE is upgraded to 3.10 in our 5.4.70_2.3.0 release. I'll work on upgrading meta-freescale this morning._" (https://github.com/Freescale/meta-freescale/issues/686#issuecomment-800219904)

otavio, 2021-03-16:
@thochstein are you going to look at this or @sfalbesoner should take a look at it?

thochstein, 2021-03-16:
I'm working on it :-)

PR #690 (2021-03-16): optee-imx: Upgrade to 3.10.0
Update all recipes to pull sources from imx_5.4.70_2.3.0 branch in NXP
upstream and update SRCREV to match release tags.

Update PREFERRED_VERSION of optee for mx8 to point to 3.10.0.imx.

Following notes are additions to the version update of recipes:
--------------------------
optee-client:
- Drop CFLAGS patch as it is already applied upstream

optee-os:
- Drop gcc10 patch as it is already applied upstream
- Add dependency on python3-pycryptodomex-native
- Drop NOWERROR flag as package is reworked to disable warnings by default
- Rework mkimage load address since [... cut]
commit be6a39f2 (2021-03-16): optee-imx: Upgrade to 3.10.0

Update all recipes to pull sources from imx_5.4.70_2.3.0 branch in NXP
upstream and update SRCREV to match release tags.

Update PREFERRED_VERSION of optee for mx8 to point to 3.10.0.imx.

Following notes are additions to the version update of recipes:
--------------------------
optee-client:
- Drop CFLAGS patch as it is already applied upstream

optee-os:
- Drop gcc10 patch as it is already applied upstream
- Add dependency on python3-pycryptodomex-native
- Drop NOWERROR flag as package is reworked to disable warnings by default
- Rework mkimage load address sinc [... cut]
closed 2021-03-16 by otavio