Freescale/meta-freescale #699: mesa build break on NXP BSP for i.MX with GPU

thochstein, 2021-03-24:
mesa has dropped support for `swrast`:

```
| ../mesa-21.0.0/meson.build:21:0: ERROR: Options "swrast" are not in allowed choices: "auto, i915, i965, r100, r200, nouveau"
```

thochstein, 2021-03-24:
I'm checking to see how far we are from being able to drop mesa entirely for i.MX with GPU. Neena has dropped `osmesa` and `swrast` and that seems to be building.

kraj, 2021-04-01:
you need something like this perhaps 

```patch
diff --git a/recipes-graphics/mesa/mesa_%.bbappend b/recipes-graphics/mesa/mesa_%.bbappend
index fce3802e..79a5fd16 100644
--- a/recipes-graphics/mesa/mesa_%.bbappend
+++ b/recipes-graphics/mesa/mesa_%.bbappend
@@ -23,20 +23,20 @@ PACKAGECONFIG_append_use-mainline-bsp = " gallium etnaviv kmsro freedreno"
 # For parts with no GPU, use gallium and dri
 PACKAGECONFIG_REMOVE_NXPBSP               = ""
 PACKAGECONFIG_APPEND_NXPBSP               = ""
-DRIDRIVERS_NXPBSP                         = ""
+GALLIUMDRIVERS_NXPBSP                     = ""
 #
 # For parts with GPU but no DRM, use gallium
 PACKAGECONFIG_REMOVE_NXPBSP_imxgpu        = "dri"
-DRIDRIVERS_NXPBSP_imxgpu                  = ""
+GALLIUMDRIVERS_NXPBSP_imxgpu              = ""
 #
 # For parts with GPU and DRM, use osmesa, dri, and swrast
 PACKAGECONFIG_REMOVE_NXPBSP_imxgpu_imxdrm = "gallium"
 PACKAGECONFIG_APPEND_NXPBSP_imxgpu_imxdrm = "osmesa"
-DRIDRIVERS_NXPBSP_imxgpu_imxdrm           = "swrast"
+GALLIUMDRIVERS_NXPBSP_imxgpu_imxdrm       = "swrast"
 #
 PACKAGECONFIG_remove_use-nxp-bsp = "${PACKAGECONFIG_REMOVE_NXPBSP}"
 PACKAGECONFIG_append_use-nxp-bsp = " ${PACKAGECONFIG_APPEND_NXPBSP}"
-DRIDRIVERS_use-nxp-bsp           = "${DRIDRIVERS_NXPBSP}"
+GALLIUMDRIVERS_use-nxp-bsp           = "${GALLIUMDRIVERS_NXPBSP}"
 
 BACKEND = \
     "${@bb.utils.contains('DISTRO_FEATURES', 'wayland', 'wayland', \

```

thochstein, 2021-04-06:
Thanks, Khem.

```
 PACKAGECONFIG_REMOVE_NXPBSP_imxgpu_imxdrm = "gallium"
 PACKAGECONFIG_APPEND_NXPBSP_imxgpu_imxdrm = "osmesa"
-DRIDRIVERS_NXPBSP_imxgpu_imxdrm           = "swrast"
+GALLIUMDRIVERS_NXPBSP_imxgpu_imxdrm       = "swrast"
```

Does this make sense to set `GALLIUMDRIVERS` when `gallium` is disabled?

I found that `gallium` was originally removed and `osmesa` added to work around build issues [here](https://github.com/Freescale/meta-freescale/commit/3b0fc734634f8f0db1f48fc13c64cacd3d99dd92). I don't see those build issues any more and propose in https://github.com/Freescale/meta-freescale/pull/702 to drop the specialization in this section and align on removing `dri` for `imxgpu_use-nxp-bsp`.

kraj, 2021-04-06:
you could still use swrast/llvmpile gallium driver but I think upsream is removing a bunch of code around this so perhaps we should wait until 2.22 and see where it ends up.

otavio, 2021-04-14:
Fixed in #702.

closed 2021-04-14 by otavio