Freescale/meta-freescale #704: Bootloader choosing logic breaks compatibility for existing machines

thochstein, 2021-04-01:
A problem I'm running into after https://github.com/Freescale/meta-freescale/issues/696  is my local machines are not defaulting to the same values.

```
cp: cannot stat '/home/bamboo/build/5.10-hn-full/fsl-imx-internal-fb/temp_build_dir/build_fsl-imx-internal-fb/tmp/work/imx6ul7d-poky-linux-gnueabi/u-boot-imx/1_2020.04-r0/git/mx6ul_14x14_evk_optee_config/u-boot.u-boot-imx': No such file or directory
```

We fixed it locally:

```
UBOOT_MAKE_TARGET_pn-u-boot-imx = "u-boot.imx"
UBOOT_SUFFIX_pn-u-boot-imx = "imx"
```

But it would be nice to fix this so we don't break compatibility.

_Originally posted by @thochstein in https://github.com/Freescale/meta-freescale/issues/696#issuecomment-811933523_

otavio, 2021-04-10:
I've successfully built the `u-boot-fslc`and `u-boot-imx` for ìmx6ulevk`.

You can see this below:
```
% ls tmp/*/deploy/images/*
tmp/fslc/deploy/images/imx6ulevk:
SPL
SPL-imx6ulevk
SPL-imx6ulevk-sd
SPL-imx6ulevk-v2020.10+gitAUTOINC+80c23498b4-r0-sd-v2020.10+gitAUTOINC+80c23498b4-r0
SPL-sd
u-boot.img
u-boot-fslc-initial-env-imx6ulevk-sd
u-boot-fslc-initial-env-imx6ulevk-sd-v2020.10+gitAUTOINC+80c23498b4-r0
u-boot-fslc-initial-env-sd
u-boot-imx6ulevk.img
u-boot-imx6ulevk.img-sd
u-boot-sd-v2020.10+gitAUTOINC+80c23498b4-r0.img
u-boot.img-sd

tmp/imx/deploy/images/imx6ulevk:
u-boot.imx                           u-boot-imx-initial-env-imx6ulevk-sd-2020.04-r0
u-boot-imx6ulevk.imx                 u-boot-imx-initial-env-sd
u-boot-imx6ulevk.imx-sd              u-boot-sd-2020.04-r0.imx
u-boot-imx-initial-env-imx6ulevk-sd  u-boot.imx-sd
```

I spent some time looking at this, and I think this is happening in an internal machine definition where you mix two or more machine types. Am I right?

If that is the case, it is indeed possible it'll need small adjustments as the logic has indeed change.

thochstein, 2021-04-10:
The problem occurs with our own machine config files. As I see it, the default settings for these uboot variables have changed and will thus break any machine config using the default. Here is the full set of overrides we are now using:

```
# Overrides for imx-base.inc
IMX_DEFAULT_BSP = "nxp"

# The default bootloader is set to -fslc in
# meta-freescale/cond/machine/include/imx-base.inc, so we must
# override it here.
IMX_DEFAULT_BOOTLOADER_imx = "u-boot-imx"

UBOOT_MAKE_TARGET_pn-u-boot-imx_mx6 = "u-boot.imx"
UBOOT_MAKE_TARGET_pn-u-boot-imx_mx7 = "u-boot.imx"
UBOOT_SUFFIX_pn-u-boot-imx_mx6 = "imx"
UBOOT_SUFFIX_pn-u-boot-imx_mx7 = "imx"
UBOOT_SUFFIX_pn-u-boot-imx_mx8 = "bin"
UBOOT_MAKE_TARGET_pn-u-boot-imx-mfgtool_mx6 = "u-boot.imx"
UBOOT_MAKE_TARGET_pn-u-boot-imx-mfgtool_mx7 = "u-boot.imx"
UBOOT_SUFFIX_pn-u-boot-imx-mfgtool_mx6 = "imx"
UBOOT_SUFFIX_pn-u-boot-imx-mfgtool_mx7 = "imx"
UBOOT_SUFFIX_pn-u-boot-imx-mfgtool_mx8 = "bin"
```

otavio, 2021-04-10:
It makes sense; I think we can close this as we understood what happened. Do you agree?

thochstein, 2021-04-10:
Would be nice not to break compatibility, but if you're okay with it, at least we know it breaks compatibility.

otavio, 2021-04-10:
We try hard not to break, but we eventually need; this seems to be a step in the right direction, so I think we can live with that. I am closing this now.

closed 2021-04-10 by otavio