Freescale/meta-freescale #707: LOCALVERSION (and MXC_GPU_VIV) does work since dunfell

msalvinik, 2021-04-06:
`CONFIG_LOCALVERSION` setting done in `fsl-kernel-localversion.bbclass` does not work anymore since dunfell, because since this release `linux-imx.inc` includes also `kernel-yocto.bbclass`.

In this class, task `kernel_configme` task overwrites the `.config` file generated by `preconfigure` task in `fsl-kernel-localversion.bbclass`

Same result for `CONFIG_MXC_GPU_VIV` set in `fsl-vivante-kernel-driver-handler.bbclass`.

otavio, 2021-04-10:
@msalvinik, thanks for reporting the local version bug; this is now fixed. I couldn't reproduce the GPU_VIV error, and it worked fine in my tests. Please open a specific issue for it with more detail about how to reproduce it.

closed 2021-04-10 by otavio

msalvinik, 2021-04-12:
@otavio  I'm sorry, my mistake: you are right, GPU_VIV worked fine. Thanks for fixing LOCALVERSION.

denix0, 2021-04-16:
This actually breaks standard capability to apply config fragments!

commit 49299998 (2021-04-10): classes: fsl-kernel-localversion: Fix task dependency

We need to do any change to the generated `.config` file after it is
finished to be modified. In the case of `kernel-yocto` class being in use,
we need to do it after `kernel_configme` task.

Fixes: c7e23876 ("linux-imx.inc: inherit kernel-yocto class")
Fixes: #707.

Change-Id: I7d32b47595dff269defdc5b3f98aa0b9d89b603a
commit ffeb0556 (2021-04-10): classes: fsl-kernel-localversion: Fix task dependency

We need to do any change to the generated `.config` file after it is
finished to be modified. In the case of `kernel-yocto` class being in use,
we need to do it after `kernel_configme` task.

Fixes: c7e23876 ("linux-imx.inc: inherit kernel-yocto class")
Fixes: #707.

Change-Id: I7d32b47595dff269defdc5b3f98aa0b9d89b603a
(cherry picked from commit 492999986124b5e960d0e21fc81d341458d6c5ad)
commit d6cb0a63 (2021-04-10): classes: fsl-kernel-localversion: Fix task dependency

We need to do any change to the generated `.config` file after it is
finished to be modified. In the case of `kernel-yocto` class being in use,
we need to do it after `kernel_configme` task.

Fixes: c7e23876 ("linux-imx.inc: inherit kernel-yocto class")
Fixes: #707.

Change-Id: I7d32b47595dff269defdc5b3f98aa0b9d89b603a
(cherry picked from commit 492999986124b5e960d0e21fc81d341458d6c5ad)