Freescale/meta-freescale #733: Applying config fragments no longer works after #707

denix0, 2021-04-16:
Fix for #707 in commit d6cb0a63d1a77da21e7f9cb7ca6561371b3fc896 breaks standard capability to apply config fragments:

foo.cfg:
CONFIG_FOO=y

linux%.bbappend:
SRC_URI += "file://foo.cfg"

otavio, 2021-04-16:
What mechanism are you using for the fragments? We are using the kernel-yocto schema.

denix0, 2021-04-16:
Yes, nothing special about it. Using dunfell. Kernel recipe is linux-fslc-imx on imx8mqevk.
The config fragment is pretty much like above. bbappend also has FILESEXTRAPATHS change, of course.
Used to work for quite some time until this change, i.e. 5.4.107. Then it stopped working. Reverting d6cb0a63d1a77da21e7f9cb7ca6561371b3fc896 makes it work again with 5.4.112

otavio, 2021-04-16:
I'll need to dig to find where it is getting wrong. I imagine we need to add an extra order somewhere.

I'll try to look at it during the weekend. No promises...

otavio, 2021-04-17:
I am not in position to check if this patch fixes the issue but I believe it should:

```
diff --git a/classes/fsl-kernel-localversion.bbclass b/classes/fsl-kernel-localversion.bbclass
index c73b60a6..b34f51cd 100644
--- a/classes/fsl-kernel-localversion.bbclass
+++ b/classes/fsl-kernel-localversion.bbclass
@@ -42,4 +42,4 @@ do_kernel_localversion() {
 	fi
 }
 
-addtask kernel_localversion before do_configure after do_patch do_kernel_configme
+addtask kernel_localversion before do_configure after do_patch do_kernel_configme do_kernel_metadata
```

If someone can check this, it'd be very helpful.

csarmat, 2021-04-17:
@otavio, that patch does not fix the issue. The order of tasks looks like this:

```
do_fetch (22418): log.do_fetch.22418
do_unpack (22422): log.do_unpack.22422
do_prepare_recipe_sysroot (22423): log.do_prepare_recipe_sysroot.22423
do_symlink_kernsrc (22456): log.do_symlink_kernsrc.22456
do_kernel_checkout (22458): log.do_kernel_checkout.22458
do_validate_branches (23248): log.do_validate_branches.23248
do_kernel_metadata (23272): log.do_kernel_metadata.23272
do_patch (23565): log.do_patch.23565
do_kernel_version_sanity_check (23566): log.do_kernel_version_sanity_check.23566
do_kernel_configme (23594): log.do_kernel_configme.23594
do_deploy_source_date_epoch (23595): log.do_deploy_source_date_epoch.23595
do_populate_lic (23596): log.do_populate_lic.23596
do_kernel_localversion (25469): log.do_kernel_localversion.25469
do_configure (25474): log.do_configure.25474
do_kernel_configcheck (25744): log.do_kernel_configcheck.25744
do_compile (6605): log.do_compile.6605
```
The patch that fixes this issue:
```patch
diff --git a/classes/fsl-kernel-localversion.bbclass b/classes/fsl-kernel-localversion.bbclass
index c73b60a6..c782a3d3 100644
--- a/classes/fsl-kernel-localversion.bbclass
+++ b/classes/fsl-kernel-localversion.bbclass
@@ -42,4 +42,4 @@ do_kernel_localversion() {
        fi
 }

-addtask kernel_localversion before do_configure after do_patch do_kernel_configme
+addtask kernel_localversion before do_configure after do_patch do_kernel_metadata
```
The order of tasks after using changes above (code works fine):
```
do_fetch (2816): log.do_fetch.2816
do_unpack (2820): log.do_unpack.2820
do_prepare_recipe_sysroot (2821): log.do_prepare_recipe_sysroot.2821
do_symlink_kernsrc (2854): log.do_symlink_kernsrc.2854
do_kernel_checkout (2856): log.do_kernel_checkout.2856
do_validate_branches (3647): log.do_validate_branches.3647
do_kernel_metadata (3671): log.do_kernel_metadata.3671
do_patch (3964): log.do_patch.3964
do_kernel_version_sanity_check (3965): log.do_kernel_version_sanity_check.3965
do_kernel_localversion (3993): log.do_kernel_localversion.3993
do_kernel_configme (3994): log.do_kernel_configme.3994
do_deploy_source_date_epoch (3995): log.do_deploy_source_date_epoch.3995
do_populate_lic (3996): log.do_populate_lic.3996
do_configure (5877): log.do_configure.5877
do_kernel_configcheck (6147): log.do_kernel_configcheck.6147
do_compile (19086): log.do_compile.19086
```

As you can see the difference is that in working case the task **do_kernel_localversion** runs before **do_kernel_configme**, **do_deploy_source_date_epoch**, **do_populate_lic**. And the patch as below one does not work also:
```patch
diff --git a/classes/fsl-kernel-localversion.bbclass b/classes/fsl-kernel-localversion.bbclass
index c73b60a6..e5f94379 100644
--- a/classes/fsl-kernel-localversion.bbclass
+++ b/classes/fsl-kernel-localversion.bbclass
@@ -42,4 +42,4 @@ do_kernel_localversion() {
        fi
 }

-addtask kernel_localversion before do_configure after do_patch do_kernel_configme
+addtask kernel_localversion before do_configure do_deploy_source_date_epoch do_populate_lic after do_patch do_kernel_configme
```
So, it looks like the reason for this issue is that **do_kernel_configme** starts before **do_kernel_localversion**.

otavio, 2021-04-17:
Awsome. Do you want to send a PR with the fix?

csarmat, 2021-04-17:
I might do this but I will not manage to do this till Monday. So, if you can commit this change then do it

otavio, 2021-04-17:
@denix0 can you test #734? It seems it fixes the issue.

denix0, 2021-04-18:
yes, I can confirm the fix from #734 works in dunfell

PR #734 (2021-04-18): classes: fsl-kernel-localversion: Fix config fragments apply
We have to ensure the task run do_kernel_metadata so the config
fragments can be applied properly.

Refs: #707.
Fixes: #733.
Fixes: 49299998 ("classes: fsl-kernel-localversion: Fix task dependency")
Fixes: c7e23876 ("linux-imx.inc: inherit kernel-yocto class")
commit 5943060c (2021-04-18): classes: fsl-kernel-localversion: Fix config fragments apply

We have to ensure the task run do_kernel_metadata so the config
fragments can be applied properly.

Refs: #707.
Fixes: #733.
Fixes: 49299998 ("classes: fsl-kernel-localversion: Fix task dependency")
Fixes: c7e23876 ("linux-imx.inc: inherit kernel-yocto class")
commit 2e8dd740 (2021-04-18): classes: fsl-kernel-localversion: Fix config fragments apply

We have to ensure the task run do_kernel_metadata so the config
fragments can be applied properly.

Refs: #707.
Fixes: #733.
Fixes: 49299998 ("classes: fsl-kernel-localversion: Fix task dependency")
Fixes: c7e23876 ("linux-imx.inc: inherit kernel-yocto class")

(cherry picked from commit 5943060c8ea892054f2d995d398e21bcb2237543)
commit e43a9f2e (2021-04-18): classes: fsl-kernel-localversion: Fix config fragments apply

We have to ensure the task run do_kernel_metadata so the config
fragments can be applied properly.

Refs: #707.
Fixes: #733.
Fixes: 49299998 ("classes: fsl-kernel-localversion: Fix task dependency")
Fixes: c7e23876 ("linux-imx.inc: inherit kernel-yocto class")

(cherry picked from commit 5943060c8ea892054f2d995d398e21bcb2237543)
closed 2021-04-18 by otavio