Freescale/meta-freescale #759: Cannot run core-image-weston + suggestions for fixes

dv1, 2021-04-29:
I found two issues with the weston support in current master.

1. When I tried to build core-image-weston, there was a build error, since xserver-xorg-wayland was not found. As documented in https://lists.yoctoproject.org/g/meta-freescale/message/24666 , the reason for this is that xserver-xorg-wayland was replaced with a standalone xwayland recipe. This change in weston_9.0.0.imx.bb fixed it for me:
    -RDEPENDS_${PN}-xwayland += "xserver-xorg-xwayland"
    +RDEPENDS_${PN}-xwayland += "xwayland"

2. After building, I booted the machine, but weston did not show up. It turns out that the galcore kernel module was not installed, and the imx-variant of weston needs it. I added this line to weston_9.0.0.imx.bb , which fixed this:
    +RDEPENDS_${PN}_mx8 += "kernel-module-galcore"

I post these as an issue instead of a MR since I am not sure if these are the proper ways to fix these problems.

otavio, 2021-04-29:
The first, yes. The second, better way to fix this is to add a rrecommends to the kernel module on one of the `imx-gpu-viv` libraries.

Could you cook two PRs for this?

dv1, 2021-04-29:
I tried to add `kernel-module-galcore` to the rrecommends of all `imx-gpu-viv` libraries, and yet, `kernel-module-galcore` was not added to built images. So, I'll try to add it to the rdepends of `libgal-imx` instead.

otavio, 2021-04-29:
It should be rrecommends.

dv1, 2021-04-29:
OK, done, [this is the MR](https://github.com/Freescale/meta-freescale/pull/760). I close this now to continue discussions over there instead.

closed 2021-04-29 by dv1