Freescale/meta-freescale #783: imx6 weston-init build error gatesgarth

chrisdimich, 2021-05-14:
I am seeing an issue with the BSP when building nitrogen6x on gatesgarth. The issue is in weston-init.bbappend. There are variables defined which essentially will uncomment out lines in weston.ini. The problem is, due to the MACHINE overrides in nitrogen6x, this gets called twice and throws an error:

ERROR: weston-init-1.0-r0 do_install: Commented setting '#use-g2d=1' not found in file /backup/jenkins/jobs/yocto_gatesgarth/workspace/build-boundary-xwayland-nitrogen6x/tmp/work/nitrogen6x-fslc-linux-gnueabi/weston-init/1.0-r0/image/etc/xdg/weston/weston.ini

So I need to comment out/remove one of these in weston-init.bbappend in meta-freescale:

#INI_UNCOMMENT_ASSIGNMENTS_append_mx6dl = " \
#use-g2d=1 \
#"
INI_UNCOMMENT_ASSIGNMENTS_append_mx6q = " \
    use-g2d=1 \
"
Then it builds fine. 

@gibsson and I feel it's better to remove all the mx6/mx7/mx8 checks and just have:

INI_UNCOMMENT_ASSIGNMENTS_append_imxgpu2d = " \
    use-g2d=1 \
"

otavio, 2021-05-14:
The error when uncommenting the block has been fixed in master: https://github.com/Freescale/meta-freescale/commit/258063936a4256a40ba55bc2305c9989ef09598e; please prepare a backport PR so we merge this fix.

Please prepare a PR for the proposal of the override change. Please mention on commit log  it affects following SoCs:

```sh
% git grep :imxgpu2d: | sed 's,.*MACHINEOVERRIDES_EXTENDER_\(.*\) .*=.*,\1,g'
mx6q
mx6dl
mx6sx
mx6sl
mx7ulp
mx8qm
mx8mm
mx8mp
mx8qxp
```

chrisdimich, 2021-05-14:
Thanks @otavio ! Will do.

otavio, 2021-05-14:
The override change proposal should be a new PR against master.

thochstein, 2021-05-14:
Generally using `_imxgpu2d` is fine, but my understanding for 8QM and 8QXP is that the 3D rendering will provide better performance.

otavio, 2021-05-17:
@chrisdimich please make a PR reworking the append for the override.

chrisdimich, 2021-05-17:
No problem, will do.

chrisdimich, 2021-05-18:
Created PR #787

gibsson, 2021-06-29:
@chrisdimich if this issue is gone as the PR was merged, please close it.

closed 2021-06-29 by chrisdimich