Freescale/meta-freescale #788: Kernel crash on imx8qm-mek with fslc kernel

sapiippo, 2021-05-19:
The linux-fslc-imx kernel (5.4.114+gf5b1a0c923d0) crashes on imx8qm-mek board on boot, while the linux-imx kernel (5.4.70+g5f2220e87dd8) works.

```
[    1.903184] SError Interrupt on CPU4, code 0xbf000000 -- SError
[    1.903186] CPU: 4 PID: 1 Comm: swapper/0 Not tainted 5.4.114+gf5b1a0c923d0 #1
[    1.903186] Hardware name: Freescale i.MX8QM MEK (DT)
[    1.903187] pstate: 40000005 (nZcv daif -PAN -UAO)
[    1.903188] pc : caam_probe+0xa40/0xe5c
[    1.903189] lr : caam_probe+0x148/0xe5c
[    1.903190] sp : ffff80001004baf0
[    1.903191] x29: ffff80001004baf0 x28: 0000000000000000 
[    1.903193] x27: ffff0008f65b2800 x26: ffff8000115d3b00 
[    1.903195] x25: ffff800011548638 x24: ffff800011bfc000 
[    1.903197] x23: ffff800013b30f00 x22: ffff0008f65b2810 
[    1.903199] x21: ffff800013b00000 x20: 0000000001934211 
[    1.903201] x19: ffff0008f7168680 x18: ffff800011617328 
[    1.903203] x17: ffff0008f7540a00 x16: ffff0008f75405c8 
[    1.903204] x15: ffffffffffffffff x14: ffffffffff000000 
[    1.903206] x13: ffffffffffffffff x12: 0000000000000001 
[    1.903208] x11: 0000000000000007 x10: 0101010101010101 
[    1.903210] x9 : fffffffffffffffc x8 : 0000000000000008 
[    1.903211] x7 : 0000000000000004 x6 : 0000000000000000 
[    1.903214] x5 : 0000000000000001 x4 : 0000000000000003 
[    1.903215] x3 : ffff0008ff435890 x2 : 0000000000000000 
[    1.903217] x1 : 0000000000000001 x0 : 0000000000000000 
[    1.903220] Kernel panic - not syncing: Asynchronous SError Interrupt
[    1.903221] CPU: 4 PID: 1 Comm: swapper/0 Not tainted 5.4.114+gf5b1a0c923d0 #1
[    1.903222] Hardware name: Freescale i.MX8QM MEK (DT)
[    1.903223] Call trace:
[    1.903223]  dump_backtrace+0x0/0x140
[    1.903224]  show_stack+0x14/0x20
[    1.903225]  dump_stack+0xb4/0x110
[    1.903226]  panic+0x158/0x324
[    1.903226]  __stack_chk_fail+0x0/0x1c
[    1.903227]  arm64_serror_panic+0x74/0x80
[    1.903228]  enter_from_user_mode+0x0/0x4
[    1.903229]  el1_error+0x84/0xf8
[    1.903229]  caam_probe+0xa40/0xe5c
[    1.903230]  platform_drv_probe+0x50/0xa0
[    1.903231]  really_probe+0xd8/0x420
[    1.903232]  driver_probe_device+0x54/0xe4
[    1.903233]  device_driver_attach+0xb4/0xc0
[    1.903233]  __driver_attach+0x80/0x114
[    1.903234]  bus_for_each_dev+0x6c/0xbc
[    1.903235]  driver_attach+0x20/0x30
[    1.903236]  bus_add_driver+0xfc/0x1e0
[    1.903236]  driver_register+0x74/0x120
[    1.903237]  __platform_driver_register+0x44/0x50
[    1.903238]  caam_driver_init+0x18/0x20
[    1.903239]  do_one_initcall+0x50/0x1b0
[    1.903240]  kernel_init_freeable+0x194/0x23c
[    1.903240]  kernel_init+0x10/0x100
[    1.903241]  ret_from_fork+0x10/0x24
[    1.903255] SMP: stopping secondary CPUs
[    1.903256] Kernel Offset: disabled
[    1.903257] CPU features: 0x0002,2100600c
[    1.903258] Memory Limit: none
```

```
Build Configuration:
BB_VERSION           = "1.49.2"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-fsl-linux"
MACHINE              = "imx8qm-mek"
DISTRO               = "fsl-wayland"
DISTRO_VERSION       = "3.4-snapshot-20210519"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa72-cortexa53 crypto"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "master:eec64066073428e38d5b0b6947ff302943188f65"
meta-python          
meta-oe              = "master:589aa162cead42acdd7e8dbd7c0243b95e341f19"
meta-freescale       = "master:2c4efeea83b9edeb63167ed79163c1e55baab7e6"
meta-freescale-distro = "master:d94fbaad1e6355c144090b2e782de6193d42dadc"
```

onurcalili, 2021-05-20:
My problem caused by the change in source code if mmc controller in linux kernel. The change in mmc controller source code changes the indexing of mmc devices indexing (ex. /dev/mmcblk1 becomes /dev/mmcblk0, /dev/mmcblk3 becomes /dev/mmcblk1 if sdhc2 and sdhc4 are used in kernel). So I patch the kernel as follows; 

diff --git a/drivers/mmc/core/core.c b/drivers/mmc/core/core.c
index 460a456bcdd2..986fc225f19a 100644
--- a/drivers/mmc/core/core.c
+++ b/drivers/mmc/core/core.c
@@ -52,6 +52,8 @@
 
 static const unsigned freqs[] = { 400000, 300000, 200000, 100000 };
 
+static int __mmc_max_reserved_idx = -1;
+
 /*
  * Enabling software CRCs on the data blocks can be a significant (30%)
  * performance cost, and for other reasons may not always be desired.
@@ -2392,10 +2394,46 @@ void mmc_stop_host(struct mmc_host *host)
 	mmc_release_host(host);
 }
 
+/*
+ * mmc_first_nonreserved_index() - get the first index that
+ * is not reserved
+ */
+int mmc_first_nonreserved_index(void)
+{
+	return __mmc_max_reserved_idx + 1;
+}
+EXPORT_SYMBOL(mmc_first_nonreserved_index);
+
+/*
+ * mmc_get_reserved_index() - get the index reserved for this host
+ * Return: The index reserved for this host or negative error value
+ *        if no index is reserved for this host
+ */
+int mmc_get_reserved_index(struct mmc_host *host)
+{
+	return of_alias_get_id(host->parent->of_node, "mmc");
+}
+EXPORT_SYMBOL(mmc_get_reserved_index);
+
+static void mmc_of_reserve_idx(void)
+{
+	int max;
+
+	max = of_alias_max_index("mmc");
+	if (max < 0)
+		return;
+
+	__mmc_max_reserved_idx = max;
+	pr_debug("MMC: reserving %d slots for of aliases\n",
+			__mmc_max_reserved_idx + 1);
+}
+
 static int __init mmc_init(void)
 {
 	int ret;
 
+	mmc_of_reserve_idx();
+
 	ret = mmc_register_bus();
 	if (ret)
 		return ret;
diff --git a/drivers/mmc/core/host.c b/drivers/mmc/core/host.c
index f7339590a84c..7c6314cc3002 100644
--- a/drivers/mmc/core/host.c
+++ b/drivers/mmc/core/host.c
@@ -433,6 +433,7 @@ EXPORT_SYMBOL(mmc_of_parse_voltage);
 struct mmc_host *mmc_alloc_host(int extra, struct device *dev)
 {
 	int err;
+	int alias_id;
 	struct mmc_host *host;
 
 	host = kzalloc(sizeof(struct mmc_host) + extra, GFP_KERNEL);
@@ -442,7 +443,14 @@ struct mmc_host *mmc_alloc_host(int extra, struct device *dev)
 	/* scanning will be enabled when we're ready */
 	host->rescan_disable = 1;
 
-	err = ida_simple_get(&mmc_host_ida, 0, 0, GFP_KERNEL);
+	alias_id = mmc_get_reserved_index(host);
+	if (alias_id >= 0)
+		err = ida_simple_get(&mmc_host_ida, alias_id,
+				alias_id + 1, GFP_KERNEL);
+	else
+		err = ida_simple_get(&mmc_host_ida,
+					mmc_first_nonreserved_index(),
+					0, GFP_KERNEL);
 	if (err < 0) {
 		kfree(host);
 		return NULL;

zandrey, 2021-05-25:
@sapiippo 

Did you try to bisect the kernel repo to find the offending commit?

The upgrade only included commits from _stable korg_, therefore it would be beneficial to know which commit caused this behavior.

Also, did you try to build and run _mainline_ BSP as a sanity check to see if it exposes the same problem? That BSP flavor should contain the latest `linux-5.12.y` branch and can be used to separate NXP-specific commits from upstream commits.

For the record, I've just booted the _NXP BSP_ on `imx8mpevk` machine, and saw no crashes occurred:
```
root@imx8mpevk:~# uname -a
Linux imx8mpevk 5.4.119+gaf18e9f3baa8 #1 SMP PREEMPT Fri May 14 21:10:12 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux
```

-- andrey

zandrey, 2021-05-26:
@onurcalili Thanks for pointing this out!

A colleague of mine came up with the same investigation and the patch, I'd queue it in for the next kernel update.

-- andrey

zandrey, 2021-05-27:
@sapiippo @onurcalili 

I've opened a PR #796 with kernel update that includes the patch to fix the merge issue. Please test your setup with this bump.

-- andrey

otavio, 2021-05-27:
I merged it together with the other Linux kernel bumps. Once it is confirmed to fix the issue, we can close this.

otavio, 2021-05-31:
Fixed in #796.

closed 2021-05-31 by otavio