Freescale/meta-freescale #791: UBOOT_ENTRYPOINT not set for mx8m machines built with mainline bsp

pberginkonsult, 2021-05-21:
When building mainline bsp for, as example, imx8mm_lpddr4_evk the variable UBOOT_ENRTYPOINT is not set. This causes the fitImage generation to be corrupt as 'Load Address' in the uImage header gets wrong value.

The MACHINEOVERRIDES mx8m was removed from mainline bsp in the commit 069482f24b3e07c8118f4cf9af7564220ac5e82b. This caused that this change ac25d3a3d09a2707d9ee614d3ed506fd2ebe2f83, setting DEFAULTTUNE and UBOOT_ENTRYPOINT for mx8m machines will fail for mainline bsp.

zandrey, 2021-05-22:
@pberginkonsult

`master` branch contains U-Boot `2021.07-rc1`, which has the _distro boot_ enabled by default. In this case, I'm not sure if the kernel FIT load support is still enabled and operable.

In fact, even before ac25d3a3d09a2707d9ee614d3ed506fd2ebe2f83 ("imx-base.inc: Consolidate DEFAULTTUNE and UBOOT_ENTRYPOINT for 8M") was merged - original `MACHINEOVERRIDES` were anyway filtered out in the _mainline_ BSP, which suggests that it was not working even before.

Is it necessary that you use FIT image load, or maybe you can switch to use _extlinux_ instead to align with upstream U-Boot? You can see the 122bfa162197075949adf826a39119cc21c6955f ("conf: machine: imx8m[m,n,p]: enable extlinux support") for the reference and reasoning.

-- andrey

pberginkonsult, 2021-05-24:
> In fact, even before [ac25d3a](https://github.com/Freescale/meta-freescale/commit/ac25d3a3d09a2707d9ee614d3ed506fd2ebe2f83) ("imx-base.inc: Consolidate DEFAULTTUNE and UBOOT_ENTRYPOINT for 8M") was merged - original `MACHINEOVERRIDES` were anyway filtered out in the _mainline_ BSP, which suggests that it was not working even before.
>
Could be tha case, haven't checked. 
> Is it necessary that you use FIT image load, or maybe you can switch to use _extlinux_ instead to align with upstream U-Boot? You can see the [122bfa1](https://github.com/Freescale/meta-freescale/commit/122bfa162197075949adf826a39119cc21c6955f) ("conf: machine: imx8m[m,n,p]: enable extlinux support") for the reference and reasoning.

In my project I don't want a generic distro boot. I need to bundle an initramfs and by using fitImage I can easily bundle kernel and initramfs together. I'm not blocked by this as the simple workaround is to add UBOOT_ENTRYPOINT in my config and problem is solved.

From my perspective the issue here is that we have variables that seems to be valid for the config but is filtered out. UBOOT_ENTRYPOINT is used by kernel-fitimage.bbclass from oe-core and whe I use imx8mm-lpddr4-evk.conf I assume this variable is set. 

I have tried to build with and without ac25d3a3d09a2707d9ee614d3ed506fd2ebe2f83 and from  a quick check in buildhistory the difference is that when I revert the commit imx-boot is added to the image. But haven't digged down the actual dependecy chain but my guess is that this is the root cause:
```
../sources/meta-freescale/conf/machine/include/imx-base.inc:WKS_FILE_DEPENDS_append_mx8m = " imx-boot"
```

pberginkonsult, 2021-09-10:
> What is the status of this issue? I have noted some change in the master recently, could you, please comment what is your thoughts now?
> 
> @pberginkonsult

@angolini I've tested this again and I can still see the issue.

My build setup:
```Build Configuration:
BB_VERSION           = "1.51.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-fslc-linux"
MACHINE              = "imx8mp-lpddr4-evk"
DISTRO               = "fslc-wayland"
DISTRO_VERSION       = "3.4-snapshot-20210910"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa53 crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "master:fd9c3a5a14fb54268707f6ece6a5fa11e4378200"
meta-oe              
meta-multimedia      
meta-python          = "master:c2a8057647a0483e7faf1b55d4d89d4c9647e597"
meta-freescale       = "HEAD:41a108b3453b5855d090a62dd451855ab6f9dad9"
meta-freescale-3rdparty = "master:10a3c870807fa9df36fb21ad036860789b491f78"
meta-freescale-distro = "master:819524f714226cd88bfb0eb2646aabbe424bf7c5"
```

Steps to reproduce is to add the following configuration to the build:
```
KERNEL_CLASSES += "kernel-fitimage"
KERNEL_IMAGETYPES:append += "fitImage"
```
And build the kernel. This will make a fitImage with this header:
```$ mkimage -l /tmp/fitImage
FIT description: Kernel fitImage for FSLC Wayland/5.14+gitAUTOINC+35f3f4421d/imx8mp-lpddr4-evk
Created:         Mon Aug 30 14:05:58 2021
 Image 0 (kernel-1)
  Description:  Linux kernel
  Created:      Mon Aug 30 14:05:58 2021
  Type:         Kernel Image
  Compression:  gzip compressed
  Data Size:    11529168 Bytes = 11258.95 KiB = 11.00 MiB
  Architecture: AArch64
  OS:           Linux
  Load Address: 0x01314c40
  Entry Point:  0x01314c40
  Hash algo:    sha256
  Hash value:   f3be5f892cea457538237ea4c9e03281384721c02e26183223fa2019af8bda2d
 Image 1 (fdt-freescale_imx8mp-evk.dtb)
  Description:  Flattened Device Tree blob
  Created:      Mon Aug 30 14:05:58 2021
  Type:         Flat Device Tree
  Compression:  uncompressed
  Data Size:    24911 Bytes = 24.33 KiB = 0.02 MiB
  Architecture: AArch64
  Hash algo:    sha256
  Hash value:   63ba9ece2314032434753ac0a34d4e4986695cb1bc98d3731fd7b5d2c890e6b4
 Default Configuration: 'conf-freescale_imx8mp-evk.dtb'
 Configuration 0 (conf-freescale_imx8mp-evk.dtb)
  Description:  1 Linux kernel, FDT blob
  Kernel:       kernel-1
  FDT:          fdt-freescale_imx8mp-evk.dtb
  Hash algo:    sha256
  Hash value:   unavailable
```

And when booting on target:
```u-boot=> bootm ${fitimage_addr}
## Loading kernel from FIT Image at 43800000 ...
   Using 'conf-freescale_imx8mp-evk.dtb' configuration
   Trying 'kernel-1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  gzip compressed
     Data Start:   0x43800100
     Data Size:    11529168 Bytes = 11 MiB
     Architecture: AArch64
     OS:           Linux
     Load Address: 0x01314c40
     Entry Point:  0x01314c40
     Hash algo:    sha256
     Hash value:   f3be5f892cea457538237ea4c9e03281384721c02e26183223fa2019af8bda2d
   Verifying Hash Integrity ... sha256+ OK
## Loading fdt from FIT Image at 43800000 ...
   Using 'conf-freescale_imx8mp-evk.dtb' configuration
   Trying 'fdt-freescale_imx8mp-evk.dtb' fdt subimage
     Description:  Flattened Device Tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x442fedec
     Data Size:    24911 Bytes = 24.3 KiB
     Architecture: AArch64
     Hash algo:    sha256
     Hash value:   63ba9ece2314032434753ac0a34d4e4986695cb1bc98d3731fd7b5d2c890e6b4
   Verifying Hash Integrity ... sha256+ OK
   Booting using the fdt blob at 0x442fedec
   Uncompressing Kernel Image
"Error" handler, esr 0xbf000002
elr: 000000004026fbc4 lr : 000000003e233970 (reloc)
elr: 0000000054f90bc4 lr : 0000000052f54970
x0 : 0000000052bf6e78 x1 : 0000000000000014
x2 : 0000000001314c43 x3 : 000000004380019d
x4 : 0000000052f5431c x5 : 00000000000ae77a
x6 : 0000000000000000 x7 : 00000000000000fa
x8 : 0000000052f53a30 x9 : 00000000d4a4e2f5
x10: 00000000442fecca x11: 0000000000000009
x12: 0000000000000000 x13: 0000000005314b3e
x14: 00000000000001ff x15: 000000000000003f
x16: 0000000000000000 x17: 0000000052f53f80
x18: 0000000052bffdd0 x19: 0000000000000001
x20: 0000000000000000 x21: 0000000000000000
x22: 00000000ffffffff x23: 0000000000000000
x24: 0000000000000000 x25: 0000000000000000
x26: 0000000000afeb38 x27: 0000000000000017
x28: 0000000052f53ac0 x29: 0000000052bf6d10

Code: 9ac120c6 8b060084 8b0400a5 11004021 (8a0501c4) 
Resetting CPU ...

resetting ...
```

The solution to get it working is to add `UBOOT_ENTRYPOINT = "0x40480000"` to the configuration. That will produce a fitImage with "Load Address" and "Entry Point" set in the header and the kernel boot as expected.

I would like some suggestion how we best solve this or if fitImage is unsupported in the standard config. Then I can try to create a PR for it.

otavio, 2021-09-10:
This happens on mainline only, correct?

pberginkonsult, 2021-09-10:
> This happens on mainline only, correct?

Yes, correct.

When setting `IMX_DEFAULT_BSP="nxp"` the variable `UBOOT_ENTRYPOINT` will get a value because `mx8m` is in `OVERRIDES`.

zandrey, 2021-09-10:
Hello @pberginkonsult 

If you set the `UBOOT_ENTRYPOINT` in your machine unconditional and regardless of BSP flavor - does this solve the issue with _FIT_ booting?

I was having the similar issue with `DEFAULTTUNE` and has it set in 1d044d1 for mainline BSP in include files for `use-mainline-bsp` override.

Perhaps the same could be done for `UBOOT_ENTRYPOINT` as well in a similar manner?

-- andrey

pberginkonsult, 2021-09-10:
> Hello @pberginkonsult
> 
> If you set the `UBOOT_ENTRYPOINT` in your machine unconditional and regardless of BSP flavor - does this solve the issue with _FIT_ booting?
> 

Yes. This issue has a very simple solution and it is to define `UBOOT_ENTRYPOINT` with correct address somewhere in your configuration, machine config or elsewhere.

> I was having the similar issue with `DEFAULTTUNE` and has it set in [1d044d1](https://github.com/Freescale/meta-freescale/commit/1d044d1351dcff2603ae064875bcd976a5bc4c93) for mainline BSP in include [...]

Good suggestion. As you say this is a similar issue. I will use 1d044d1 as template and send a PR.

otavio, 2021-09-10:
I am wondering if we can come up with a generic solution ... still scratching my head ...

otavio, 2021-10-05:
I believe I have come up with a solution for this. I am working on a PoC to propose the needed changes. Essentially, we extend the overrides to a `generic`, `nxp-bsp`, and `mainline-bsp`.

So, essentially, the `mx8m` override would be split into:

- `mx8m-generic`: common use for both BSP variants
- `mx8m-nxp-bsp`: specific for NXP BSP use
- `mx8m-mainline-bsp`: specific for Mainline BSP use

The extender mechanism would be responsible for extending the override list to include the `generic` override. We can then use the three different variants to handle the metadata correctly.

Generically speaking, the conversion can be mostly automated. The existing overrides which don't use the `use-mainline-bsp` will be moved to the `nxp-bsp` one. I plan to use the generic override (for example, `mx8m`) to automatically expand to the final BSP override automatically so simple machines would need no changes. In contrast, more advanced ones would still be simple to convert.

What do you thing about this proposal?

angolini, 2021-12-01:
Do we have any update on this issue?
@otavio @pberginkonsult

pberginkonsult, 2021-12-01:
Sorry, no update from my side. I see that I missed to comment on Otavios suggestion. I think it looks good.

IniterWorker, 2022-01-18:
After @mferland pointed to me a `loadaddr` issue in the fitImage for mfgtool, I hurt my head on the same question:
> I am wondering if we can come up with a generic solution ... still scratching my head ...

### Current workaround:

It's to enforce UBOOT_ENTRYPOINT = "{your machine spec addr}" in your machie.conf.

otavio, 2022-02-19:
@pberginkonsult @IniterWorker please check the PR above. It seems to greatly fix this issue.

closed 2022-02-23 by otavio