Freescale/meta-freescale #795: Mesa build is broken after the upstream upgrade to 21.1.1

zandrey, 2021-05-26:
Upstream OE commit 8579142183 ("mesa: update 21.0.3 -> 21.1.1") has bumped up the version of _Mesa_ to `21.1.1.`

This causes the patch [0002-dri-bring-back-use-flags-for-createImageWithModifier.patch](../blob/master/recipes-graphics/mesa/mesa/0002-dri-bring-back-use-flags-for-createImageWithModifier.patch) not to apply clean and build breaks.

Patches were merged with commit bfb9e699 ("mesa/etnaviv: Add patches to fix redraw"), which lists `Upstream-Status: Submitted [https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/7603]` in patches submitted into the layer.

The mentioned _Mesa_ PR however does not have the patch included in the layer, as it has been updated several times after the original submission, which eventually lead to several patches dropped (including the "offending" one).

I'm not sure if the commit bfb9e699 ("mesa/etnaviv: Add patches to fix redraw") applies to the new version, and should be either re-worked or reverted. Reverting the patch does solve the compilation issue, but I cannot test that the runtime is sane.

-- andrey

otavio, 2021-05-26:
@schnitzeltony is it possible for you to take a look at this?

schnitzeltony, 2021-05-26:
Yes I should do that because there were promising changes at the etnaviv area. Frankly spoken: I was waiting for this update. 
Bad news: Daily business makes it impossible for me to announce some schedule - hope this is something for weekend...

otavio, 2021-05-31:
Fixed in #797.

closed 2021-05-31 by otavio