Freescale/meta-freescale #798: MACHINE_SOCARCH_FILTER breaks build

schnitzeltony, 2021-05-31:
With latest oe-core master I get:

> ERROR: extra-cmake-modules-native-5.82.0-r0 do_prepare_recipe_sysroot: Manifest /home/superandy/tmp/oe-core-glibc/sstate-control/manifest-x86_64_x86_64-nativesdk-qtbase.populate_sysroot not found in imx6qdl_variscite_som cortexa9hf-neon cortexa9t2hf-neon cortexa9t2hf-vfp cortexa9hf-neon cortexa9hf-vfp armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
> ERROR: Logfile of failure stored in: /home/superandy/tmp/oe-core-glibc/work/x86_64-linux/extra-cmake-modules-native/5.82.0-r0/temp/log.do_prepare_recipe_sysroot.2524
> ERROR: Task (virtual:native:/home/superandy/data/oe-core/sources/meta-qt5-extra/recipes-kde/kf5/untier/extra-cmake-modules/extra-cmake-modules.bb:do_prepare_recipe_sysroot) failed with exit code '1'
> 

Warning seen for long time now get errors by https://github.com/openembedded/openembedded-core/commit/64b89f3c8fc31842256c482a3039d90d3f12c1cc.

Some analysis:
* qtbase is marked as SOC_ARCH [1]
* as soon as a **native** recipe DEPENDS on **qtbase (or other recipe marked as SOC-ARCH)**, manifest cannot be found. This happens often to me because in meta-qt5-extra there are many native recipes inheriting cmake_qt5 which adds qtbase to depends.
* removing `INHERIT += "fsl-dynamic-packagearch"`- few lines below [1] - fixes build

[1] https://github.com/Freescale/meta-freescale/blob/733e1683ce8895a960c01a3ee1fce40118d82982/conf/machine/include/imx-base.inc#L246

shr-project, 2021-05-31:
native recipe depending on anything TUNE_PKGARCH, MACHINE_ARCH or even allarch is definitely wrong, right?

sstate-diff-machine.sh would report the same.

The warning was changed to error 4 months ago (before hardknott release):
https://git.openembedded.org/openembedded-core/commit/meta/lib/oe/sstatesig.py?id=95607a26854d873399d2b9d7e5fcbffc0cbdba4c

the linked commit from master only changes it from error to fatal.

schnitzeltony, 2021-05-31:
> the linked commit from master only changes it from error to fatal.

As far as I remember build continued. Note that I start builds for imx on master only to be a good boy: Our companies builds are based on dunfell.
> native recipe depending on anything TUNE_PKGARCH, MACHINE_ARCH or even allarch is definitely wrong, right?

Yes maybe but it happens here and there. I started to fix but stopped - there were too many cases. BTW 90% of my trouble is caused by meta-qt5/cmake_qt5.bbclass: It adds qtbase-native and qtbase to DEPENDS. As soon as a native recipe inherits it, the damage is done. Shall I send a fix there as a start?

schnitzeltony, 2021-06-01:
@otavio: Hope you did not invest too much effort on this one: as far as I can see there is no activity required here.

@shr-project Thanks for bringing me back to the right track.

Shall we close this one or leave it open for others?

otavio, 2021-06-01:
@schnitzeltony I didn't get to it but I am glad you got it right :-) I'll close this as there is no point in leaving this open for good.

closed 2021-06-01 by otavio