Freescale/meta-freescale #808: Upgrade [master] to NXP release 5.10.9-1.0.0

zandrey, 2021-06-11:
Hello all,

I see that NXP started to provide binary images, which are based on the new release `5.10.9-1.0.0`. There are also branches and tags available for `meta-imx` on the _CodeAurora_, which corresponds to this release:
- Source URL: https://source.codeaurora.org/external/imx/meta-imx
- Branch: _gatesgarth-5.10.9-1.0.0_
- Tag: _lf-5.10.y-1.0.0_

Since there is no announcement from NXP regarding the update of the layers, and it is not listed in [Alpha / Beta BSPs for Microprocessors](https://www.nxp.com/pages/alpha-beta-bsps-for-microprocessors:IMXPRERELEASES), I was just wondering should the upgrade effort be started here, or it makes sense to rather wait until this release is fully announced?

-- andrey

otavio, 2021-06-11:
@thochstein, what do you think? I want to start to upgrade so we can have it on the October release if possible.

thochstein, 2021-06-11:
We've got another release coming by end of month, 5.10.35. Should we wait for that?

otavio, 2021-06-15:
@thochstein I'd say to wait for it and once it is released we can start the work.

otavio, 2021-06-29:
So it is time :-)

thochstein, 2021-06-29:
Okay, I'll get started on kernel and uboot.

angolini, 2021-08-06:
#818

closed 2021-08-06 by angolini