Freescale/meta-freescale #821: Driver SGTL5000 codec audio freescale

cristianod3, 2021-07-01:
Does meta-freescale contain the driver to control the freescale SGTL5000 codec chip?

pberginkonsult, 2021-07-01:
meta-freescale contains meta data to help you build your Linux distribution. The driver for SGTL5000 can be found in the Linux kernel tree. There are a couple of kernel recipes in this layer that you can use to build your kernel.

cristianod3, 2021-07-01:
in the linux kernel files tree i found sgtl500.c and sgtl5000.h; how to add these files when compiling the new kernel with yocto? If you could explain it to me in more detail, as I am still learning to work with yocto.
thanks

zandrey, 2021-07-01:
@cristianod3

The issue you're facing now is not related to Yocto, it is rather a kernel configuration that you need to adjust for your own needs. This is what @pberginkonsult was pointing out in his reply above.

As I've replied to you [on the OE-Core ML](https://lists.openembedded.org/g/openembedded-core/topic/sgtl5000_audio_codec/83922991?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,83922991), this codec support is included in the kernel since 2011. You need to enable it in your kernel configuration, and then build the kernel with that option enabled.

And to quote the official Yocto Project official front page (https://www.yoctoproject.org/):
```
The Yocto Project.  It's not an embedded Linux Distribution,
It creates a custom one for you.
```

This means that Yocto does not provide an out-of-the-box Linux distribution, it does create one that you define using meta data. I strongly recommend to go through the official Yocto documentation to get an idea on the project and how it can help you to create your own Linux distribution.

-- andrey

angolini, 2021-08-06:
Basically, you have to make sure the defconfig file for the linux recipe being used on the target machine includes the sgtl5* device enabled.

I'm going to close this issue as there was no update in last 30+ days. If case anything else is needed, please feel free to reopen it @cristianod3

closed 2021-08-06 by angolini