Freescale/meta-freescale #827: What is libsdl2's disable-video-vivante for?

whsplnk, 2021-07-15:
Hi,
Sorry not sure if this is the best place/way to ask a question.

We are attemping to bring up Kivy to an imx8mp via Yocto and Kivy requires libsdl2. Unfortunately, we are having many problems doing so. (It is currently segfaulting somewhere on sdl2 -> libEGL->libwayland-client->pthread_mut-x_lock).

Anyways, there's a switch in meta-freescale's libsdl2_%.bbappend that disables video vivante.

I was wondering what was the switch for? I tried enabling it, but I don't see a difference

otavio, 2021-07-15:
Hello @whsplnk, we at O.S. Systems, enabled Kivy support on i.MX6 some time ago; we used Etnaviv for this specific project and it worked fine.

whsplnk, 2021-07-16:
Hi @otavio, I recognize your name on the Kivy discord while I was researching on how to get the Kivy going with Yocto! Was the imx6 running on X11 or wayland?

I got Kivy going on a rockchip device applying a version of the patch you posted (but for some reason it keeps falling back to llvm pipe even with gpu drivers installed). Unfortunately, like I mentioned it does not run on imx8/wayland for somereason.

Would it be possible to share the steps your team has done? (I would understand if you can't). I could also post the recipe that I've been using as well.

otavio, 2021-07-16:
We used Wayland with Linux mainline (Etnaviv driver). I'll check if we did the upstreaming of the work we did.

whsplnk, 2021-07-16:
I was wondering if you had to do something special when compiling sdl2? Aside from configuring it for gles?

whsplnk, 2021-07-16:
Hi otavio,
I managed to get Kivy going on the imx8 finally. I switched over to completely wayland instead of xwayland.

otavio, 2021-07-16:
Oh! Maybe you could upstream it to meta-python? :-)

closed 2021-09-08 by angolini