Freescale/meta-freescale #834: Build process for imx6ull-microgea fails because of pxp

jfernandz, 2021-07-28:
I'm trying to build Dunfell for an `imx6ull-microgea` and I got that output:

```
Initialising tasks: 100% |########################################################################################################| Time: 0:00:06
Sstate summary: Wanted 1259 Found 0 Missed 1259 Current 793 (0% match, 38% complete)
NOTE: Executing Tasks
ERROR: imx-lib-1_5.8+AUTOINC+3f777974c0-r0 do_compile: oe_runmake failed
ERROR: imx-lib-1_5.8+AUTOINC+3f777974c0-r0 do_compile: Execution of '/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/temp/run.do_compile.2184599' failed with exit code 1:
make -C pxp all
make[1]: Entering directory '/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/git/pxp'
arm-poky-linux-gnueabi-gcc  -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a7 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/recipe-sysroot -D -I/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/recipe-sysroot/usr/include/imx -Wall -fPIC  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0=/usr/src/debug/imx-lib/1_5.8+AUTOINC+3f777974c0-r0                      -fdebug-prefix-map=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0=/usr/src/debug/imx-lib/1_5.8+AUTOINC+3f777974c0-r0                      -fdebug-prefix-map=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/recipe-sysroot-native=  -c pxp_lib.c -o pxp_lib.o
<command-line>: error: macro names must be identifiers
In file included from pxp_lib.c:26:
pxp_lib.h:19:10: fatal error: linux/pxp_device.h: No such file or directory
   19 | #include <linux/pxp_device.h>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
Makefile:30: recipe for target 'pxp_lib.o' failed
make[1]: *** [pxp_lib.o] Error 1
make[1]: Leaving directory '/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/git/pxp'
Makefile:17: recipe for target 'pxp' failed
make: *** [pxp] Error 2
WARNING: /home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/temp/run.do_compile.2184599:1 exit 1 from 'exit 1'

ERROR: Logfile of failure stored in: /home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/temp/log.do_compile.2184599
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 1 CROSS_COMPILE=arm-poky-linux-gnueabi- PLATFORM= INCLUDE=-I/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/recipe-sysroot/usr/include/imx all
| make -C pxp all
| make[1]: Entering directory '/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/git/pxp'
| arm-poky-linux-gnueabi-gcc  -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a7 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/recipe-sysroot -D -I/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/recipe-sysroot/usr/include/imx -Wall -fPIC  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0=/usr/src/debug/imx-lib/1_5.8+AUTOINC+3f777974c0-r0                      -fdebug-prefix-map=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0=/usr/src/debug/imx-lib/1_5.8+AUTOINC+3f777974c0-r0                      -fdebug-prefix-map=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/recipe-sysroot-native=  -c pxp_lib.c -o pxp_lib.o
| <command-line>: error: macro names must be identifiers
| In file included from pxp_lib.c:26:
| pxp_lib.h:19:10: fatal error: linux/pxp_device.h: No such file or directory
|    19 | #include <linux/pxp_device.h>
|       |          ^~~~~~~~~~~~~~~~~~~~
| compilation terminated.
| Makefile:30: recipe for target 'pxp_lib.o' failed
| make[1]: *** [pxp_lib.o] Error 1
| make[1]: Leaving directory '/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/git/pxp'
| Makefile:17: recipe for target 'pxp' failed
| make: *** [pxp] Error 2
| ERROR: oe_runmake failed
| WARNING: /home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/temp/run.do_compile.2184599:1 exit 1 from 'exit 1'
| ERROR: Execution of '/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/temp/run.do_compile.2184599' failed with exit code 1:
| make -C pxp all
| make[1]: Entering directory '/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/git/pxp'
| arm-poky-linux-gnueabi-gcc  -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a7 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/recipe-sysroot -D -I/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/recipe-sysroot/usr/include/imx -Wall -fPIC  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0=/usr/src/debug/imx-lib/1_5.8+AUTOINC+3f777974c0-r0                      -fdebug-prefix-map=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0=/usr/src/debug/imx-lib/1_5.8+AUTOINC+3f777974c0-r0                      -fdebug-prefix-map=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/recipe-sysroot-native=  -c pxp_lib.c -o pxp_lib.o
| <command-line>: error: macro names must be identifiers
| In file included from pxp_lib.c:26:
| pxp_lib.h:19:10: fatal error: linux/pxp_device.h: No such file or directory
|    19 | #include <linux/pxp_device.h>
|       |          ^~~~~~~~~~~~~~~~~~~~
| compilation terminated.
| Makefile:30: recipe for target 'pxp_lib.o' failed
| make[1]: *** [pxp_lib.o] Error 1
[... cut ...]
| WARNING: /home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/temp/run.do_compile.2184599:1 exit 1 from 'exit 1'
| 
ERROR: Task (/home/pokyuser/building/joifi-engicam-mx6-dunfell/sources/meta-freescale/recipes-bsp/imx-lib/imx-lib_git.bb:do_compile) failed with exit code '1'
```

Apparently the machine conf file is bringing it in [line 9](https://github.com/engicam-stable/meta-engicam-nxp/blob/dunfell-community-bsp/conf/machine/imx6ull-microgea.conf#L9).

What do you think? Thank you all.

angolini, 2021-07-28:
let me double check if I can reproduce the same for another board

angolini, 2021-07-28:
When I try to build `imx-lib` for `imx6ullevk` I get a "nothing provides"  message:

```
$ MACHINE=imx6ullevk bitbake imx-lib
...
Loaded 3275 entries from dependency cache.
ERROR: Nothing PROVIDES 'imx-lib'
imx-lib was skipped: incompatible with machine imx6ullevk (not in COMPATIBLE_MACHINE)
```

I reviewed your machine conf file and I cannot see anything which would cause an error. I wonder if you are using dunfell on all your meta-layers

jfernandz, 2021-07-29:
I get the very same output when I try to build just `imx-lib`:

```pokyuser@dc0ad4b53062:~/building/joifi-engicam-mx6-dunfell/build$ bitbake imx-lib
Loading cache: 100% |#############################################################################################################| Time: 0:00:04
Loaded 3621 entries from dependency cache.
Parsing recipes: 100% |###########################################################################################################| Time: 0:00:03
Parsing of 2458 .bb files complete (2442 cached, 16 parsed). 3617 targets, 246 skipped, 0 masked, 0 errors.
Removing 1 recipes from the cortexa7t2hf-neon sysroot: 100% |#####################################################################| Time: 0:00:02
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "imx6ull-microgea"
DISTRO               = "joifi"
DISTRO_VERSION       = "3.1.10"
TUNE_FEATURES        = "arm vfp cortexa7 neon thumb callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            = "HEAD:2a848e95074318f3a243df7b3f40513a13173a82"
meta-oe              
meta-multimedia      
meta-python          
                     = "HEAD:85f8047c71047d93cf79a954142157f85079968d"
meta-freescale       = "HEAD:12c3eed6a90b436565126d8ba6934f794b816e25"
meta-freescale-3rdparty = "HEAD:537cb85e4db17d1f6b356a92112b32ae2e39b235"
meta-freescale-distro = "HEAD:5d882cdf079b3bde0bd9869ce3ca3db411acbf3b"
meta-engicam-nxp     = "HEAD:797a1c8f6a12ca8e2c6eb3115f24d66aa7d15dab"
meta-joifi           = "HEAD:929fee1062b311558ef6760eb6c17af59739b9df"
meta-iot-cloud       = "HEAD:5431f91969087d631972d73cd03b16eed0a61987"

Initialising tasks: 100% |########################################################################################################| Time: 0:00:01
Sstate summary: Wanted 7 Found 0 Missed 7 Current 137 (0% match, 95% complete)
NOTE: Executing Tasks
ERROR: imx-lib-1_5.8+AUTOINC+3f777974c0-r0 do_compile: oe_runmake failed
ERROR: imx-lib-1_5.8+AUTOINC+3f777974c0-r0 do_compile: Execution of '/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/temp/run.do_compile.2586035' failed with exit code 1:
make -C pxp all
make[1]: Entering directory '/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/git/pxp'
arm-poky-linux-gnueabi-gcc  -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a7 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/recipe-sysroot -D -I/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/recipe-sysroot/usr/include/imx -Wall -fPIC  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0=/usr/src/debug/imx-lib/1_5.8+AUTOINC+3f777974c0-r0                      -fdebug-prefix-map=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0=/usr/src/debug/imx-lib/1_5.8+AUTOINC+3f777974c0-r0                      -fdebug-prefix-map=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/recipe-sysroot-native=  -c pxp_lib.c -o pxp_lib.o
<command-line>: error: macro names must be identifiers
In file included from pxp_lib.c:26:
pxp_lib.h:19:10: fatal error: linux/pxp_device.h: No such file or directory
   19 | #include <linux/pxp_device.h>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
Makefile:30: recipe for target 'pxp_lib.o' failed
make[1]: *** [pxp_lib.o] Error 1
make[1]: Leaving directory '/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/git/pxp'
Makefile:17: recipe for target 'pxp' failed
make: *** [pxp] Error 2
WARNING: /home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/temp/run.do_compile.2586035:1 exit 1 from 'exit 1'

ERROR: Logfile of failure stored in: /home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/temp/log.do_compile.2586035
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 1 CROSS_COMPILE=arm-poky-linux-gnueabi- PLATFORM= INCLUDE=-I/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/recipe-sysroot/usr/include/imx all
| make -C pxp all
[... cut ...]
| make[1]: Leaving directory '/home/pokyuser/building/joifi-engicam-mx6-dunfell/build/tmp/work/cortexa7t2hf-neon-mx6ul-poky-linux-gnueabi/imx-lib/1_5.8+AUTOINC+3f777974c0-r0/git/pxp'
[... 77 lines cut ...]
    lprng \
    \ 
    minicom \
    mtd-utils \
    mtd-utils-ubifs \
    \ 
    opkg \
    \
    parted \
    psplash \
    \ 
    rng-tools \
    \ 
    serialtools \
    \ 
    tar \
    \ 
    usbutils \
    util-linux \
    "

EXTRA_IMAGE_FEATURES = " ssh-server-openssh hwcodecs "
```

Do you think the problem could be because I'm including `imx-kobs` and `imx-test`? I think I'll try to build the image without them.

jfernandz, 2021-07-29:
Exactly, the problem was I was installing `imx-test` which apparently brings `pxp`, the problem was I was using the same recipe for my image than when I was using warrior, so my recipe inherited `ixm-test` because I followed [this example](https://github.com/engicam-stable/meta-engicam/blob/warrior/recipes-images/images/engicam-test-hw.bb) to crate my recipe. When I remove `imx-test` from my image recipe it builds as expected.

Thank you very mucho @angolini you helped me a lot! :grin:

closed 2021-07-29 by jfernandz