Freescale/meta-freescale #838: imx-alsa-plugins: Build error linux/mxc_asrc.h: No such file or directory

angolini, 2021-08-05:
```
ERROR: Logfile of failure stored in: /media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/temp/log.do_compile.3378664
Log data follows:
| DEBUG: Executing python function autotools_aclocals
| DEBUG: SITE files ['endian-little', 'bit-64', 'arm-common', 'arm-64', 'common-linux', 'common-glibc', 'aarch64-linux', 'common']
| DEBUG: Python function autotools_aclocals finished
| DEBUG: Executing shell function do_compile
| NOTE: make -j 8
| make  all-recursive
| make[1]: Entering directory '/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/build'
| Making all in asrc
| make[2]: Entering directory '/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/build/asrc'
| ../aarch64-fsl-linux-libtool  --tag=CC   --mode=compile aarch64-fsl-linux-gcc  -mcpu=cortex-a53 -march=armv8-a+crc+crypto -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -Wdate-time --sysroot=/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../../git/asrc -I..    -Wall -g   -I/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/recipe-sysroot/usr/include/imx -c -o rate_asrcrate.lo ../../git/asrc/rate_asrcrate.c
| ../aarch64-fsl-linux-libtool  --tag=CC   --mode=compile aarch64-fsl-linux-gcc  -mcpu=cortex-a53 -march=armv8-a+crc+crypto -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -Wdate-time --sysroot=/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../../git/asrc -I..    -Wall -g   -I/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/recipe-sysroot/usr/include/imx -c -o asrc_pair.lo ../../git/asrc/asrc_pair.c
| aarch64-fsl-linux-libtool: compile:  aarch64-fsl-linux-gcc -mcpu=cortex-a53 -march=armv8-a+crc+crypto -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -Wdate-time --sysroot=/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../../git/asrc -I.. -Wall -g -I/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/recipe-sysroot/usr/include/imx -c ../../git/asrc/asrc_pair.c  -fPIC -DPIC -o .libs/asrc_pair.o
| aarch64-fsl-linux-libtool: compile:  aarch64-fsl-linux-gcc -mcpu=cortex-a53 -march=armv8-a+crc+crypto -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -Wdate-time --sysroot=/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../../git/asrc -I.. -Wall -g -I/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/recipe-sysroot/usr/include/imx -c ../../git/asrc/rate_asrcrate.c  -fPIC -DPIC -o .libs/rate_asrcrate.o
| In file included from ../../git/asrc/rate_asrcrate.c:24:
| ../../git/asrc/asrc_pair.h:28:10: fatal error: linux/mxc_asrc.h: No such file or directory
|    28 | #include <linux/mxc_asrc.h>
|       |          ^~~~~~~~~~~~~~~~~~
| compilation terminated.
| make[2]: *** [Makefile:444: rate_asrcrate.lo] Error 1
| make[2]: *** Waiting for unfinished jobs....
| ../../git/asrc/asrc_pair.c:29:10: fatal error: linux/mxc_asrc.h: No such file or directory
|    29 | #include <linux/mxc_asrc.h>
|       |          ^~~~~~~~~~~~~~~~~~
| compilation terminated.
| make[2]: *** [Makefile:444: asrc_pair.lo] Error 1
| make[2]: Leaving directory '/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/build/asrc'
| make[1]: *** [Makefile:404: all-recursive] Error 1
| make[1]: Leaving directory '/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/build'
| make: *** [Makefile:336: all] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/temp/run.do_compile.3378664' failed with exit code 1:
| make  all-recursive
| make[1]: Entering directory '/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/build'
| Making all in asrc
| make[2]: Entering directory '/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/build/asrc'
| ../aarch64-fsl-linux-libtool  --tag=CC   --mode=compile aarch64-fsl-linux-gcc  -mcpu=cortex-a53 -march=armv8-a+crc+crypto -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -Wdate-time --sysroot=/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../../git/asrc -I..    -Wall -g   -I/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/recipe-sysroot/usr/include/imx -c -o rate_asrcrate.lo ../../git/asrc/rate_asrcrate.c
| ../aarch64-fsl-linux-libtool  --tag=CC   --mode=compile aarch64-fsl-linux-gcc  -mcpu=cortex-a53 -march=armv8-a+crc+crypto -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -Wdate-time --sysroot=/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../../git/asrc -I..    -Wall -g   -I/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/recipe-sysroot/usr/include/imx -c -o asrc_pair.lo ../../git/asrc/asrc_pair.c
| aarch64-fsl-linux-libtool: compile:  aarch64-fsl-linux-gcc -mcpu=cortex-a53 -march=armv8-a+crc+crypto -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -Wdate-time --sysroot=/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../../git/asrc -I.. -Wall -g -I/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/recipe-sysroot/usr/include/imx -c ../../git/asrc/asrc_pair.c  -fPIC -DPIC -o .libs/asrc_pair.o
| aarch64-fsl-linux-libtool: compile:  aarch64-fsl-linux-gcc -mcpu=cortex-a53 -march=armv8-a+crc+crypto -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -Wdate-time --sysroot=/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/recipe-sysroot -DHAVE_CONFIG_H -I. -I../../git/asrc -I.. -Wall -g -I/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/recipe-sysroot/usr/include/imx -c ../../git/asrc/rate_asrcrate.c  -fPIC -DPIC -o .libs/rate_asrcrate.o
| In file included from ../../git/asrc/rate_asrcrate.c:24:
| ../../git/asrc/asrc_pair.h:28:10: fatal error: linux/mxc_asrc.h: No such file or directory
|    28 | #include <linux/mxc_asrc.h>
|       |          ^~~~~~~~~~~~~~~~~~
| compilation terminated.
| make[2]: *** [Makefile:444: rate_asrcrate.lo] Error 1
| make[2]: *** Waiting for unfinished jobs....
| ../../git/asrc/asrc_pair.c:29:10: fatal error: linux/mxc_asrc.h: No such file or directory
|    29 | #include <linux/mxc_asrc.h>
|       |          ^~~~~~~~~~~~~~~~~~
| compilation terminated.
| make[2]: *** [Makefile:444: asrc_pair.lo] Error 1
| make[2]: Leaving directory '/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/build/asrc'
| make[1]: *** [Makefile:404: all-recursive] Error 1
| make[1]: Leaving directory '/media/daiane/TRIFORCE/yocto/master/build/tmp/work/cortexa53-crypto-fsl-linux/imx-alsa-plugins/1.0.26-r0/build'
| make: *** [Makefile:336: all] Error 2
| WARNING: exit code 1 from a shell command.
```

angolini, 2021-08-05:
@otavio  I haven't yet looked deeply, but reporting as it might be some package missing update

otavio, 2021-08-05:
@thochstein it might be a missing alsa-lib or kernel headers bump. Can you take a look?

thochstein, 2021-08-06:
I haven't been able to work on updating multimedia and graphics for our Q2 release. I can take a look now, but the layer is incompatible now? Will you merge Khem's change soon?

`ERROR: Layer freescale-layer is not compatible with the core layer which only supports these series: honister (layer is compatible with gatesgarth hardknott)`

otavio, 2021-08-06:
Yes, I can merge it tomorrow.

angolini, 2021-08-09:
I confirm it's fixed with #839

closed 2021-08-09 by angolini