Freescale/meta-freescale #854: Layer freescale-layer is not compatible with the core layer which only supports these series: hardknott (layer is compatible with honister)

jasonlic, 2021-08-28:
bitbake-layers add-layer ../meta-freescale/ output errors:
NOTE: Starting bitbake server...
ERROR: Layer freescale-layer is not compatible with the core layer which only supports these series: hardknott (layer is compatible with honister)
ERROR: Parse failure with the specified layer added, aborting.

otavio, 2021-08-28:
You need to sue the hardknott branch for it.

zandrey, 2021-08-30:
@jasonlic Indeed, `master` branch of the layer only supports _honister_ starting from commit 13fc081bf70a9438be3a09c91ee38af9798af207. This is done in order to accommodate new append mechanism introduced in OE-Core.

Perhaps you have your core layer setup checked out on `hardknott`, while trying to use `master` from this layer?

zandrey, 2021-09-23:
@jasonlic Ping. Any chance you verified your setup? Is this issue still valid?

jasonlic, 2021-09-24:
Yes, it still has error.

从 Windows 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>发送

zandrey, 2021-09-24:
> 
> Yes, it still has error.

Can you post your layer setup here? Which layers do you include? Which branches?

From you original post, it is not clear to which layer setup you're adding `meta-freescale` to, and which branches those layers are checked out on.

If you have core layers checked out at _master_, but adding _hardknott_ from `meta-freescale` - then you for sure have the error you've posted originally. I've already pointed out above, that starting from commit 13fc081bf70a9438be3a09c91ee38af9798af207 in `meta-freescale` layer compatibility has been changed since the new override index is introduced upstream.

You have 2 options to go ahead:
1. If you intend to build _hardknott_ - make sure **_all_** your layers are checked out at that branch.
2. If you're aiming for _master_ branch - make sure that `meta-freescale` is checked out at the tip of _master_, or at least past the 13fc081bf70a9438be3a09c91ee38af9798af207

jasonlic, 2021-09-27:
It can workl now. I use branch honister of poky.
Here is my steps:

git clone -b honister git://git.yoctoproject.org/poky.git
cd poky
git clone https://github.com/Freescale/meta-freescale.git
bitbake-layers add-layer ../meta-freescale/

thanks

从 Windows 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>发送

If you have core layers checked out at master, but adding hardknott from meta-freescale - then you for sure have the error you've posted originally. I've already pointed out above, that starting from commit 13fc081<https://github.com/Freescale/meta-freescale/commit/13fc081bf70a9438be3a09c91ee38af9798af207> in meta-freescale layer compatibility has been changed since the new override index is introduced upstream.

otavio, 2021-09-27:
I'll close this as it is not an issue. In case of doubts please ask for reopening it.

closed 2021-09-27 by otavio