Freescale/meta-freescale #864: Building master branch for imx53 -> kernel panic

woutervanh, 2021-09-14:
Building linux 5.14.0+g682d702b426b (from kernel recipe in master branch).  Only added own devicetree (which was working fine on 5.10). I don't really have a clue what's going on...
And now I'm popping on to a kernel panic:

> [    2.323612] 63f90000.serial: ttymxc4 at MMIO 0x63f90000 (irq = 102, base_baud = 4166666) is a IMX
[    2.576862] random: fast init done
[    3.149450] printk: console [ttymxc4] enabled
[    3.171713] [drm] using 16m VRAM carveout
[    3.237037] msm msm: [drm:msm_drm_bind] VRAM: 88100000->89100000
[    3.243334] adreno 30000000.gpu: Could not find a clock rate. Using a reasonable default
[    3.253874] adreno 30000000.gpu: supply vdd not found, using dummy regulator
[    3.262871] adreno 30000000.gpu: supply vddcx not found, using dummy regulator
[    3.275493] msm msm: bound 30000000.gpu (ops a3xx_ops)
[    3.287174] [drm] Initialized msm 1.8.0 20130625 for msm on minor 0
[    3.293915] 8<--- cut here ---
[    3.297012] Unable to handle kernel NULL pointer dereference at virtual address 00000028
[    3.305244] pgd = (ptrval)
[    3.307989] [00000028] *pgd=00000000
[    3.311624] Internal error: Oops: 805 [#1] SMP ARM
[    3.316430] Modules linked in:
[    3.319503] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 5.14.0+g682d702b426b #1
[    3.326652] Hardware name: Freescale i.MX53 (Device Tree Support)
[    3.332754] PC is at __mutex_init+0x14/0x54
[    3.336969] LR is at msm_disp_snapshot_init+0x24/0xa0
[    3.342039] pc : [<c0181024>]    lr : [<c0775168>]    psr: a0000053
[    3.348314] sp : c20cbd10  ip : bf30efad  fp : d7ebb000
[    3.353547] r10: 00000000  r9 : c2414c00  r8 : c20ca000
[    3.358780] r7 : 00000000  r6 : c1e69078  r5 : c12d31e8  r4 : 00000028
[    3.365316] r3 : c2414c00  r2 : c1e69078  r1 : c12d31e8  r0 : 00000028
[    3.371854] Flags: NzCv  IRQs on  FIQs off  Mode SVC_32  ISA ARM  Segment none
[    3.379089] Control: 10c5387d  Table: 70004019  DAC: 00000051
[    3.384842] Register r0 information: non-paged memory
[    3.389911] Register r1 information: non-slab/vmalloc memory
[    3.395588] Register r2 information: non-slab/vmalloc memory
[    3.401262] Register r3 information: slab kmalloc-1k start c2414c00 pointer offset 0 size 1024
[    3.409919] Register r4 information: non-paged memory
[    3.414985] Register r5 information: non-slab/vmalloc memory
[    3.420657] Register r6 information: non-slab/vmalloc memory
[    3.426330] Register r7 information: NULL pointer
[    3.431048] Register r8 information: non-slab/vmalloc memory
[    3.436722] Register r9 information: slab kmalloc-1k start c2414c00 pointer offset 0 size 1024
[    3.445373] Register r10 information: NULL pointer
[    3.450179] Register r11 information: non-slab/vmalloc memory
[    3.455939] Register r12 information: non-paged memory
[    3.461091] Process swapper/0 (pid: 1, stack limit = 0x(ptrval))
[    3.467109] Stack: (0xc20cbd10 to 0xc20cc000)
[    3.471479] bd00:                                     00000000 c21eb800 00000000 c2414010
[    3.479671] bd20: c20ca000 c0775168 00000000 c077898c 89100000 00000000 c0778664 c24142d4
[    3.487864] bd40: 60000053 00000000 c24142d4 60000053 c24142d4 c0e6e7b0 c240e700 c24142f8
[    3.496055] bd60: 60000053 c07dac8c 00008124 92a80910 00008124 c240e540 c240e680 c240e6c0
[    3.504246] bd80: c17692d0 00000018 00000001 c244a040 c176927c c07cdb80 c176927c 00000000
[    3.512437] bda0: c0f84f64 c240e6c0 c176927c c2414010 c0f84f64 c20ca000 00000000 c1800000
[    3.520629] bdc0: c15003e0 c07cdfe4 c0f8fdd4 c2414010 ffffffff d7babe60 c2414010 c2414010
[    3.528820] bde0: c2414010 c0778e3c 00000000 c2449bc8 c2449af8 c244a040 00000000 c03812e0
[    3.537010] be00: 00000000 92a80910 c2449af8 00000000 c2414010 c1766d48 c2414010 00000000
[    3.545201] be20: 00000000 c1800000 c15003e0 c07d852c c2414010 00000000 c1766d48 c2414010
[    3.553392] be40: 00000000 c07d5c08 c2414010 00000000 c1766d48 c2414010 00000000 c07d5f40
[    3.561583] be60: c1e6b3c4 c1e6b3c8 c24140a0 c07d6008 c2414010 c1766d48 c24140a0 c20ca000
[    3.569775] be80: c16093cc 00000000 c1800000 c07d66cc 00000000 c1766d48 c07d6618 c07d3c50
[    3.577966] bea0: c2437300 c20ae6b0 c240a414 92a80910 c20ae6e4 c1766d48 c2437300 00000000
[    3.586157] bec0: c1769db8 c07d4f64 c1303428 c05f93f4 c1766d48 c1766d48 00000000 c2004300
[    3.594349] bee0: c17f4fe0 c07d7474 c20ca000 c154371c c2004300 c01022e8 000005c8 00000000
[    3.602540] bf00: c2004376 c2004300 c2004375 c014ee30 c1306a40 00000128 00000000 00000000
[    3.610731] bf20: c15003e0 ffffffff 00000006 c17f4fe0 c16093cc ffffe000 c155c870 c155c854
[    3.618922] bf40: c13d00a8 92a80910 c1573b90 00000007 c2004300 c155c874 c155c854 c13d00a8
[    3.627113] bf60: c1800000 c15011ec 00000006 00000006 00000000 c15003e0 c16097cc 00000128
[    3.635303] bf80: c155c808 00000000 c0e6414c 00000000 00000000 00000000 00000000 00000000
[    3.643494] bfa0: 00000000 c0e6415c 00000000 c010013c 00000000 00000000 00000000 00000000
[    3.651686] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    3.659876] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[    3.668065] [<c0181024>] (__mutex_init) from [<c0775168>] (msm_disp_snapshot_init+0x24/0xa0)
[    3.676530] [<c0775168>] (msm_disp_snapshot_init) from [<c077898c>] (msm_drm_bind+0x3a0/0x610)
[    3.685169] [<c077898c>] (msm_drm_bind) from [<c07cdb80>] (try_to_bring_up_master+0x15c/0x1ac)
[    3.693804] [<c07cdb80>] (try_to_bring_up_master) from [<c07cdfe4>] (component_master_add_with_match+0xb4/0xf4)
[    3.703912] [<c07cdfe4>] (component_master_add_with_match) from [<c0778e3c>] (msm_pdev_probe+0x240/0x294)
[    3.713499] [<c0778e3c>] (msm_pdev_probe) from [<c07d852c>] (platform_probe+0x58/0xbc)
[    3.721443] [<c07d852c>] (platform_probe) from [<c07d5c08>] (really_probe.part.0+0x9c/0x334)
[    3.729914] [<c07d5c08>] (really_probe.part.0) from [<c07d5f40>] (__driver_probe_device+0xa0/0x138)
[    3.738983] [<c07d5f40>] (__driver_probe_device) from [<c07d6008>] (driver_probe_device+0x30/0x10c)
[    3.748055] [<c07d6008>] (driver_probe_device) from [<c07d66cc>] (__driver_attach+0xb4/0x174)
[    3.756603] [<c07d66cc>] (__driver_attach) from [<c07d3c50>] (bus_for_each_dev+0x78/0xb8)
[    3.764805] [<c07d3c50>] (bus_for_each_dev) from [<c07d4f64>] (bus_add_driver+0xf0/0x1d8)
[    3.773005] [<c07d4f64>] (bus_add_driver) from [<c07d7474>] (driver_register+0x88/0x118)
[    3.781118] [<c07d7474>] (driver_register) from [<c01022e8>] (do_one_initcall+0x80/0x3a4)
[    3.789320] [<c01022e8>] (do_one_initcall) from [<c15011ec>] (kernel_init_freeable+0x198/0x228)
[    3.798052] [<c15011ec>] (kernel_init_freeable) from [<c0e6415c>] (kernel_init+0x10/0x128)
[    3.806340] [<c0e6415c>] (kernel_init) from [<c010013c>] (ret_from_fork+0x14/0x38)
[    3.813928] Exception stack(0xc20cbfb0 to 0xc20cbff8)
[    3.818991] bfa0:                                     00000000 00000000 00000000 00000000
[    3.827181] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    3.835370] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[    3.842000] Code: e3a07000 e1a04000 e1a05001 e1a06002 (e4807004)
[    3.848241] ---[ end trace 07bcb2c7524c4e46 ]---
[    3.853061] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[    3.860753] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b ]---

otavio, 2021-09-14:
@twoerner can you verify if you can reproduce it?

twoerner, 2021-09-14:
yes, i'm seeing the exact same kernel panic with a core-image-base build from master:

```
Build Configuration:
BB_VERSION           = "1.51.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "opensuseleap-15.2"
TARGET_SYS           = "arm-oe-linux-gnueabi"
MACHINE              = "imx53qsb"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "arm vfp cortexa8 neon thumb callconvention-hard"
TARGET_FPU           = "hard"
meta-freescale       = "master:6cb49377476a39c4de9c223c07e1ad57df06427f"
meta                 = "master:50154869b496cf63a4fae2c8278350fcf9a2591d"
```

my version is:
```
[    0.000000] Linux version 5.14.0+g35f3f4421d3d (oe-user@oe-host) (arm-oe-linux-gnueabi-gcc (GCC) 11.2.0, GNU ld (GNU Binutils) 2.37.20210721) #1 SMP Tue Sep 14 14:40:59 UTC 2021
```

```
$ cd tmp-glibc/work/imx53qsb-oe-linux-gnueabi/linux-fslc/5.14+gitAUTOINC+35f3f4421d-r0/git
$ git remote -v
origin  git://github.com/Freescale/linux-fslc.git (fetch)
origin  git://github.com/Freescale/linux-fslc.git (push)
$ git log --oneline -1
35f3f4421d3d (HEAD -> 5.14.x+fslc, origin/5.14.x+fslc) ARM64: defconfig: Enable KPROBES so allow LTTng use
```

otavio, 2021-09-14:
Did you check if it is due to a mismatch on the root partition? @fabioestevam ideas?

woutervanh, 2021-09-14:
@otavio , it has nothing to do with the root partition. I'm booting from nfs and even disabled my nfs server to verify. It's purely something in kernel

otavio, 2021-09-14:
Have you tried 5.13? might be good to find a good kernel version to try to find out the culprit.

fabioestevam, 2021-09-14:
I have just sent the attached patch upstream.

Please test it.

fabioestevam, 2021-09-14:
Looks like replying via email does not allow attachment.

Here is the fix I sent for this issue:
https://lists.freedesktop.org/archives/dri-devel/2021-September/323297.html

twoerner, 2021-09-14:
@fabioestevam brilliant!
works for me :+1: 

```
…
[    2.993534] [drm] using 16m VRAM carveout
[    3.054585] msm msm: [drm:msm_drm_bind] VRAM: cc100000->cd100000
[    3.060852] adreno 30000000.gpu: Could not find a clock rate. Using a reasonable default
[    3.071139] adreno 30000000.gpu: supply vdd not found, using dummy regulator
[    3.079715] adreno 30000000.gpu: supply vddcx not found, using dummy regulator
[    3.091568] msm msm: bound 30000000.gpu (ops a3xx_ops)
[    3.102191] [drm] Initialized msm 1.8.0 20130625 for msm on minor 0
[    3.131616] imx-drm display-subsystem: bound imx-ipuv3-crtc.2 (ops ipu_crtc_ops)
[    3.139936] imx-drm display-subsystem: bound imx-ipuv3-crtc.3 (ops ipu_crtc_ops)
[    3.148037] imx-drm display-subsystem: bound 63ff0000.tve (ops imx_tve_ops)
[    3.159181] [drm] Initialized imx-drm 1.0.0 20120507 for display-subsystem on minor 1
[    3.167189] imx-ipuv3 18000000.ipu: IPUv3M probed
…

OpenEmbedded nodistro.0 imx53qsb /dev/ttymxc0

imx53qsb login: root
root@imx53qsb:~# uname -a
Linux imx53qsb 5.14.0+gd400ec6e72b3 #1 SMP Tue Sep 14 19:48:25 UTC 2021 armv7l GNU/Linux
root@imx53qsb:~#
```

twoerner, 2021-09-14:
I guess we'll need to carry this patch locally until it makes its way through the various projects?
@fabioestevam are you preparing a patch for meta-freescale?

fabioestevam, 2021-09-14:
Hi Trevor,

On Tue, Sep 14, 2021 at 5:03 PM Trevor Woerner ***@***.***> wrote:
>
> I guess we'll need to carry this patch locally until it makes its way through the various projects?
> @fabioestevam are you preparing a patch for meta-freescale?

Thanks for testing.

Could you please help to prepare the patch for meta-freescale?

twoerner, 2021-09-14:
> Thanks for testing. Could you please help to prepare the patch for meta-freescale?

No problem!

I have no idea if @otavio will like my approach, but here's a first draft: https://github.com/Freescale/meta-freescale/pull/865

fabioestevam, 2021-09-14:
Hi Trevor,

On Tue, Sep 14, 2021 at 5:42 PM Trevor Woerner ***@***.***> wrote:
>
> Thanks for testing. Could you please help to prepare the patch for meta-freescale?
>
> No problem!
>
> I have no idea if @otavio will like my approach, but here's a first draft: #865

Thanks! I have pushed the fix now:
https://github.com/Freescale/linux-fslc/commit/4b88a1ee28e7be1fce90e389d786575408924d65

otavio, 2021-09-14:
No way 👎🏼 We carry forks for a reason ... heheh. As @fabioestevam mentioned, it just need the bump now.

twoerner, 2021-09-14:
ah! update on its way…

closed 2021-09-15 by angolini

zandrey, 2021-09-16:
Hello @woutervanh @twoerner @fabioestevam !

Can you maybe tell me which commit caused this in upstream? I'm now doing the stable update, and this got me thinking on if the commit from upstream went into stable trees (`linux-5.10.y` and `linux-5.4.y`) - they would also be broken... 😞 

Any chance anyone could bisect the `linux-5.14.y` to get an offending commit, just to make sure it went only into `master` and `linux-5.14.y` for now, or is already taken into stable?

I can't do it myself - I do not poses the needed _i.MX53_ derivative to test it, so I would have to rely on any hints from you here.

-- andrey

fabioestevam, 2021-09-17:
Hi Andrey,

On Thu, Sep 16, 2021 at 6:48 PM Andrey Zhizhikin
***@***.***> wrote:
>
> Hello @woutervanh @twoerner @fabioestevam !
>
> Can you maybe tell me which commit caused this in upstream? I'm now doing the stable update, and this got me thinking on if the commit from upstream went into stable trees (linux-5.10.y and linux-5.4.y) - they would also be broken... 😞

The commit that broke i.MX53 is
https://github.com/Freescale/linux-fslc/commit/98659487b845c05b6bed85d881713545db674c7c

It only affects 5.14, not 5.10.

I have sent a fix to meta-freescale and also submitted the fix upstream:
https://github.com/Freescale/linux-fslc/commit/4b88a1ee28e7be1fce90e389d786575408924d65

Regards,

Fabio Estevam

zandrey, 2021-09-17:
Hello Fabio,

> Hi Andrey,
>
> On Thu, Sep 16, 2021 at 6:48 PM Andrey Zhizhikin
> >
> > Hello @woutervanh @twoerner @fabioestevam !
> >
> > Can you maybe tell me which commit caused this in upstream? I'm now
> doing the stable update, and this got me thinking on if the commit from
> upstream went into stable trees (linux-5.10.y and linux-5.4.y) - they would
> also be broken... 😞
>
> The commit that broke i.MX53 is
>
> https://github.com/Freescale/linux-fslc/commit/98659487b845c05b6bed85d881713545db674c7c
>
> It only affects 5.14, not 5.10.

Thanks for the quick confirmation! Now I see that the offending commit is
not in stable branches, and I can proceed with stable kernel updates in
recipes so this crash would not appear after it.

>
> I have sent a fix to meta-freescale and also submitted the fix upstream:
>
> https://github.com/Freescale/linux-fslc/commit/4b88a1ee28e7be1fce90e389d786575408924d65

Great catch! I'd note this commit for myself for further updates of
linux-5.14.y

>
>
> Regards,
>
> Fabio Estevam