Freescale/meta-freescale #878: imx-cst: do_packagedata failing

kraj, 2021-10-01:
seems this is the failure
```
fatal: unable to access 'https://gitlab.apertis.org/pkg/imx-code-signing-tool.git/': server certificate verification failed. CAfile: none CRLfile: none
```
see

https://errors.yoctoproject.org/Errors/Details/609986/

otavio, 2021-10-01:
@tprrt, can you take a look?

tprrt, 2021-10-11:
Hi,

Have you seen any new occurrences?

I tried to reproduce this issue without success.
Moreover, the DST Root CA X3, has expired Sept. 30, 2021, causing quite a bit of trouble, which could explain this.

Best regards,
Thomas

kraj, 2021-10-11:
yes cert perhaps was the reason. We should still apply this patch since it pins to right branch

see
https://github.com/Freescale/meta-freescale/pull/881

angolini, 2022-02-01:
Closing as I think the issue is solved now

closed 2022-02-01 by angolini