Freescale/meta-freescale #922: cortex-m support?

RobertBerger, 2021-11-05:
The boards supported here have Cortex-A cores which run Linux. But some also have Cortex-M cores. 

- Do you plan to support e.g. FreeRTOS and/or Zephyr on those Cortex-M cores as well? 
- How are they currently supported if want to use only poky and this layer?

MaxKrummenacher, 2021-11-05:
Hi Robert

While I don't know the answers to your questions some related comments:

Alejandro Hernandez gave an excellent talk on the ELCE 2019 on the subject 'One Build to Rule Them All: Building FreeRTOS & Linux Using Yocto' along with a layer [2] which implements the FreeRTOS side in this multiarch setup.

However, IMHO, building a (Cross) FreeRTOS application outside of a buildsystem like OE is quite easy. Download a precompiled toolchain, download FreeRTOS, add your application source code, run make and you have your binary blob vor the Cortex-M core.
My guess is that doing just that and have a OE recipe which just installs the binary blob into the resulting image is likely a less complicated way to go.

Max

[1] https://www.youtube.com/watch?v=mFgiIXv7b5U
[2] https://github.com/aehs29/meta-freertos

otavio, 2021-11-05:
I tend to agree with @MaxKrummenacher here. I'd build it externally and package the binary blob.

RobertBerger, 2021-11-05:
Thank you @MaxKrummenacher and @otavio. 
Ideally, I would like to distinguish between a "developer workflow" where it's fine to build it externally and a "release workflow" where it would be nice to build it with multiconfig. I am not sure if it's worth bothering to build a custom toolchain for the cortex-m, since most people will most likely use the one which comes with the IDE and since this could be a tricky thing to get it right.

nsdrude, 2022-01-17:
@RobertBerger You may find our recipe freertos-variscite helpful: https://github.com/varigit/meta-variscite-fslc/tree/hardknott/recipes-bsp/freertos-variscite

It fetches the prebuilt toolchain and builds the MCUXpresso example projects for Variscite's i.MX8 family. The way it's currently written, it is specific to Variscite's machines. But it could be adapted to be more generic and configurable.

This enables a "release workflow" where the Cortex-M firmware is revision controlled and built by Yocto.

closed 2022-06-03 by angolini