Freescale/meta-freescale #929: devregs license appears to be wrong

dargo600, 2021-11-19:
Looking at:
https://github.com/boundarydevices/devregs

There are both GPL-1 and GPL2 licenses.  Though if you look at the right side it's clearly marked as GPL-2 not GPL-1 so I think that's the license that should be used.  I'm not too clear as to why this is marked as GPL1 here.  It possibly could be a combo of both licenses as they may not be incompatible...

otavio, 2021-11-19:
https://github.com/boundarydevices/devregs/issues/7

@gibsson can you ask for Troy to look at this?

gibsson, 2021-11-24:
I fixed the project, it's now clearly GPL-2.0 as you can see in both the source file `devregs.cpp` and the `COPYING` file.

otavio, 2021-11-24:
@gibsson can you send a PR bumping its revision, fixing it, and closing this?

gibsson, 2021-11-24:
Here: https://github.com/Freescale/meta-freescale/pull/931

otavio, 2021-11-24:
Closed by #931.

closed 2021-11-24 by otavio