Freescale/meta-freescale #938: imx-atf build fails with ld-is-gold

sapiippo, 2022-01-13:
After changes in https://github.com/Freescale/meta-freescale/pull/933 for imx-atf, its build fails when ld-is-gold is in DISTRO_FEATURES

Built with master branch from poky and meta-freescale using imx8qm-mek:
```
|   LD      /home/qt/work/build/build/tmp/work/cortexa72-cortexa53-crypto-mx8-poky-linux/imx-atf/2.4+gitAUTOINC+bb4957067d-r0/git/build/imx8qm/release/bl31/bl31.elf
| aarch64-poky-linux-ld: warning: creating a segment to contain the file and program headers outside of any MEMORY region
| aarch64-poky-linux-ld: error: load segment overlap [0x80000000 -> 0x8000b8c0] and [0x8000a0c8 -> 0x8000c588]
| aarch64-poky-linux-ld: error: load segment overlap [0x8000a0c8 -> 0x8000c588] and [0x8000b8c0 -> 0x8000f560]
| aarch64-poky-linux-ld: error: invalid reference to dot symbol outside of SECTIONS clause
| aarch64-poky-linux-ld: error: invalid reference to dot symbol outside of SECTIONS clause
| aarch64-poky-linux-ld: error: invalid reference to dot symbol outside of SECTIONS clause
| make: *** [Makefile:1105: /home/qt/work/build/build/tmp/work/cortexa72-cortexa53-crypto-mx8-poky-linux/imx-atf/2.4+gitAUTOINC+bb4957067d-r0/git/build/imx8qm/release/bl31/bl31.elf] Error 1
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
NOTE: recipe imx-atf-2.4+gitAUTOINC+bb4957067d-r0: task do_compile: Failed
ERROR: Task (/home/qt/work/build/sources/meta-freescale/recipes-bsp/imx-atf/imx-atf_2.4.bb:do_compile) failed with exit code '1'
```

otavio, 2022-01-13:
@sapiippo are you going to look at the issue?

denniedegroot, 2022-05-03:
I am running into the same issue with ld-is-gold enabled. Any progress on this?

angolini, 2022-06-03:
What is the status of this issue using current master, @sapiippo ?

otavio, 2024-08-24:
I couldn't reproduce that for a long time, so I'm closing this issue.

closed 2024-08-24 by otavio