Freescale/meta-freescale #941: uuu: COMPATIBLE_MACHINE issue with nativesdk build

IniterWorker, 2022-01-13:
From https://github.com/Freescale/meta-freescale/pull/892,
 
It seems that nativesdk build don't share MACHINEOVERRIDES. And, with the `COMPATIBLE_MACHINE = "(imx)"`, we are not able to build in nativesdk (host).

Should we revert the change/commit? Or alternative?

B. r

otavio, 2022-01-14:
@kraj how do you suggest we fix this?

kraj, 2022-01-14:
hmm try `COMPATIBLE_MACHINE:class-nativesdk = "(.*)"`

IniterWorker, 2022-01-14:
I didn't think about it earlier. Thank you, @kraj, it works fine. Do you want to propose the change?

kraj, 2022-01-15:
> I didn't think about it earlier. Thank you, @kraj, it works fine. Do you want to propose the change?

Go for it

otavio, 2022-01-17:
@IniterWorker can you prepare a PR?

IniterWorker, 2022-01-18:
@otavio and @kraj,

> A regular expression that resolves to one or more target machines with which a recipe is compatible.

As a consequence, we are limiting the recipe integration between different machines.

The current recipe doesn't use `COMPATIBLE_MACHINE` for its main purpose of limiting machines, we should consider removing it.

Use case: you may want to use uuu from another stack like meta-ti, ...

If that makes sense to you, I will prepare PRs to remove the `COMPATIBLE_MACHINE`.

kraj, 2022-01-18:
No need to remove compatible machine it’s added for a reason because it does not work universally for all machines it’s fine to widen the check to include more machines where it is known to work

IniterWorker, 2022-01-18:
Build target and "end-user" available target are two different things.
- You may want to use a beagle bone to flash your imx target with uuu.

So, from my point of you, if libusb/libuuu is not limited to a specific machine, you should be able to build from any machine.

As I said before, it seems to me, we are using `COMPATIBLE_MACHINE` in the wrong way. We don't require a runtime dependency on the imx machine. But, we use it as a classifier for "documentation" purposes.

kraj, 2022-01-18:
> Build target and "end-user" available target are two different things.
> 
> * You may want to use a beagle bone to flash your imx target with uuu.

ok in theory its fine. 
meta-ti would not prefer a dependency on meta-freescale both of them are BSP layers.

> 
> So, from my point of you, if libusb/libuuu is not limited to a specific machine, you should be able to build from any machine.

OK.

> 
> As I said before, it seems to me, we are using `COMPATIBLE_MACHINE` in the wrong way. We don't require a compile/runtime dependency on the imx machine. But, we are using it as a classifier for "documentation" purposes.

if it is a generic tool, move it to a generic layer, it does not belong to a BSP layer.

IniterWorker, 2022-01-19:
> meta-ti would not prefer a dependency on meta-freescale both of them are BSP layers.

Sure, as an example, it was a bit odd and unclear to use. My bad ;-)

> if it is a generic tool, move it to a generic layer, it does not belong to a BSP layer.

Do we have a generic layer for meta-freescale/imx **tools**?

kraj, 2022-01-20:
perhaps `meta-oe`

IniterWorker, 2022-01-25:
@otavio are you okay with moving uuu recipe to the meta-oe layer?

otavio, 2022-01-25:
Sure; no worries from my side.

angolini, 2022-06-03:
@IniterWorker , are you planning to work on this? Or can I close the issue?

IniterWorker, 2022-06-08:
Thank you @angolini, I close it. Because the next step should be a different task with moving uuu-tools to the meta-oe layer.

closed 2022-06-08 by IniterWorker