Freescale/meta-freescale #951: fitImage: fsl-image-mfgtool-initramfs doesn't allow ramfs

IniterWorker, 2022-01-19:
Hi,

I am trying to generate a fitImage from `fsl-image-mfgtool-initramfs`. After a `loadaddr` issue already reported in meta-freescale issue, I am struggling with the initramfs.

As the `its` file wasn't reporting the ramdisk, I tried to put some configurations.

### INITRAMFS Configurations

From **local.conf**

```
KERNEL_CLASSES  = " kernel-fitimage"
KERNEL_IMAGETYPES = "fitImage zImage"
IMAGE_FSTYPES += "tar.bz2 cpio.gz cpio.gz.u-boot"
INITRAMFS_IMAGE = "fsl-image-mfgtool-initramfs"
INITRAMFS_IMAGE_BUNDLE = "1"
```
meta:  **honister**
meta-freescale: **honister**

### Behaviour

```
ERROR: Task (/workdir/meta-custom/recipes-kernel/linux/linux-fslc-lts-mfgtool_5.10.bb:do_bundle_initramfs) failed with exit code '1'
ERROR: linux-fslc-lts-5.10.91+gitAUTOINC+aa8525ec05-r0 do_bundle_initramfs: Could not find any /workdir/build/tmp/deploy/images/imx6ull-custom/fsl-image-mfgtool-initramfs-imx6ull-telflex-radio-boardv2.cpio{.gz|.lz4|.lzo|.lzma|.xz|.zst) for bundling; INITRAMFS_IMAGE_NAME might be wrong.
```

The current behaviour is that the file **cpio.gz** is not present in the deploy directory and can't be added to the fitImage.

Did I miss something?

### Maybe related issue

I have seen a very unlucky bug like that before : 
- https://www.yoctoproject.org/pipermail/yocto/2017-August/037454.html
- https://bugzilla.yoctoproject.org/show_bug.cgi?id=11928

IniterWorker, 2022-01-25:
After few experimentations, I have a fitImage with all in it: dtbs + zImage + initramfs.

But, I can't achieve this without removing these lines from **mfgtool-initramfs-image.bbclass**.

```
IMAGE_FSTYPES = "cpio.gz.u-boot"
IMAGE_FSTYPES:mxs = "cpio.gz.u-boot"
```

I am a bit confused with the purpose of forcing IMAGE_FSTYPES during the build of `fsl-image-mfgtool-initramfs.bb` from `mfgtool-initramfs-image.bbclass`. @otavio and @kraj could you give me a hint about it?

Thanks,

otavio, 2022-01-25:
What machine and `IMX_DEFAULT_BSP` values are you using?

IniterWorker, 2022-01-25:
I am using `IMX_DEFAULT_BSP="mainline"`.

angolini, 2022-06-03:
Are we going to use a weaker assignment for that variable, @otavio ?

What is the status of this issue?

otavio, 2022-06-03:
@angolini we need to have a use case for this. If @IniterWorker can describe what he is trying to do and provide a way for test, sure.

IniterWorker, 2022-06-08:
@otavio, sure, this work was related to the flash process of the imx6ull via uuu-tools in fastboot mode. The `IMAGE_FSTYPES` overrides the machine configuration and fitImage was not populated properly with `cpio.gz.u-boot`.

To make it work, I create my own __bbclass__ and __flash-image__ (mfgtool-initramfs-image).

<details close>
  <summary>meta-customer/classes/custom-initramfs-image.bbclass</summary>

```
# Generates a Manufacturing Tool Initramfs image
#
# This generates the initramfs used for the installation process. The
# image provides the utilities which are used, in the target, during
# the process and receive the commands from the MfgTool application.
#
# Copyright 2014-2017 (C) O.S. Systems Software LTDA.

DEPENDS += "u-boot-mfgtool linux-mfgtool"

FEATURE_PACKAGES_mtd = "packagegroup-fsl-mfgtool-mtd"
FEATURE_PACKAGES_extfs = "packagegroup-fsl-mfgtool-extfs"
FEATURE_PACKAGES_f2fs = "packagegroup-fsl-mfgtool-f2fs"

IMAGE_ROOTFS_SIZE ?= "8192"

# Filesystems enabled by default
DEFAULT_FS_SUPPORT = " \
    mtd \
    extfs \
"

IMAGE_FEATURES = " \
    ${DEFAULT_FS_SUPPORT} \
    \
    read-only-rootfs \
"

# Avoid installation of syslog
BAD_RECOMMENDATIONS += "busybox-syslog"

# Avoid static /dev
USE_DEVFS = "1"

inherit core-image

CORE_IMAGE_BASE_INSTALL = " \
    ${CORE_IMAGE_EXTRA_INSTALL} \
"

```
</details>

<details close>
  <summary>meta-customer/images/flash-image.bb</summary>

```
DESCRIPTION = "Small image to be used with Manufacturing Tool \
(mfg-tool) in a production environment."

LICENSE = "MIT"

inherit custom-initramfs-image

IMAGE_INSTALL += " \
    parted \
"
```
</details>

<details close>
  <summary>meta-customer/conf/machine/imx6ull-customer-machine.conf</summary>

```

MACHINEOVERRIDES =. "mx6:mx6ull:"

include conf/machine/include/imx-base.inc
include conf/machine/include/arm/armv7a/tune-cortexa7.inc

MACHINE_FEATURES:remove = "wifi bluetooth bcm4339 bcm43455 touchscreen"

PREFERRED_PROVIDER_virtual/kernel = "linux-fslc-lts"

#	imx6ull-dart6ul-6ulcustomboard.dtb EVK
KERNEL_DEVICETREE = " \
	imx6ull-dart6ul-customer-custom.dtb \
"

KERNEL_CLASSES  += " kernel-fitimage"
KERNEL_IMAGETYPES = "fitImage"

UBOOT_MAKE_TARGET = "SPL u-boot.img"
UBOOT_INITIAL_ENV = "u-boot-initial-env"
UBOOT_SUFFIX = "img"
UBOOT_ENTRYPOINT = "0x80008000"
SPL_BINARY = "SPL"

UBOOT_CONFIG ??= "emmc"

UBOOT_CONFIG[emmc]     = "/customer_custom_emmc_config"
UBOOT_CONFIG[mfgtool]  = "/customer_custom_mfg_config"

# Use the gadget usb serial
# SERIAL_CONSOLES = "115200;ttymxc0 115200;ttyGS0"
SERIAL_CONSOLES = "115200;ttymxc0"
IMAGE_FSTYPES = "tar.bz2 tar.gz cpio.gz"

# Use a custom image for uuu-tools / fastboot process
INITRAMFS_IMAGE = "flash-image"

DISTRO_FEATURES += "acl agrp ext2 largefile usbgadget usbhost xattr multiarch"
```
</details>

Do you need more information?

otavio, 2022-06-08:
Can you prepare a PR?

IniterWorker, 2022-06-08:
@otavio, I must ask two questions before:
- Can we make `IMAGE_FSTYPES` weak?
- Did you make it work before with `cpio.gz.u-boot`? (Because, I need to remove it on imx6ull machine. See https://github.com/Freescale/meta-freescale/issues/951#issuecomment-1021247221)

otavio, 2022-06-09:
> @otavio, I must ask two questions before:
> 
> * Can we make `IMAGE_FSTYPES` weak?

Yes. 

> * Did you make it work before with `cpio.gz.u-boot`? (Because, I need to remove it on imx6ull machine. See [fitImage: fsl-image-mfgtool-initramfs doesn't allow ramfs #951 (comment)](https://github.com/Freescale/meta-freescale/issues/951#issuecomment-1021247221))

Yes but without fitImage use.

PR #1104 (2022-06-13): classes: mfgtool-initramfs-image use weak variable for IMAGE_FSTYPES
Close #951 by changing `IMAGE_FSTYPES` to a weak variable
closed 2022-06-13 by otavio