Freescale/meta-freescale #959: imx-vpu-hantro-daemon fetch fails with ERROR 304

zandrey, 2022-01-23:
After 4d713c80, `imx-vpu-hantro-daemon` has been upgraded to version `1.1.1`.

Unfortunately, fetcher fails to pull the archive from https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//imx-vpu-hantro-daemon-1.1.1.tar.gz

Trying to fetch it manually using `wget` results in the following output:
```
$ wget https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//imx-vpu-hantro-daemon-1.1.1.tar.gz
--2022-01-23 21:34:54--  https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//imx-vpu-hantro-daemon-1.1.1.tar.gz
Resolving www.nxp.com (www.nxp.com)... 2.20.219.131
Connecting to www.nxp.com (www.nxp.com)|2.20.219.131|:443... connected.
HTTP request sent, awaiting response... 304 Not Modified
2022-01-23 21:34:54 ERROR 304: Not Modified.
```

Previous version can be fetched without any issues:
```
$ wget https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//imx-vpu-hantro-daemon-1.1.0.tar.gz
--2022-01-23 21:37:06--  https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//imx-vpu-hantro-daemon-1.1.0.tar.gz
Resolving www.nxp.com (www.nxp.com)... 2.20.219.131
Connecting to www.nxp.com (www.nxp.com)|2.20.219.131|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 129316 (126K) [application/x-gzip]
Saving to: ‘imx-vpu-hantro-daemon-1.1.0.tar.gz’

imx-vpu-hantro-daemon-1.1.0.tar.gz                                  100%[==================================================================================================================================================================>] 126.29K  --.-KB/s    in 0.007s  

2022-01-23 21:37:07 (18.0 MB/s) - ‘imx-vpu-hantro-daemon-1.1.0.tar.gz’ saved [129316/129316]
```

**Cc:** @thochstein

otavio, 2022-01-23:
It works for me. See 
```console
otavio@yocto-env ~/src/ossystems/oel/master/build % ye pv imx-vpu-hantro-daemon_
~/src/ossystems/oel/master/build/tmp/deploy/ipk/cortexa53-crypto-mx8m/imx-vpu-hantro-daemon_1.1.1-r0_cortexa53-crypto-mx8m.ipk
drwxr-xr-x root/root         0 2021-11-10 07:32 ./usr/
drwxr-xr-x root/root         0 2021-11-10 07:32 ./usr/bin/
-rwxr-xr-x root/root    155088 2021-11-10 07:32 ./usr/bin/vsidaemon
/home/otavio/src/ossystems/oel/master/build/tmp/deploy/ipk/cortexa53-crypto-mx8m/imx-vpu-hantro-daemon_1.1.1-r0_cortexa53-crypto-mx8m.ipk
```

otavio, 2022-01-23:
Maybe, it was a transient error? or has been fixed on their end? Could you double-check it?

zandrey, 2022-01-24:
@otavio Just double-checked, and the issue is still there, I get `2022-01-24 09:42:03 ERROR 304: Not Modified.` when I try to fetch the new archive manually.

MaxKrummenacher, 2022-01-24:
@thochstein
My guess is that NXP's distributed download server system doesn't sync the servers reliably.

If I try to fetch it from home: www.nxp.com evaluates to 2.20.219.131 and I get the same `304: Not Modified.` as Andrey does.
If I fetch on our build server: www.nxp.com evaluates to 2.19.72.166 and wget (and a build) succeeds.

See also:
https://community.nxp.com/t5/i-MX-Processors/i-MX-Yocto-build-fails-with-quot-ERROR-304-Not-Modified-quot/m-p/1378976
https://community.nxp.com/t5/i-MX-Processors/Download-of-sof-imx-1-5-0-2-tar-gz-Fails/m-p/1217070

otavio, 2022-01-24:
@laurenpost, could you forward this to internal IT team?

thochstein, 2022-01-24:
Sorry for the trouble. I've got a ticket in. Will let you know.

thochstein, 2022-01-25:
I saw the same failure yesterday, today it works. I haven't heard from IT but maybe you can try it, @zandrey ?

zandrey, 2022-01-26:
@thochstein 

Unfortunately, it still persists...

```
$ wget https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//imx-vpu-hantro-daemon-1.1.1.tar.gz
--2022-01-26 08:18:57--  https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//imx-vpu-hantro-daemon-1.1.1.tar.gz
Resolving www.nxp.com (www.nxp.com)... 2.20.219.131
Connecting to www.nxp.com (www.nxp.com)|2.20.219.131|:443... connected.
HTTP request sent, awaiting response... 304 Not Modified
2022-01-26 08:18:57 ERROR 304: Not Modified.

$ wget https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//imx-vpu-hantro-daemon-1.1.0.tar.gz
--2022-01-26 08:19:11--  https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//imx-vpu-hantro-daemon-1.1.0.tar.gz
Resolving www.nxp.com (www.nxp.com)... 2.20.219.131
Connecting to www.nxp.com (www.nxp.com)|2.20.219.131|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 129316 (126K) [application/x-gzip]
Saving to: ‘imx-vpu-hantro-daemon-1.1.0.tar.gz’

imx-vpu-hantro-daemon-1.1.0.tar.gz                               100%[==========================================================================================================================================================>] 126.29K  --.-KB/s    in 0.007s  

2022-01-26 08:19:11 (17.8 MB/s) - ‘imx-vpu-hantro-daemon-1.1.0.tar.gz’ saved [129316/129316]

```

_www.nxp.com_ resolves to `2.20.219.131` and it still returns ERROR: 304

thochstein, 2022-01-26:
@zandrey, can you try using `curl`?

zandrey, 2022-01-28:
@thochstein 

Unfortunately, `curl` returns the same status:

```
curl -D- https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//imx-vpu-hantro-daemon-1.1.1.tar.gz
HTTP/2 304
content-type: application/x-gzip
last-modified: Wed, 10 Nov 2021 07:33:08 GMT
etag: "99e96b2e00516d56a78a95a80851076b:1638935413.611178"
cache-control: max-age=1800
expires: Fri, 28 Jan 2022 18:43:39 GMT
date: Fri, 28 Jan 2022 18:13:39 GMT
set-cookie: CC=CH; path=/; secure
server-timing: cdn-cache; desc=HIT
server-timing: edge; dur=1
x-content-type-options: nosniff
strict-transport-security: max-age=31536000; includeSubdomains;
x-xss-protection: 1; mode=block
x-frame-options: SAMEORIGIN
```

Can fetch it using any tool...

-- andrey

thochstein, 2022-01-28:
Thanks Andrey. It fails again for me. Can you try this and attach the output?
`curl -v -o /dev/null https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//imx-vpu-hantro-daemon-1.1.1.tar.gz`

zandrey, 2022-01-28:
Sure, here is the output:

```
$ curl -v -o /dev/null https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//imx-vpu-hantro-daemon-1.1.1.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*   Trying 2.20.219.131:443...
* TCP_NODELAY set
* Connected to www.nxp.com (2.20.219.131) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: /etc/ssl/certs
} [5 bytes data]
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
} [512 bytes data]
* TLSv1.3 (IN), TLS handshake, Server hello (2):
{ [122 bytes data]
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
{ [29 bytes data]
* TLSv1.3 (IN), TLS handshake, Certificate (11):
{ [4024 bytes data]
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
{ [264 bytes data]
* TLSv1.3 (IN), TLS handshake, Finished (20):
{ [52 bytes data]
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
} [1 bytes data]
* TLSv1.3 (OUT), TLS handshake, Finished (20):
} [52 bytes data]
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384
* ALPN, server accepted to use h2
* Server certificate:
*  subject: C=US; ST=Arizona; L=Chandler; O=NXP SEMICONDUCTORS USA, INC.; CN=www.nxp.com
*  start date: Dec 17 00:00:00 2021 GMT
*  expire date: Sep 13 23:59:59 2022 GMT
*  subjectAltName: host "www.nxp.com" matched cert's "www.nxp.com"
*  issuer: C=US; O=DigiCert Inc; OU=www.digicert.com; CN=GeoTrust RSA CA 2018
*  SSL certificate verify ok.
* Using HTTP2, server supports multi-use
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
} [5 bytes data]
* Using Stream ID: 1 (easy handle 0x5622b9716880)
} [5 bytes data]
> GET /lgfiles/NMG/MAD/YOCTO//imx-vpu-hantro-daemon-1.1.1.tar.gz HTTP/2
> Host: www.nxp.com
> user-agent: curl/7.68.0
> accept: */*
>
{ [5 bytes data]
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
{ [249 bytes data]
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
{ [249 bytes data]
* old SSL session ID is stale, removing
{ [5 bytes data]
* Connection state changed (MAX_CONCURRENT_STREAMS == 100)!
} [5 bytes data]
< HTTP/2 304
< content-type: application/x-gzip
< last-modified: Wed, 10 Nov 2021 07:33:08 GMT
< etag: "99e96b2e00516d56a78a95a80851076b:1638935413.611178"
< cache-control: max-age=1800
< expires: Fri, 28 Jan 2022 22:10:52 GMT
< date: Fri, 28 Jan 2022 21:40:52 GMT
< set-cookie: CC=CH; path=/; secure
< server-timing: cdn-cache; desc=HIT
< server-timing: edge; dur=1021
< x-content-type-options: nosniff
< strict-transport-security: max-age=31536000; includeSubdomains;
< x-xss-protection: 1; mode=block
< x-frame-options: SAMEORIGIN
<
{ [0 bytes data]
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
* Connection #0 to host www.nxp.com left intact
```

zandrey, 2022-02-04:
@thochstein 

Do you maybe have an update here? It is still failing, and unfortunately blocks CI builds that I have...

If this problem persists, perhaps a revert of 4d713c80302466a3d4d2b33e071cfeeb866aa548 should be considered...

-- andrey

thochstein, 2022-02-04:
Sorry, I was told they had it narrowed down and were following up with a third-party vendor. Let me bug them again.

thochstein, 2022-02-04:
I'm told they added a workaround around 1:00 PM MST yesterday. It failed for me as well, so they'll continue looking. I'll revert it for now.

thochstein, 2022-02-04:
@zandrey or @otavio, would you happen to know if I would need to revert `imx-vpu-hantro` as well?

otavio, 2022-02-04:
If this isn't fixed, it is better or it could misbehave.

Got feedback from the vendor?

thochstein, 2022-02-04:
Yes, the vendor put a workaround in place that they thought would fix it, but seems not.

thochstein, 2022-02-09:
@zandrey, please try again, seems to be fixed.

thochstein, 2022-02-16:
Yes, sorry for all the trouble here. We heard of a problem yesterday and IT is looking into it again...

otavio, 2022-02-16:
I reopened it so we have it as a central point of contact for people facing this problem.

zandrey, 2022-09-05:
I've hit the same issue after upgrade to _1.1.3_ has been submitted, the issue is intermittent and currently gives `HTTP/2 304`... 😞 

Internal CI shown that tarball was briefly available on 02/09/2022, then until today - not anymore...

jclsn, 2022-09-05:
It is a matter of luck. They set up a premirror somewhere I think. I have holidays though 😌

thochstein, 2022-09-06:
I've alerted our IT. Sorry again for the trouble.

zandrey, 2022-09-08:
Hello @thochstein 

> I've alerted our IT. Sorry again for the trouble.

Thanks a lot, it's working again. I was able to fetch the `imx-vpu-hantro-daemon-1.1.3.tar.gz` tarball.

thochstein, 2022-09-12:
Glad to hear it is working. I don't think IT did anything, but they did today clear a cache to hopefully help anything intermittent that might still be lingering.

Felix1165, 2022-11-07:
Got the error aggain today for imx-vpu-hantro-daemon version 1.1.3. can someone look into it? im a litte pressed for time, to get a new yocto build.

otavio, 2022-11-07:
@thochstein

thochstein, 2022-11-10:
Please try again.

lbabaly, 2022-12-15:
I bumped into this problem as well:
wget -t 2 -T 30 --passive-ftp -P /workdir/downloads/ 'https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//imx-vpu-hantro-daemon-1.1.3.tar.gz' --progress=dot -v
--2022-12-15 06:56:46--  https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//imx-vpu-hantro-daemon-1.1.3.tar.gz
Resolving www.nxp.com (www.nxp.com)... 104.103.110.198
Connecting to www.nxp.com (www.nxp.com)|104.103.110.198|:443... connected.
HTTP request sent, awaiting response... 304 Not Modified
2022-12-15 06:56:46 ERROR 304: Not Modified.

otavio, 2022-12-15:
@thochstein could you ask internally?

thochstein, 2022-12-15:
@otavio, reported.
@lbabaly, it works for me, so please try again in case it is fixed.

thochstein, 2022-12-15:
And now our IT has definitely done a 'reset', so should be fixed.

lbabaly, 2022-12-15:
@thochstein Now, it's working, thank you

zandrey, 2023-01-26:
@thochstein,

FYI, I've hit this issue today after e9b11f4ce4b786e777f506eb8764d2988f2841b7 was merged upgrading `firmware-sof-imx` package:
```
$ wget https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//sof-imx-2.3.0.tar.gz
--2023-01-26 21:13:29--  https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//sof-imx-2.3.0.tar.gz
Resolving www.nxp.com (www.nxp.com)... 104.79.20.32
Connecting to www.nxp.com (www.nxp.com)|104.79.20.32|:443... connected.
HTTP request sent, awaiting response... 304 Not Modified
2023-01-26 21:13:29 ERROR 304: Not Modified.
```

thochstein, 2023-01-26:
Hmm, it did work for me just now. Please try once more? I will let IT know.

thochstein, 2023-01-26:
Oh, weird, it worked just once, now is failing for me.

thochstein, 2023-01-30:
Please try again, IT has done a short-term fix again. They also have a long-term fix they are trying, so hopefully we'll finally see this go away.

pacho94, 2023-02-21:
Hello i'm having the same issue on the url https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//imx-vpu-hantro-daemon-1.1.3.tar.gz.
The wget output gives 304 Not Modified

thochstein, 2023-02-21:
It is working for me. I will let IT know.

thochstein, 2023-02-21:
@pacho94, please try again. IT informs me that they hope to have eliminated this problem going forward.

pacho94, 2023-02-23:
I tried now and is working. Thanks!

Akbar-arm, 2023-02-23:
imx-vpu-hantro-daemon now works but now zephyr-demo-imx_3.1.0 fails in other builds for what I think is a related reason! I have to still do the trick of editing /etc/hosts and mapping ip addresses to www.nxp.com for it to succeed.

thochstein, 2023-02-23:
Hi @Akbar-arm, your issue sounds earlier than the problem we have been fighting. Can you check your DNS and/or proxy settings with your network mgmt?

kaspt, 2023-02-24:
We have the same error 304 while download the file. We try it also on a mobile network. 
[https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//sof-imx-2.3.0.tar.gz](https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//sof-imx-2.3.0.tar.gz)

thochstein, 2023-02-24:
Sorry @kaspt, IT missed that one but thinks it is fixed now. Please try again.

goshiw, 2023-03-20:
Hello

I face same issue regarding following files.
https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//sof-imx-2.2.0.tar.gz
https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//sof-zephyr-2.2.0.tar.gz
https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//imx-vpu-hantro-daemon-1.1.4.tar.gz

I tried wget but 304 error is occured.
Does anyone have same issue now?

goshiw, 2023-03-23:
Hi @thochstein , 
Now I can access these files.
It might be proxy issue.  
Thank you for your cooperation.

otavio, 2023-10-12:
It appears that this issue has been resolved since March, so I will mark it as closed.

closed 2023-10-12 by otavio