Freescale/meta-freescale #961: SCMVERSION ?= "y" Image creation fails when `linux-imx` is populated from sstate cache.

niziak, 2022-01-25:
Hello.
When external additional kernel modules are built by separate recipes following error is shown:
```
ERROR: XXX-image-1.0-r0 do_rootfs: Unable to install packages.
...
Solver encountered 2 problem(s):
Problem 1/2:

- package jailhouse-0.12-r0.imx8mmevk requires kernel-module-jailhouse-5.10.35-lts-5.10.y+gd5366be4cdf6, but none of the providers can be installed
- conflicting requests
- nothing provides kernel-5.10.35-lts-5.10.y+gd5366be4cdf6 needed by kernel-module-jailhouse-5.10.35-lts-5.10.y+gd5366be4cdf6-0.12-r0.imx8mmevk

Solution 1:
- do not ask to install a package providing jailhouse
```

Problem occurs in some specific scenario, where linux-imx is downloaded directly from sstate cache and additional modules (i.e. `jailhouse`) are built separately. Modules requires kernel source. Kernel source is unpacked and then patched by our local patches.
Every patch generates new commit in kernel source (every time different hash because current timestamp is part of hash base),.

When SCMVERSION is enabled (which is default) the kernel package name is different each time so the built module package name doesn't match to kernel.

Solution is to disable SCMVERSION by setting `SCMVERSION=""`

otavio, 2022-01-25:
Is there a way for me to reproduce it?

angolini, 2022-06-03:
I will close this issue. If there is a way to reproduce it on our side, please reopen and update with the how-to.

closed 2022-06-03 by angolini

junzhuimx, 2023-11-29:
This patch is not friendly when SRCREV="${AUTOREV}", which will break as 
git rev-parse --verify --short AUTOINC 2> /dev/null

otavio, 2023-11-29:
How would you recommend we tackle this issue, @niziak and @junzhuimx?

dwagenk, 2023-11-30:
Probably [`SRCPV`](https://docs.yoctoproject.org/singleindex.html#term-SRCPV) is the correct value to use then. I don't know if it is univerally set (and we can just assume it to always be present) or if we need a special case handling for `AUTOREV`.

dwagenk, 2023-11-30:
I looked through the `SRCPV` documentation and implementation. It changed with Nanbield and for the `fsl-kernel-localversion.bbclass` would require an implementation handling the version differences. Both use the underlying bitbake function `bb.fetch2.get_srcrev`, so we could use that. But with the 64 character limit for the kernel localversion (according to kconfig help for `CONFIG_LOCALVERSION`) we'd have to strip the value to limit it's length.