Freescale/meta-freescale #966: devtool failing in do_kernel_localversion on dunfell

BWhitten, 2022-01-28:
Hi,

It seems that the attempt to move defconfig in fsl-kernel-localversion.bbclass fails when trying to use devtool to modify the kernel.
If I make a bbappend in my layer to add a defconfig as per the manual I still get the same error, is the defconfig at the wrong location?

**Steps to reproduce**
```
repo init -u https://github.com/Freescale/fsl-community-bsp-platform -b dunfell
repo sync
MACHINE=imx8mmevk DISTRO=fslc-wayland EULA=1 source ./setup-environment build
devtool modify virtual/kernel
```

**Results in:**
```
ERROR: Execution of '/home/ben/Projects/yocto-fslc/build/tmp/work/imx8mmevk-fslc-linux/linux-fslc-imx/5.4.147+gitAUTOINC+074a74780f-r0/devtooltmp-z9gf6f_i/temp/run.do_kernel_localversion.348332' failed with exit code 2
ERROR: Logfile of failure stored in: /home/ben/Projects/yocto-fslc/build/tmp/work/imx8mmevk-fslc-linux/linux-fslc-imx/5.4.147+gitAUTOINC+074a74780f-r0/devtooltmp-z9gf6f_i/temp/log.do_kernel_localversion.348332
Log data follows:
| DEBUG: Executing shell function do_kernel_localversion
| /home/ben/Projects/yocto-fslc/build/tmp/work/imx8mmevk-fslc-linux/linux-fslc-imx/5.4.147+gitAUTOINC+074a74780f-r0/devtooltmp-z9gf6f_i/temp/run.do_kernel_localversion.348332: 121: cannot open /home/ben/Projects/yocto-fslc/build/tmp/work/imx8mmevk-fslc-linux/linux-fslc-imx/5.4.147+gitAUTOINC+074a74780f-r0/devtooltmp-z9gf6f_i/workdir/defconfig: No such file
| WARNING: exit code 2 from a shell command.
| ERROR: Execution of '/home/ben/Projects/yocto-fslc/build/tmp/work/imx8mmevk-fslc-linux/linux-fslc-imx/5.4.147+gitAUTOINC+074a74780f-r0/devtooltmp-z9gf6f_i/temp/run.do_kernel_localversion.348332' failed with exit code 2
NOTE: Tasks Summary: Attempted 428 tasks of which 0 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 1 seconds
ERROR: Extracting source for linux-fslc-imx failed
```

BWhitten, 2022-01-28:
From running devtool with --keep-temp it appears that devtool puts the defconfig in ${WORKDIR}/../oe-local-files/

otavio, 2022-01-29:
I am not a `devtool` user so I never facet it; are you going to investigate it?

BWhitten, 2022-01-31:
I will try against master this evening. Just so I understand, what is this class trying to achieve? I see the population of a .config happens in the kernel class already, not sure why it needs to happen again?

angolini, 2022-01-31:
I just tested on master and this is the output. For me it worked.

Please, try it on master only to make sure if it worked or not on master for you.

```
$ devtool modify virtual/kernel
NOTE: Starting bitbake server...
NOTE: Reconnecting to bitbake server...
NOTE: Retrying server connection (#1)...
WARNING: Host distribution "ubuntu-21.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |###############################################################################################################################################################################| Time: 0:00:00
Loaded 3816 entries from dependency cache.

Summary: There was 1 WARNING message shown.
INFO: Mapping virtual/kernel to linux-fslc
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.53.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-fslc-linux"
MACHINE              = "imx8mm-lpddr4-evk"
DISTRO               = "fslc-framebuffer"
DISTRO_VERSION       = "3.4-snapshot-20220131"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa53 crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "master:92bb6f72ceb39c99e5c93c0a99b70fb210233acb"
meta-oe              
meta-multimedia      
meta-python          = "master:7c5dbbc9ef632cfe2fb913bf166c9a36812d96de"
meta-freescale       = "master:5f806a7d75dc18477b68a72adbe1efd461bbd635"
meta-freescale-3rdparty = "master:60595861de106c48c94875e49303ca6bb6270a1a"
meta-freescale-distro = "master:c99cb8577e5a9f75a9c8d3765df03325da95d7ad"
workspace            = "<unknown>:<unknown>"

Initialising tasks: 100% |##########################################################################################################################################################################| Time: 0:00:00
Sstate summary: Wanted 51 Local 51 Mirrors 0 Missed 0 Current 49 (100% match, 100% complete)
NOTE: Executing Tasks
NOTE: Tasks Summary: Attempted 456 tasks of which 456 didn't need to be rerun and all succeeded.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 1 seconds
INFO: Copying kernel config to workspace
INFO: Recipe linux-fslc now set up to build from /media/....../yocto/master/build/workspace/sources/linux-fslc

```

BWhitten, 2022-01-31:
Fails here issuing
```
repo init -u https://github.com/Freescale/fsl-community-bsp-platform -b master
repo sync
MACHINE=imx8mm-lpddr4-evk DISTRO=fslc-wayland EULA=1 source ./setup-environment build
devtool modify virtual/kernel
```

Failure:
```
Build Configuration:
BB_VERSION           = "1.53.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "linuxmint-20"
TARGET_SYS           = "aarch64-fslc-linux"
MACHINE              = "imx8mm-lpddr4-evk"
DISTRO               = "fslc-wayland"
DISTRO_VERSION       = "3.4-snapshot-20220131"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa53 crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:92bb6f72ceb39c99e5c93c0a99b70fb210233acb"
meta-oe              
meta-multimedia      
meta-python          = "HEAD:c597a154dd66c615aa636b071bb8220904ee19a5"
meta-freescale       = "HEAD:5f806a7d75dc18477b68a72adbe1efd461bbd635"
meta-freescale-3rdparty = "HEAD:60595861de106c48c94875e49303ca6bb6270a1a"
meta-freescale-distro = "HEAD:c99cb8577e5a9f75a9c8d3765df03325da95d7ad"
workspace            = "<unknown>:<unknown>"

NOTE: Fetching uninative binary shim http://downloads.yoctoproject.org/releases/uninative/3.4/x86_64-nativesdk-libc-3.4.tar.xz;sha256sum=126f4f7f6f21084ee140dac3eb4c536b963837826b7c38599db0b512c3377ba2 (will check PREMIRRORS first)
Initialising tasks: 100% |#############################################################################################################################################################| Time: 0:00:00
Sstate summary: Wanted 100 Local 0 Mirrors 0 Missed 100 Current 0 (0% match, 0% complete)
NOTE: Executing Tasks
ERROR: ExecutionError('/home/ben/Projects/yocto-fslc-master/build/tmp/work/imx8mm_lpddr4_evk-fslc-linux/linux-fslc/5.15.16+gitAUTOINC+d084d16632-r0/devtooltmp-ff1j780s/temp/run.do_kernel_localversion.764334', 2, None, None)
ERROR: Logfile of failure stored in: /home/ben/Projects/yocto-fslc-master/build/tmp/work/imx8mm_lpddr4_evk-fslc-linux/linux-fslc/5.15.16+gitAUTOINC+d084d16632-r0/devtooltmp-ff1j780s/temp/log.do_kernel_localversion.764334
Log data follows:
| DEBUG: Executing shell function do_kernel_localversion
| /home/ben/Projects/yocto-fslc-master/build/tmp/work/imx8mm_lpddr4_evk-fslc-linux/linux-fslc/5.15.16+gitAUTOINC+d084d16632-r0/devtooltmp-ff1j780s/temp/run.do_kernel_localversion.764334: 162: cannot open /home/ben/Projects/yocto-fslc-master/build/tmp/work/imx8mm_lpddr4_evk-fslc-linux/linux-fslc/5.15.16+gitAUTOINC+d084d16632-r0/devtooltmp-ff1j780s/workdir/defconfig: No such file
| WARNING: exit code 2 from a shell command.
NOTE: Tasks Summary: Attempted 455 tasks of which 0 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 2 seconds
ERROR: Extracting source for linux-fslc failed
```

I think if I were to build an image then issue devtool it would pass, it's the fact I'm coming at it from a blank directory?

angolini, 2022-02-01:
I don't know. I don't use devtool that much, and everytime I use it is on top of some build

Let me test your lines on my machine, I will try to test it today.

closed 2022-08-01 by otavio