Freescale/meta-freescale #969: imx-atf build fails with ccache

tprrt, 2022-02-02:
EXTRA_OEMAKE isn't correctly appended when ccache is enabled, because the function remove_options_tail doesn't support this case.

```
ERROR: imx-atf-2.4+gitAUTOINC+5782363f92-r0 do_compile: oe_runmake failed
ERROR: imx-atf-2.4+gitAUTOINC+5782363f92-r0 do_compile: Execution of '/imx-atf/2.4+gitAUTOINC+5782363f92-r0/temp/run.do_compile.2139094' failed with exit code 1
ERROR: Logfile of failure stored in: /imx-atf/2.4+gitAUTOINC+5782363f92-r0/temp/log.do_compile.2139094
Log data follows:
| DEBUG: Executing shell function do_compile                                                                                                                                                                                                  | NOTE: make -j 8 CROSS_COMPILE=aarch64-linux- PLAT=imx8qx LD=aarch64-linux-ld CC=ccache aarch64-linux-gcc bl31
| make: *** No rule to make target 'aarch64-linux-gcc'.  Stop.
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/imx-atf/2.4+gitAUTOINC+5782363f92-r0/temp/run.do_compile.2139094' failed with exit code 1
ERROR: Task (/recipes-bsp/imx-atf/imx-atf_2.4.bb:do_compile) failed with exit code '1'
```

tprrt, 2022-02-02:
Hello @otavio,

Do you want me to improve the remove_options_tail function?
Otherwise I can just disable ccache on this recipe.

tprrt, 2022-02-03:
I will push a fix that disable ccache on imx-atf because when I updated the recipe to support the case when ccache is enabled, a new error is raised:

> ERROR: imx-atf-2.4+gitAUTOINC+5782363f92-r0 do_compile: oe_runmake failed
> ERROR: imx-atf-2.4+gitAUTOINC+5782363f92-r0 do_compile: ExecutionError('/src/build/tmp-glibc/work/cortexa35-mx8-oe-linux/imx-atf/2.4+gitAUTOINC+5782363f92-r0/temp/run.do_compile.1367342', 1, None, None)
> ERROR: Logfile of failure stored in: /src/build/tmp-glibc/work/cortexa35-mx8-oe-linux/imx-atf/2.4+gitAUTOINC+5782363f92-r0/temp/log.do_compile.1367342
> Log data follows:
> | DEBUG: Executing shell function do_compile
> | NOTE: make -j 8 CROSS_COMPILE=aarch64-oe-linux- PLAT=imx8qx LD="aarch64-oe-linux-ld" CC="ccache aarch64-oe-linux-gcc" bl31
> | /bin/sh: 1: Syntax error: Unterminated quoted string
> | make: *** No rule to make target 'aarch64-oe-linux-gcc"'.  Stop.
> | ERROR: oe_runmake failed
> | WARNING: exit code 1 from a shell command.
> ERROR: Task (/src/layers/meta-freescale/recipes-bsp/imx-atf/imx-atf_2.4.bb:do_compile) failed with exit code '1'
> NOTE: Tasks Summary: Attempted 765 tasks of which 757 didn't need to be rerun and 1 failed.
> NOTE: Writing buildhistory
> NOTE: Writing buildhistory took: 0 seconds
> NOTE: Build completion summary:
> NOTE:   do_deploy_source_date_epoch: 0.0% sstate reuse(0 setscene, 1 scratch)
> NOTE:   do_populate_lic: 0.0% sstate reuse(0 setscene, 1 scratch)

thochstein, 2022-02-03:
> a new error is raised

Hi @tprrt, it seems that quotation marks are needed around the value.

`EXTRA_OEMAKE += "CC="${@remove_options_tail(d.getVar('CC'))}""`

tprrt, 2022-02-03:
Hello @thochstein,

> > a new error is raised
> 
> Hi @tprrt, it seems that quotation marks are needed around the value.
> 
> `EXTRA_OEMAKE += "CC="${@remove_options_tail(d.getVar('CC'))}""`

When I tried to add quotation marks around the value, I obtained the error above, whereas there quotation marks seems correct:

> NOTE: make -j 8 CROSS_COMPILE=aarch64-oe-linux- PLAT=imx8qx LD="aarch64-oe-linux-ld" CC="ccache aarch64-oe-linux-gcc" bl31
> | /bin/sh: 1: Syntax error: Unterminated quoted string

thochstein, 2022-02-03:
Okay, it seems I'm not getting `ccache` when I build. Did you do anything to enable it? That `NOTE` actually doesn't make sense to me. I don't see how it comes from the original line that looks like this:

```
EXTRA_OEMAKE += "CC=${@remove_options_tail(d.getVar('CC'))}"
```

This is what I see in `log.do_compile`:

```
NOTE: make -j 8 CROSS_COMPILE=aarch64-fsl-linux- PLAT=imx8mq LD=aarch64-fsl-linux-ld CC=aarch64-fsl-linux-gcc  bl31
```

thochstein, 2022-02-03:
I tried adding a bogus entry:

```
EXTRA_OEMAKE += "CC=${@remove_options_tail(d.getVar('CC'))} bogus"
```

The error I see doesn't match yours but does fail to treat it as part of the CC argument:

```
| NOTE: make CROSS_COMPILE=aarch64-fsl-linux- PLAT=imx8mq LD=aarch64-fsl-linux-ld CC=aarch64-fsl-linux-gcc bogus clean
| make: *** No rule to make target 'bogus'.  Stop.
```
When I add the quotation marks at least it now treats it like part of the compiler:

```
| NOTE: make -j 8 CROSS_COMPILE=aarch64-fsl-linux- PLAT=imx8mq LD=aarch64-fsl-linux-ld CC=aarch64-fsl-linux-gcc  bogus bl31
|   CC      bl31/bl31_main.c
|   CC      bl31/interrupt_mgmt.c
|   CC      common/runtime_svc.c
|   CC      drivers/arm/gic/v3/arm_gicv3_common.c
|   CC      drivers/arm/gic/v3/gic-x00.c
|   CC      bl31/bl31_context_mgmt.c
|   CC      drivers/arm/gic/v3/gicrv3_helpers.c
|   CC      drivers/arm/gic/v3/gicdv3_helpers.c
| aarch64-fsl-linux-gcc: warning: bogus: linker input file unused because linking not done
| aarch64-fsl-linux-gcc: error: bogus: linker input file not found: No such file or directory
```

tprrt, 2022-02-04:
I also added an escape character before the mark quotation, it works much better without. :)

PR #973 (2022-02-05): imx-atf: add mark quotation to prevent build issue with ccache 
Otherwise, when ccache is enable the following issue occurs:
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 8 CROSS_COMPILE=aarch64-oe-linux- PLAT=imx8qx LD=aarch64-oe-linux-ld CC=ccache aarch64-oe-linux-gcc bl31
| make: *** No rule to make target 'aarch64-oe-linux-gcc'.  Stop.
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.

Fix #969
closed 2022-02-05 by otavio