Freescale/meta-freescale #971: optee-test build issues

tprrt, 2022-02-02:
There are some optee-test build issues:

Because python3-pycrypto has been removed from meta-openembedded for security reason:
> ERROR: Nothing PROVIDES 'python3-pycrypto-native' (but /src/layers/meta-freescale/recipes-security/optee-imx/optee-test_3.10.0.imx.bb DEPENDS on or otherwise requires it). Close matches:
>   python3-m2crypto-native
>   python3-pycryptodome-native
>   python3-pycryptodomex-native
> ERROR: Required build target 'optee-test' has no buildable providers.
> Missing or unbuildable dependency chain was: ['optee-test', 'python3-pycrypto-native']

> commit 84769a963974cffc5d877e57a46a289a84bc293d
> Author: Ross Burton <ross@burtonini.com>
> Date:   Fri Jan 14 14:10:08 2022 +0000
> 
>     python3-pycrypto: remove obsolete pycrypto module
>     
>     pycrypto hasn't had a commit upstream since June 2014, which for a
>     security-critical module is not a good sign.
>     
>     Everyone has migrated to the pycryptodome or cryptography modules, so
>     remove pycrypto from meta-python.
>     
>     Signed-off-by: Ross Burton <ross.burton@arm.com>
>     Signed-off-by: Khem Raj <raj.khem@gmail.com>
>     Signed-off-by: Trevor Gamblin <trevor.gamblin@windriver.com>

In addition, there are some compilation issues:

> | In file included from xtest_test.h:9,
> |                  from regression_7000.c:11:
> | regression_7000.c: In function 'xtest_tee_init.part.0':
> | ./adbg/include/adbg.h:136:9: error: '<unknown>' may be used uninitialized [-Werror=maybe-uninitialized]
> |   136 |         Do_ADBG_ExpectPointerNotNULL(Case_p, __FILE__, __LINE__, Got, #Got)
> |       |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> | regression_7000.c:25:14: note: in expansion of macro 'ADBG_EXPECT_NOT_NULL'
> |    25 |         if (!ADBG_EXPECT_NOT_NULL(c, SHARE_MEM01))
> |       |              ^~~~~~~~~~~~~~~~~~~~
> | ./adbg/include/adbg.h:223:6: note: by argument 4 of type 'const void *' to 'Do_ADBG_ExpectPointerNotNULL' declared here
> |   223 | bool Do_ADBG_ExpectPointerNotNULL(ADBG_Case_t *const Case_p,
> |       |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
> | cc1: all warnings being treated as errors
> | make[1]: *** [Makefile:205: /src/build/tmp-glibc/work/cortexa35-oe-linux/optee-test/3.10.0.imx-r0/build/xtest/regression_7000.o] Error 1
> | make[1]: *** Waiting for unfinished jobs....
> |   CC      /src/build/tmp-glibc/work/cortexa35-oe-linux/optee-test/3.10.0.imx-r0/git/host/xtest/xtest_helpers.c
> | make[1]: Leaving directory '/src/build/tmp-glibc/work/cortexa35-oe-linux/optee-test/3.10.0.imx-r0/git/host/xtest'
> | make: *** [Makefile:37: xtest] Error 2
> | make: Leaving directory '/src/build/tmp-glibc/work/cortexa35-oe-linux/optee-test/3.10.0.imx-r0/git'
> | ERROR: oe_runmake failed
> | WARNING: exit code 1 from a shell command.

tprrt, 2022-02-09:
is anyone already working on this issue?

thochstein, 2022-02-09:
Hi Thomas. I'm not working on it. I have been working on porting our latest release (into `meta-freescale` `master`) but got side-tracked. The `optee` recipes are upgraded to 3.15. Perhaps the newer version has fixed this problem.

https://source.codeaurora.org/external/imx/meta-imx/tree/meta-bsp/recipes-security/optee-imx?h=hardknott-5.10.72-2.2.0

tprrt, 2022-02-10:
HI @thochstein,

Thanks, so I will test the latest release, I hope tests will work.

Otherwise, how do you do to functionally validate your new versions without using optee-test?

Best regards,
Thomas

thochstein, 2022-02-10:
When updating components from NXP releases we only do sanity checks (does it boot to a desktop) and rely on the functional testing that has already been done on the component (while in the meta-imx layer) and then the testing of the community itself.

tprrt, 2022-02-10:
Hi @thochstein,

So, I bumped OP-TEE to 3.15, then tests are built successfully, but a few are failed, especially PKCS#11 tests:

```

[...]

* pkcs11_1001 PKCS11: List PKCS#11 slots and get information from
ERR [856] ckteec:ckteec_invoke_init:304: TEEC open session failed ffff0008 from 3

pkcs11_1000.c:216: rv has an unexpected value: 0x30 = CKR_DEVICE_ERROR, expected 0x0 = CKR_OK
  pkcs11_1001 FAILED
 
* pkcs11_1002 PKCS11: Open and close PKCS#11 sessions
ERR [856] ckteec:ckteec_invoke_init:304: TEEC open session failed ffff0008 from 3

pkcs11_1000.c:445: rv has an unexpected value: 0x30 = CKR_DEVICE_ERROR, expected 0x0 = CKR_OK
  pkcs11_1002 FAILED

[...]

```

thochstein, 2022-02-10:
Perhaps there's more in the `meta-imx` layer that needs updating? I'll have to dig and/or ask for help.

tprrt, 2022-02-16:
> Perhaps there's more in the `meta-imx` layer that needs updating? I'll have to dig and/or ask for help.

Yes, I updated the kernel to 5.10 to fix some regression tests errors. But pksc11 tests always fail to open a session to a Trusted Application.

clementfaure, 2022-02-16:
Hi @tprrt,

0xffff0008 -> `TEE_ERROR_ITEM_NOT_FOUND`
It looks like the PKCS#11 TA is missing from `/lib/optee_armtz/`. Could you confirm?

The PKCS#11 trusted application is generated by optee-os compilation. Make sure to also install TAs generated by optee-os compilation in your rootfs:
```
    # Install embedded TAs
    install -d ${D}${base_libdir}/optee_armtz
    find ${B}/ta -name '*.ta' | while read name; do
        install -m 444 $name ${D}${base_libdir}/optee_armtz/
    done
```

clementfaure, 2022-02-16:
Yes, the PKCS#11 TA is missing.
Please find the TA in `<build_output>/ta/pkcs11/fd02c9da-306c-48c7-a49c-bbd827ae86ee.ta` and add it to your target.

tprrt, 2022-02-16:
So, I found my issue, I appended optee-test to the image but not optee-os then only the trusted application for tests have been shipped in my images.

Thank you @clementfaure for putting me on the way. :)

otavio, 2022-02-16:
So the rdepends need to be added. Please prepare a PR ;)

thochstein, 2022-02-16:
I can move the 3.15 recipes into meta-freescale master and add the rdepends.

otavio, 2022-02-19:
@thochstein please base it on top of #980 so we reduce conflicts.

thochstein, 2022-02-24:
Okay, the update is here #985.

Hmm, I think I incorrectly decided that I didn't need the rdepends. I will build an image now to check.

otavio, 2022-02-25:
Fixed in #985.

closed 2022-02-25 by otavio