Freescale/meta-freescale #996: ALSA output broken on imx8mm-lpddr4-evk (current git master)

dv1, 2022-03-04:
I have an imx8m mini LPDDR4 Rev.B EVK board here. I use the `imx8mm-evk-revb-qca-wifi.dtb` DTB.

When I try to play any audio to an ALSA output PCM device, the board completely freezes.

List of loaded modules:

```
# lsmod | sort
Module                  Size  Used by
authenc                16384  1 caam_jr
caam                   28672  1 caam_jr
caam_jr               172032  0
caamalg_desc           40960  1 caam_jr
caamhash_desc          16384  1 caam_jr
caamkeyblob_desc       16384  1 caam_jr
crct10dif_ce           20480  1
crypto_engine          16384  1 caam_jr
error                  24576  7 caamalg_desc,secvio,caamkeyblob_desc,caamhash_desc,caam,caam_jr,fsl_jr_uio
fsl_jr_uio             20480  0
fuse                  131072  1
galcore               393216  2
gpio_ir_recv           16384  0
libdes                 24576  1 caam_jr
rc_core                40960  2 gpio_ir_recv
rng_core               24576  1 caam_jr
secvio                 20480  0
```

`uname -a` output:

`Linux imx8mm-lpddr4-evk 5.10.93+gf28a9b90c506 #1 SMP PREEMPT Thu Jan 20 20:01:30 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux`

I also looked in dmesg:

```
# dmesg | grep -i "sdma\|sound\|audio"
[    0.102559] Advanced Linux Sound Architecture Driver Initialized.
[    0.177392] imx-sdma 302c0000.dma-controller: Direct firmware load for imx/sdma/sdma-imx7d.bin failed with error -2
[    0.177409] imx-sdma 302c0000.dma-controller: Falling back to sysfs fallback for: imx/sdma/sdma-imx7d.bin
[    3.043665]   #1: imx-audio-micfil
[    3.047070]   #2: bt-sco-audio
[    3.050129]   #3: wm8524-audio
[   60.388548] imx-sdma 302c0000.dma-controller: external firmware not found, using ROM firmware
[   60.388685] imx-sdma 30bd0000.dma-controller: external firmware not found, using ROM firmware
[   60.402168] imx-sdma 302b0000.dma-controller: firmware found.
[   60.411510] imx-sdma 302b0000.dma-controller: loaded firmware 4.5
```

And these are the devices listed by `aplay -l`:

```
**** List of PLAYBACK Hardware Devices ****
card 0: imxspdif [imx-spdif], device 0: S/PDIF PCM snd-soc-dummy-dai-0 [S/PDIF PCM snd-soc-dummy-dai-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: btscoaudio [bt-sco-audio], device 0: 30020000.sai-bt-sco-pcm-wb bt-sco-pcm-wb-0 [30020000.sai-bt-sco-pcm-wb bt-sco-pcm-wb-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: wm8524audio [wm8524-audio], device 0: 30030000.sai-wm8524-hifi wm8524-hifi-0 [30030000.sai-wm8524-hifi wm8524-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

I tried all three devices (`hw:0,0`, `hw:0,2`, `hw:3,0`) this way:

`cat /dev/urandom | aplay -f S16_LE -c2 -r48000 -D hw:0,0`

It always hangs, no matter what device I specify with the `-D` argument.

Is this a known issue?

otavio, 2022-03-04:
Please add the firmware of sdma, as dmesg complains. See if it helps.

dv1, 2022-03-04:
Is this useful?

```
# opkg info linux-firmware-imx-sdma-imx7d
Package: linux-firmware-imx-sdma-imx7d
Version: 1:20220209-r0
Depends: linux-firmware-imx-sdma-license
Status: install ok installed
Architecture: all
Installed-Size: 2862
Installed-Time: 1646403328
```

The bitbake recipe is `recipes-bsp/firmware-imx/firmware-imx_8.14.bb`. meta-freescale git HEAD is `f6e0f17d2b777d15d97632055d129a10bd47e525`. Last git commit that touched the `firmware-imx` folder is `dbd0f0ae908a6b287488707cfe9f2d464b9b02d5`.

dv1, 2022-06-22:
Just tested a current Kirkstone build with the exact same machine. `aplay -l` now shows:

```
**** List of PLAYBACK Hardware Devices ****
card 0: btscoaudio [bt-sco-audio], device 0: 30020000.sai-bt-sco-pcm-wb bt-sco-pcm-wb-0 [30020000.sai-bt-sco-pcm-wb bt-sco-pcm-wb-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: wm8524audio [wm8524-audio], device 0: 30030000.sai-wm8524-hifi wm8524-hifi-0 [30030000.sai-wm8524-hifi wm8524-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: imxspdif [imx-spdif], device 0: S/PDIF PCM snd-soc-dummy-dai-0 [S/PDIF PCM snd-soc-dummy-dai-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

And `cat /dev/urandom | aplay -f S16_LE -c2 -r48000 -D hw:1,0` works now - I hear audio on the line out jack. None of the devices hang anymore.

Firmware version:

```
# opkg info linux-firmware-imx-sdma-imx7d
Package: linux-firmware-imx-sdma-imx7d
Version: 1:20220509-r0
Depends: linux-firmware-imx-sdma-license
Status: install ok installed
Architecture: all
Installed-Size: 2862
Installed-Time: 1655841601
```

Kernel:
```
# uname -a
Linux imx8mm-lpddr4-evk 5.15.5+gc1084c2773fc #1 SMP PREEMPT Mon Mar 7 20:03:59 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
```

Note that this is the linux-imx kernel. I included this in the local.conf:

    PREFERRED_PROVIDER_virtual/kernel:imx-nxp-bsp = "linux-imx"

I'll retry with `linux-fslc-imx` later on - I used `linux-imx` because I needed a 5.15 kernel, but `linux-fslc-imx` has been updated to 5.15 too by now.

dv1, 2022-06-22:
The error happened with `linux-fslc-imx` version 5.10.109 . But with version 5.15.48+gb6597629c15c, it went away.  Closing this issue.

closed 2022-06-22 by dv1