Freescale/meta-freescale #997: optee-os 3.15.0.imx no longer compile when the RPMB support is enabled

tprrt, 2022-03-04:
When ```CFG_RPMB_FS=y```, OP-TEE os no longer compiles because ```imx_rpmb.c``` tries to include a header that doesn't exist.
This issue seems linked to a rework that allows to compile IMX_SNVS driver without RPMB.

```
|   CC      optee-os/3.15.0.imx-r0/build/core/arch/arm/plat-imx/imx_rpmb.o
| core/arch/arm/plat-imx/imx_rpmb.c:5:10: fatal error: drivers/imx_snvs.h: No such file or directory
|     5 | #include <drivers/imx_snvs.h>
|       |          ^~~~~~~~~~~~~~~~~~~~
| compilation terminated.
| make: *** [mk/compile.mk:159: optee-os/3.15.0.imx-r0/build/core/arch/arm/plat-imx/imx_rpmb.o] Error 1
| make: *** Waiting for unfinished jobs....
|   CC     optee-os/3.15.0.imx-r0/build/core/crypto/crypto.o
|   CC      optee-os/3.15.0.imx-r0/build/core/arch/arm/plat-imx/imx_dt.o
| make: Leaving directory 'optee-os/3.15.0.imx-r0/git'
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
ERROR: Task (optee-imx/optee-os_3.15.0.imx.bb:do_compile) failed with exit code '1'
```

otavio, 2022-03-04:
@thochstein can you take a look?

thochstein, 2022-03-04:
@clementfaure, can you comment?

clementfaure, 2022-03-07:
Hi,
That's a mistake we've fixed. It will be available in the next release.
Replace `#include <drivers/imx_snvs.h>` with `#include <drivers/nxp_snvs.h>`

otavio, 2022-03-07:
@clementfaure, @thochstein could you guys prepare a PR?

thochstein, 2022-03-07:
I'll make the PR. @tprrt can you test that it fixes the build for you?

tprrt, 2022-03-08:
Hi @thochstein,

I already tried to do this before opening the ticket. :)
I also tried to replace it by ```nxp_snvs.h```.

So, it isn't enough, the following error is raised:

```
| aarch64-linux-ld.bfd: optee-os/3.15.0.imx-r0/build/core/arch/arm/plat-imx/imx_rpmb.o: in function `plat_rpmb_key_is_ready':
| /usr/src/debug/optee-os/3.15.0.imx-r0/git/core/arch/arm/plat-imx/imx_rpmb.c:15: undefined reference to `snvs_get_ssm_mode'
| aarch64-linux-ld.bfd: /usr/src/debug/optee-os/3.15.0.imx-r0/git/core/arch/arm/plat-imx/imx_rpmb.c:16: undefined reference to `snvs_get_security_cfg'
| make: Leaving directory 'optee-os/3.15.0.imx-r0/git'
| make: *** [core/arch/arm/kernel/link.mk:45: optee-os/3.15.0.imx-r0/build/core/all_objs.o] Error 1
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
ERROR: Task (optee-os_3.15.0.imx.bb:do_compile) failed with exit code '1'
```

clementfaure, 2022-03-08:
Hi @tprrt,
What is your platform?

clementfaure, 2022-03-08:
That's what I thought.
The OP-TEE SNVS driver is not supported on i.MX8QM/QXP platforms. That's why the `CFG_NXP_SNVS` is forced to `n` and `SNVS_BASE/SIZE` not defined.
That leads us to `core/arch/arm/plat-imx/sub.mk` where `srcs-$(CFG_RPMB_FS) += imx_rpmb.c` is false. Instead, it should be `srcs-$(CFG_NXP_SNVS) += imx_rpmb.c`. We will fix that in our next rebase.

clementfaure, 2022-03-08:
@tprrt Not yet. The SNVS driver for these platforms is something totally different.

PR #1001 (2022-03-16): optee-os: Fix RPMB build break
When CFG_RPMB_FS=y, OP-TEE os no longer compiles because imx_rpmb.c
tries to include a header that doesn't exist. This issue seems linked
to a rework that allows to compile i.MX Secure Non-Volatile Storage
driver without RPMB.

```
|   CC      optee-os/3.15.0.imx-r0/build/core/arch/arm/plat-imx/imx_rpmb.o
| core/arch/arm/plat-imx/imx_rpmb.c:5:10: fatal error: drivers/imx_snvs.h: No such file or directory
|     5 | #include <drivers/imx_snvs.h>
|       |          ^~~~~~~~~~~~~~~~~~~~
| compilation terminated.
| make: *** [mk/compile.mk:159: optee-os/3.15.0.imx-r0/ [... cut]
commit 3f452909 (2022-03-16): optee-os: Fix RPMB build break

When CFG_RPMB_FS=y, OP-TEE os no longer compiles because imx_rpmb.c
tries to include a header that doesn't exist. This issue seems linked
to a rework that allows to compile i.MX Secure Non-Volatile Storage
driver without RPMB.

```
|   CC      optee-os/3.15.0.imx-r0/build/core/arch/arm/plat-imx/imx_rpmb.o
| core/arch/arm/plat-imx/imx_rpmb.c:5:10: fatal error: drivers/imx_snvs.h: No such file or directory
|     5 | #include <drivers/imx_snvs.h>
|       |          ^~~~~~~~~~~~~~~~~~~~
| compilation terminated.
| make: *** [mk/compile.mk:159: optee-os/3.15.0.imx-r0 [... cut]
closed 2022-03-16 by otavio