Freescale/meta-freescale #1005: NXP Release 5.15.5-1.0.0 and Yocto 3.5 Kirkstone

zandrey, 2022-03-10:
Recent status [published by Yocto Project](https://lore.kernel.org/all/18e901d83306$9429b010$bc7d1030$@gmail.com/) sets the next LTS 3.5 _Kirkstone_ release date to `2022/04/29`

I've seen some mentions that there is an upcoming release planned from NXP (`5.15.5-1.0.0`) and wanted to check if it is planned to be included in the _meta-freescale_ that would branch out to `kirkstone` from `master`?

One reason being (among all others) is that I'm holding the Kernel branch `5.10-2.2.x-imx` for _linux-fslc-imx_ in my fork, and currently I have either the option to submit it to this layer, pick it up and got with it in _Kirkstone_, or wait till `5.15.5-1.0.0` and drop that Kernel branch altogether, replacing it with new one based on prospective `5.15.5-1.0.0`.

Another reason for this curiosity from my side would be that if the next NXP Release comes out close to the _Kirkstone_ - it might be too late to merge a lot of changes into the LTS release, while if it is imminent - then there would still be time to pick it up and have a "stabilization window" till the end of April.

@thochstein 
Do you have any information or prospective date when the next NXP `5.15.5-1.0.0` comes out?

@otavio 
Should the next NXP `5.15.5-1.0.0` be considered to be integrated into `kirkstone` branch in _meta-freescale_ or it stays with `5.10-2.2.x` instead?

Thanks a lot!

-- andrey

thochstein, 2022-03-10:
Hi @zandrey, `5.15.5-1.0.0` will release March 31 on `honister`. The next quarter `5.15.x-2.0.0` will release Jun 30 on `kirkstone`.

otavio, 2022-03-10:
I'd love to have 5.15-1.0.0 on our kirkstone; @thochstein can we start upstreaming soon?

zandrey, 2022-03-10:
Hi @thochstein!

Thanks a lot for such a prompt reply and the information on the dates!

I guess that with `5.15.5-1.0.0` release end of March, it would still give around 4 weeks to upstream it to `master` in _meta-freescale_ and eventually - to `kirkstone`. I would then rather give up on the `5.10-2.2.x-imx` Kernel and focus on `5.15.y` instead.

-- andrey

thochstein, 2022-03-10:
> I'd love to have 5.15-1.0.0 on our kirkstone; @thochstein can we start upstreaming soon?

Sure, haven't finished moving changes from the last release, mainly gstreamer that I know of.

nobody93, 2022-04-28:
There is no kirkstone branch, should we use master to build Kirkstone?

otavio, 2022-04-28:
For now, yes. We are finishing some more tests before branching but it ought to happen very soon.

escherstair, 2022-05-02:
Based on what I see [here](https://wiki.yoctoproject.org/wiki/Releases) Yocto Kirkstone number has been switched to 4.0 (no more 3.5 as it was before, and as it is in the title of this issue).

otavio, 2022-05-02:
Thanks @escherstair. Updated.

@thochstein what is missing from your side?

thochstein, 2022-05-02:
Sorry, I've had other issues to work on and this has slid. I can start upstreaming this week. Is it too late?

otavio, 2022-05-02:
Please do; we can force it in kirkstone as nobody is using it extensively yet.

franzflasch, 2022-05-04:
Hi @thochstein!

That would be great and would be very appreciated. Any thoughts on how long it will take?

thochstein, 2022-05-04:
Hi @franzflasch, I've started on it, hopefully we can get it done in a week or two.

PierreGrillet, 2022-07-08:
Hi guys,

I wonder the state of the kirkstone branch here. As this issue is still opened, is kirkstone ready?

otavio, 2022-07-11:
We have it all in sync with master and it looks good, AFAIK. Please give it a go.

I'll close this to avoid confusing users.

closed 2022-07-11 by otavio