Freescale/meta-freescale #1006: Status of support for GPU hardware acceleration on i.MX8

escherstair, 2022-03-13:
I've just started the evaluation of i.MX8 family (i.MX8M-Mini in particular) for a new project.
My need is to create a nodejs-based web app running over Linux, and Chromium would seem the most common and used client.

But based on what I found searching around and taliking to some SoMs vendors, it seems that Chromium doesn't have support for i.MX8 GPU.
And it's not clear i any other client has this kind of support.

So i have some questions:
- is this the official Yocto layer for i.MX8 family (i.MX8M-Mini)?
- is [this](https://github.com/Freescale/meta-freescale/tree/master/dynamic-layers/chromium-browser-layer/recipes-browser/chromium) the layer that adds Chromium? 
- is it GPU supported by Chromium (or by another client)?
- how can I help adding this support (if not available yet)?

Thanks in advance

otavio, 2022-03-13:
> * is this the official Yocto layer for i.MX8 family (i.MX8M-Mini)?

Yes. You can use this or NXP official release. 

> * is [this](https://github.com/Freescale/meta-freescale/tree/master/dynamic-layers/chromium-browser-layer/recipes-browser/chromium) the layer that adds Chromium?

This is enabled when meta-browser/meta-chromium is added.

> * is it GPU supported by Chromium (or by another client)?

Yes. We (at O.S. Systems) have customers using i.MX8M family with it as well as other embedded browsers.

> * how can I help adding this support (if not available yet)?

Sending PRs here.

closed 2022-03-16 by otavio

escherstair, 2022-04-01:
I reopen this issue because I tried to enable GPU usage for Chromium on iMX8M adding `--enable-features=UseOzonePlatform --ozone-platform=wayland` startup parameters.
It seems working (because [this CodeSandbox example](https://codesandbox.io/s/rrppl0y8l4?file=/src/App.js) runs) but the performance are poor.
Profiling the execution I see that CPU usage is near to 100% and so it seems that (for some reasons) the processor falls back on CPU rendering.
Can someone confirm that GPU hardware acceleration is fully supported on Chromium on iMX8M family?