Freescale/meta-freescale #1022: imx6qdlsabresd: kernel panic at boot time, using current git master

dv1, 2022-03-29:
I built an imx6qdlsabresd image with current git master (git HEAD cf88b9ac6c9d818b312c8aec774db2ae4f9d4f71 in meta-freescale). Initially it boots fine. But then I see this kernel crash:

```
[    8.015172] Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: _CMAFSLAlloc+0x134/0x134 [galcore]
[    8.026180] CPU1: stopping
[    8.026193] CPU: 1 PID: 161 Comm: udevadm Tainted: G           O      5.10.93+gf28a9b90c506 #1
[    8.026198] Hardware name: Freescale i.MX6 Quad/DualLite (Device Tree)
[    8.026226] [<8010eb74>] (unwind_backtrace) from [<80109fd0>] (show_stack+0x10/0x14)
[    8.026239] [<80109fd0>] (show_stack) from [<80c1d218>] (dump_stack+0x94/0xa8)
[    8.026252] [<80c1d218>] (dump_stack) from [<8010c98c>] (do_handle_IPI+0xf4/0x11c)
[    8.026264] [<8010c98c>] (do_handle_IPI) from [<8010c9cc>] (ipi_handler+0x18/0x20)
[    8.026279] [<8010c9cc>] (ipi_handler) from [<8017cf9c>] (__handle_domain_irq+0x7c/0xd0)
[    8.026296] [<8017cf9c>] (__handle_domain_irq) from [<80520284>] (gic_handle_irq+0x80/0x94)
[    8.026310] [<80520284>] (gic_handle_irq) from [<80100ed0>] (__irq_usr+0x50/0x80)
[    8.026316] Exception stack(0x83197fb0 to 0x83197ff8)
[    8.026324] 7fa0:                                     00aadbb2 000000ff ffffffff 000000c5
[    8.026334] 7fc0: 00aa6078 00000000 00000002 00000010 00aadbb2 00aadbb0 00000000 00000000
[    8.026343] 7fe0: 76f67008 7e8eb870 76e6169f 76e616d4 60030030 ffffffff
[    8.126194] ---[ end Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: _CMAFSLAlloc+0x134/0x134 [galcore] ]---
```

U-Boot version according to UART log:

    U-Boot 2022.01+fslc+ged9503daa3 (Jan 18 2022 - 11:54:53 +0000)

Kernel:

    5.10.93+gf28a9b90c506 (oe-user@oe-host) (arm-poky-linux-gnueabi-gcc (GCC) 11.2.0, GNU ld (GNU Binutils) 2.38.20220313) #1 SMP PREEMPT Thu Jan 20 20:01:30 UTC 2022

Is this a known issue?

otavio, 2022-03-29:
@thochstein did someone reported it internally as well?

thochstein, 2022-03-29:
Checking.

dv1, 2022-03-30:
Also of note: I tried 2 different Sabre SD boards and 2 different SD cards. Error persisted.

otavio, 2022-03-30:
@dv1 please check if using the latest version in https://github.com/Freescale/kernel-module-imx-gpu-viv helps?

otavio, 2022-03-30:
For completeness, here is the log fragment showing the current version:
[image]

dv1, 2022-03-30:
You mean version 6.4.3p4.0 ? Okay, I'll try that out.

dv1, 2022-03-30:
Nope - kernel log shows Galcore version 6.4.3.p4.398061, but still:

```
[    8.682304] Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: _CMAFSLAlloc+0x134/0x134 [galcore]
[    8.693295] CPU1: stopping
[    8.696018] CPU: 1 PID: 161 Comm: udevadm Tainted: G           O      5.10.93+gf28a9b90c506 #1
[    8.704634] Hardware name: Freescale i.MX6 Quad/DualLite (Device Tree)
[    8.711189] [<8010eb74>] (unwind_backtrace) from [<80109fd0>] (show_stack+0x10/0x14)
[    8.718945] [<80109fd0>] (show_stack) from [<80c1d218>] (dump_stack+0x94/0xa8)
[    8.726181] [<80c1d218>] (dump_stack) from [<8010c98c>] (do_handle_IPI+0xf4/0x11c)
[    8.733759] [<8010c98c>] (do_handle_IPI) from [<8010c9cc>] (ipi_handler+0x18/0x20)
[    8.741339] [<8010c9cc>] (ipi_handler) from [<8017cf9c>] (__handle_domain_irq+0x7c/0xd0)
[    8.749443] [<8017cf9c>] (__handle_domain_irq) from [<80520284>] (gic_handle_irq+0x80/0x94)
[    8.757806] [<80520284>] (gic_handle_irq) from [<80100ed0>] (__irq_usr+0x50/0x80)
[    8.765294] Exception stack(0x83033fb0 to 0x83033ff8)
[    8.770351] 3fa0:                                     fffffffe 000024e5 00000000 00000000
[    8.778536] 3fc0: 00000000 7ec43a70 02549178 7ec43a70 024ed850 02548f68 7ec43ab8 00000000
[    8.786718] 3fe0: 76f98b8c 7ec43a00 76e8da6b 76eeedf4 60030030 ffffffff
[    8.793348] ---[ end Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: _CMAFSLAlloc+0x134/0x134 [galcore] ]---
```

otavio, 2022-03-30:
@dv1 is this reproducible in other boards?

@gibsson @MaxKrummenacher @zandrey ideas?

dv1, 2022-03-30:
I did not see this on an imx8mm revB EVK board, nor on a verdin imx8mp. I currently do not have any other imx6 boards here, unfortunately.

gibsson, 2022-03-31:
Never saw that issue, but we haven't done a lot of testing with i.MX6 + 5.10 kernel, mostly i.MX 8M.

MaxKrummenacher, 2022-04-01:
We did not port to the imx 5.10 kernel and haven't seen this kernel panic with the imx 5.4 kernel.

I did a quick port to the fslc 5.10 kernel and build an image for apalis-imx6. The image booted to the console login prompt and weston on the HDMI output, no panic or kernel back trace.

That is with a snapshot of today plus the apalis-imx6 patch to meta-freescale-3rdparty.
[0001-apalis-imx6-build-with-fslc-imx-5.10-kernel.patch.txt](https://github.com/Freescale/meta-freescale/files/8400503/0001-apalis-imx6-build-with-fslc-imx-5.10-kernel.patch.txt)
[bootlog.txt](https://github.com/Freescale/meta-freescale/files/8400517/bootlog.txt)

otavio, 2022-04-06:
@MaxKrummenacher, @gibsson thanks. @dv1 we upgraded the Linux kernel and U-Boot please check if it changes something.

dv1, 2022-04-14:
@otavio Yes, thanks, works now! Using meta-freescale git master 8aa0cac47b390f127e46d1eac541e5ef1ecf711f and poky 3b2edd473ebad28383be5ecd0329ec13bbfef336 here, and it no longer crashes.

closed 2022-04-14 by dv1