Freescale/meta-freescale #1027: imx8dx-mek: Error compiling imx-boot

Ossanes, 2022-03-31:
Building a core-image-base, when compiling imx-boot this error occurs:

```
Log data follows:                                                                                                                                                                                                                          
| DEBUG: Executing shell function do_compile                                                                                        
| NOTE: 8QX boot binary build                                                                                                                                                              
| cp: failed to access '/home/aquino/src/ossystems/oel-platform/build/tmp/work/imx8dx_mek-oel-linux/imx-boot/1.0-r0/git/iMX8DX/scfw_tcm.bin': Not a directory
| WARNING: /home/aquino/src/ossystems/oel-platform/build/tmp/work/imx8dx_mek-oel-linux/imx-boot/1.0-r0/temp/run.do_compile.1809636:179 exit 1 from 'cp /home/aquino/src/ossystems/oel-platform/build/tmp/deploy/images/imx8dx-mek/imx-boot$
tools/scfw_tcm.bin /home/aquino/src/ossystems/oel-platform/build/tmp/work/imx8dx_mek-oel-linux/imx-boot/1.0-r0/git/iMX8DX/scfw_tcm.bin'
| WARNING: Backtrace (BB generated script):                                                                                                                                                                                                
|       #1: compile_mx8x, /home/aquino/src/ossystems/oel-platform/build/tmp/work/imx8dx_mek-oel-linux/imx-boot/1.0-r0/temp/run.do_compile.1809636, line 179
|       #2: do_compile, /home/aquino/src/ossystems/oel-platform/build/tmp/work/imx8dx_mek-oel-linux/imx-boot/1.0-r0/temp/run.do_compile.1809636, line 151                                                                                  
|       #3: main, /home/aquino/src/ossystems/oel-platform/build/tmp/work/imx8dx_mek-oel-linux/imx-boot/1.0-r0/temp/run.do_compile.1809636, line 189                                                                                        
ERROR: Task (/home/aquino/src/ossystems/oel-platform/sources/meta-freescale/recipes-bsp/imx-mkimage/imx-boot_1.0.bb:do_compile) failed with exit code '1'
```

It happens because imx-boot repository doesn't contain the iMX8DX directory (https://source.codeaurora.org/external/imx/imx-mkimage.git).

As workaround I tried to to use iMX8DXL directory instead by adding `BOOT_STAGING:mx8dx  = "${S}/iMX8DXL"`, but this results in another error:

```
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: 8QX boot binary build
| NOTE: building iMX8DXL - REV=C0 flash_spl
| include misc.mak
| include m4.mak
| include android.mak
| include test.mak
| include autobuild.mak
| V2X=YES ./../scripts/split_spl.sh u-boot-spl.bin 0x110000
| 65536+0 records in
| 65536+0 records out
| 65536 bytes (66 kB, 64 KiB) copied, 0.22363 s, 293 kB/s
| 67315+0 records in
| 67315+0 records out
| 67315 bytes (67 kB, 66 KiB) copied, 0.23606 s, 285 kB/s
| make[1]: *** No rule to make target 'mx8dxla1-ahab-container.img', needed by 'flash_spl'.  Stop.
| make: *** [Makefile:23: flash_spl] Error 2
| WARNING: /home/aquino/src/ossystems/oel-platform/build/tmp/work/imx8dx_mek-oel-linux/imx-boot/1.0-r0/temp/run.do_compile.1866066:158 exit 2 from 'make SOC=iMX8DXL REV=C0 dtbs=${UBOOT_DTB_NAME} ${target}'                             
| WARNING: Backtrace (BB generated script):
|       #1: do_compile, /home/aquino/src/ossystems/oel-platform/build/tmp/work/imx8dx_mek-oel-linux/imx-boot/1.0-r0/temp/run.do_compile.1866066, line 158                                                                                 
|       #2: main, /home/aquino/src/ossystems/oel-platform/build/tmp/work/imx8dx_mek-oel-linux/imx-boot/1.0-r0/temp/run.do_compile.1866066, line 189                                                                                       
ERROR: Task (/home/aquino/src/ossystems/oel-platform/sources/meta-freescale/recipes-bsp/imx-mkimage/imx-boot_1.0.bb:do_compile) failed with exit code '1' 
```

otavio, 2022-03-31:
@thochstein @zandrey

thochstein, 2022-03-31:
Looking at it.

PR #1028 (2022-04-01): Fix mx8dx machine override hierarchy
Fixes #1027.
closed 2022-04-01 by otavio