Freescale/meta-freescale #1042: All of Layerscape builds has the wrong "KERNEL_IMAGETYPE" and "SERIAL_CONSOLES" default setting

junzhuimx, 2022-04-11:
All of layerscape image meet wrong kernel image type and no serial console, after build with latest master branch.
It found that "KERNEL_IMAGETYPE" is set to "zImage" and " SERIAL_CONSOLES="" ", which comes from the default setting in poky.

The issue is caused by 2 patches to "use weak defaults" 
- https://github.com/Freescale/meta-freescale/commit/9d5c49f8c604f117da0995f8946ff04c17caada9 
- https://github.com/Freescale/meta-freescale/commit/4d3ab7284a98fc0b3e3915e5ac21fc56828e95fa

These 2 patches are not reasonable to set weak default as "??=", as original weak setting can be easier to overwrite by "=" setting in the machine configure files in the maintenance Yocto layers.

otavio, 2022-04-11:
@niebelm, could you check this? It is a regression from the PRs you did.

niebelm, 2022-04-11:
Sorry for the inconvinience:

- `SERIAL_CONSOLE` is depracated since 2013 but is used to set an empty weak default for `SERIAL_CONSOLES` (poky 
c09809f0609beb27e5670d69cbdcf713fb884f7b - https://git.yoctoproject.org/poky/commit/meta/conf/bitbake.conf?id=c09809f0609beb27e5670d69cbdcf713fb884f7b)

- kernel.bbclass decides to for whatever reason, that zImage is the best global default for `KERNEL_IMAGETYPE`.

Will provide reverts tomorrow.  I'm a bit confused.

niebelm, 2022-04-12:
See #1045

otavio, 2022-04-12:
@niebelm merged. @junzhuimx, could you give it a try? Closing this.

closed 2022-04-12 by otavio

junzhuimx, 2022-04-12:
@niebelm 
The original weak default setting KERNEL_IMAGETYPE ?= "uImage" is default setting for NXP BSP release.
current ??= default setting is meaningless as poky conf will overwrite to "zImage". 
For the customization, it is easy to overwrite it by "="