Freescale/meta-freescale #1068: Yocto  : Image-buildinfo: Exceptionerror

AlexanderVatti, 2022-05-06:
Hi Team,

We are using the following poky , with Ubuntu 18.04
DISTRO = poky
DISTRO_VERSION = 3.0.2

We have added "Image-buildinfo" , It is working fine.

But It is not working fine.
-----------------------
Layer Revisions:      |
-----------------------
meta              = <unknown>:<unknown> 
meta-poky         = <unknown>:<unknown> 
meta-oe           = <unknown>:<unknown> 
meta-multimedia   = <unknown>:<unknown> 
meta-python       = <unknown>:<unknown> 
meta-freescale    = <unknown>:<unknown> 
meta-freescale-3rdparty = <unknown>:<unknown> 
meta-freescale-distro = <unknown>:<unknown> 

 ----> did not see any changes to the following files
    1. bitbake/lib/bb/process.py
    2. meta/classes/image-buildinfo.bbclass
    3. meta/classes/metadata_scm.bbclass

**Root Cause:**
def base_get_metadata_git_branch(path, d):
    import bb.process

    try:
        rev, _ = bb.process.run('git rev-parse --abbrev-ref HEAD', cwd=path)
    except bb.process.ExecutionError:
        rev = '<unknown>'
    return rev.strip()

def base_get_metadata_git_revision(path, d):
    import bb.process

    try:
        rev, _ = bb.process.run('git rev-parse HEAD', cwd=path)
    except bb.process.ExecutionError:
        rev = '<unknown>'
    return rev.strip()

Can you please help us to resolve the issue.

otavio, 2022-05-06:
What `git rev-parse HEAD` returns? this seems like something is not right in your host.

pberginkonsult, 2022-05-06:
What you most likely see is an effect of an update of git on your host machine. https://github.blog/2022-04-12-git-security-vulnerability-announced/ This is something that has affected Yocto builds for a while. Some fixes has been pushed to oe-core but it seems to not solve issue completely.

Here is one of many threads around this in oe-core https://lists.openembedded.org/g/openembedded-core/topic/90680045#164810

And the latest update from today: https://lists.openembedded.org/g/openembedded-core/topic/90927042

So I guess your solution to this is to update poky or backport relevant fixes.

closed 2022-05-06 by otavio

AlexanderVatti, 2022-05-06:
Thanks for inputs and pointing the solutions