Freescale/meta-freescale #1081: OP-TEE tests fail on i.MX 8M Mini LPDDR4 EVKB

tprrt, 2022-05-12:
The regression test 1004 is stalled on i.MX 8M Mini LPDDR4 EVKB:

```
root@imx8mm-lpddr4-evk:~# xtest [J
Run test suite with level=0

TEE test application started overD/TC:? 0 tee_ta_init_pseudo_ta_session:299 Lookup pseudo TA d96a5b40-c3e5-21e3-8794-1002a5d5c61b
D/TC:? 0 tee_ta_init_pseudo_ta_session:312 Open invoke_tests.pta
D/TC:? 0 tee_ta_init_pseudo_ta_session:329 invoke_tests.pta : d96a5b40-c3e5-21e3-8794-1002a5d5c61b
D/TC:? 0 create_ta:382 create entry point for pseudo TA "invoke_tests.pta"
D/TC:? 0 open_session:395 open entry point for pseudo ta "invoke_tests.pta"
 default TEE instance
##########F/TC:? 0 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
D/TC:? 0 tee_ta_close_session:512 csess 0xbe0aab80 id 1
D/TC:? 0 tee_ta_close_session:531 Destroy session
D/TC:? 0 close_session:401 close entry point for pseudo ta "invoke_tests.pta"
D/TC:? 0 destroy_ta:388 destroy entry point for pseudo ta "invoke_tests.pta"
D/TC:? 0 tee_ta_init_session_with_context:607 Re-open TA d96a5b40-c3e5-21e3-8794-1002a5d5c61b
D/TC:? 0 create_ta:382 create entry point for pseudo TA "invoke_tests.pta"
D/TC:? 0 open_session:395 open entry point for pseudo ta "invoke_tests.pta"
F/TC:? 0 plat_prng_add_jitter_entropy:72 0x3B
F/TC:? 0 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
D/TC:? 0 test_entry_params:179 expect memref params: 0xbbc01600/16384 - 0x0/0zu - 0x0/0zu - 0x0/0zu
D/TC:? 0 tee_ta_close_session:512 csess 0xbe0aa9d0 id 1
D/TC:? 0 tee_ta_close_session:531 Destroy session
###############################D/TC:? 0 close_session:401 close entry point for pseudo ta "invoke_tests.pta"
D/TC:? 0 destroy_ta:388 destroy entry point for pseudo ta "invoke_tests.pta"
D/TC:? 0 tee_ta_init_session_with_context:607 Re-open TA d96a5b40-c3e5-21e3-8794-1002a5d5c61b
D/TC:? 0 create_ta:382 create entry point for pseudo TA "invoke_tests.pta"
D/TC:? 0 open_session:395 open entry point for pseudo ta "invoke_tests.pta"
D/TC:? 0 tee_ta_close_session:512 csess 0xbe0aa9d0 id 1
D/TC:? 0 tee_ta_close_session:531 Destroy session
D/TC:? 0 close_session:401 close entry point for pseudo ta "invoke_tests.pta"
D/TC:? 0 destroy_ta:388 destroy entry point for pseudo ta "invoke_tests.pta"
#############
#
# regression+pkD/TC:? 0 tee_ta_init_session_with_context:607 Re-open TA d96a5b40-c3e5-21e3-8794-1002a5d5c61b
*D/TC:? 1 __wq_rpc:34 sleep thread 1 0xbe0806d8
*D/TC:? 0 __wq_rpc:34 wake  thread 1 0xbe0806d8
*F/TC:? 1 plat_prng_add_jitter_entropy:72 0x96
*D/TC:? 0 create_ta:382 create entry point for pseudo TA "invoke_tests.pta"
*D/TC:? 1 tee_ta_init_session_with_context:607 Re-open TA d96a5b40-c3e5-21e3-8794-1002a5d5c61b
*D/TC:? 0 open_session:395 open entry point for pseudo ta "invoke_tests.pta"
*D/TC:? 1 open_session:395 open entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 0 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 1 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 0 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*D/TC:? 1 tee_ta_init_session_with_context:607 Re-open TA d96a5b40-c3e5-21e3-8794-1002a5d5c61b
*D/TC:? 0 __wq_rpc:34 sleep thread 0 0xbe0806d8
*D/TC:? 1 __wq_rpc:34 wake  thread 0 0xbe0806d8
*F/TC:? 0 plat_prng_add_jitter_entropy:72 0xBC
*D/TC:? 1 open_session:395 open entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 0 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 1 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 0 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 1 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 0 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 1 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 0 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 1 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 0 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 1 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 0 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 1 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 0 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 1 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 0 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 1 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 0 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
*F/TC:? 1 invoke_command:408 command entry point for pseudo ta "invoke_tests.pta"
[... 164 lines cut ...]
```

In addition, PKCS#11 tests fail:
```
root@imx8mm-lpddr4-evk:~# xtest -l[J[Jt pkcs11
D/TC:? 0 tee_ta_init_pseudo_ta_session:299 Lookup pseudo TA fd02c9da-306c-48c7-a49c-bbd827ae86ee
Run test suite with level=0

TED/TC:? 0 ldelf_load_ldelf:91 ldelf load address 0x40006000
D/LD:  ldelf:134 Loading TS fd02c9da-306c-48c7-a49c-bbd827ae86ee
E test application started over dF/TC:? 0 trace_syscall:151 syscall #3 (syscall_get_property)
F/TC:? 0 trace_syscall:151 syscall #5 (syscall_open_ta_session)
D/TC:? 0 ldelf_syscall_open_bin:142 Lookup user TA ELF fd02c9da-306c-48c7-a49c-bbd827ae86ee (Secure Storage TA)
efault TEE instance
######################################################
#
# pkcs11
#
######################################################

* pkcs11_1000 Initialize and close Cryptoki library

```

This issue can be reproduced with the master branch (64121b5aafd5eb25a3b2be76cf03cba4d7125e10) of meta-freescale, that contains  OP-TEE version 3.15.0, branch lf-5.10.72_2.2.0, srcrev 4d81b964a72e89a62d04187b3f055d8346b383c9.

otavio, 2022-05-12:
@thochstein have this been released in upcoming update?

thochstein, 2022-05-12:
Hi @tprrt, please see if #1085 helps.

thochstein, 2022-05-16:
Hi @tprrt, yes, I noticed that as well #1084. Hope to get this fixed today.

tprrt, 2022-05-17:
Hi @thochstein,

The same issue occurs with the revision NXP 5.15.5-1.0.0.

thochstein, 2022-05-17:
@clementfaure, can you advise on this issue?

jleni, 2022-05-23:
Hi @thochstein! Thanks for your help!
Do you have any updates on this issue? 😟

tprrt, 2022-05-24:
Hello @thochstein, @clementfaure,

Do you have any news regarding this item?

Kind regards,
Thomas Perrot

thochstein, 2023-04-13:
Okay, this is kind of old.

@tprrt, have you tried latest images to see if the issue is still valid?

If so, I hope that @clementfaure can provide his expertise.

tprrt, 2023-04-14:
Hello @thochstein,
I didn't try the latest images, so I don't know if this issue still occurs.
Kind regards,
Thomas

otavio, 2023-04-14:
@tprrt is it possible for you to test?

otavio, 2023-10-12:
We didn't hear back from @tprrt so we assume it now works. Closing.

closed 2023-10-12 by otavio