Freescale/meta-freescale #1084: Build failure for `imx-atf` on master and kirkstone

otavio, 2022-05-12:
Unfortunately when I went to build `core-image-weston` I ran into a build failure for `imx-atf`:

```
| plat/imx/imx8m/hab.c: In function 'imx_hab_handler':
| plat/imx/imx8m/hab.c:64:57: error: array subscript 0 is outside array bounds of 'uint32_t[0]' {aka 'unsigned int[]'} [-Werror=array-bounds]
|    64 | #define HAB_RVT_CHECK_TARGET_ARM64      ((unsigned long)*(uint32_t *)(HAB_RVT_BASE + 0x18))
|       |                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

_Originally posted by @thochstein in https://github.com/Freescale/meta-freescale/issues/1080#issuecomment-1124471682_

otavio, 2022-05-12:
@thochstein, I opened this issue so that it isn't forgotten.  It might work arounded using `-Warray-bounds`.

MaxKrummenacher, 2022-05-16:
I assume this discussion (and eventually the patch) will fix it from within GCC:

https://gcc.gnu.org/bugzilla/show_bug.cgi?id=99578#c39

thochstein, 2022-05-16:
Thanks, @MaxKrummenacher !

MaxKrummenacher, 2022-05-16:
@thochstein 
I may have been to optimistic. The workaround described in the gcc bugzilla entry makes the error go away if the constant value casted to pointer and then dereferenced is > 4kB, i.e. > 0x1000.
Looks like the HAB_RVT_BASE is at addresses 0x0880 or 0x0900 so those false positves may still be reported.

PR #1089 (2022-05-16): imx-atf: Suppress array-bounds error
The array-bounds error is triggered now in cases where it was silent before,
causing errors like:

```
plat/imx/imx8m/hab.c: In function 'imx_hab_handler':
plat/imx/imx8m/hab.c:64:57: error: array subscript 0 is outside array bounds of 'uint32_t[0]' {aka 'unsigned int[]'} [-Werror=array-bounds]
   64 | #define HAB_RVT_CHECK_TARGET_ARM64      ((unsigned long)*(uint32_t *)(HAB_RVT_BASE + 0x18))
      |                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

It appears that these new cases will be silenced soon in gcc [1 [... cut]
commit dc3cb910 (2022-05-16): imx-atf: Suppress array-bounds error

The array-bounds error is triggered now in cases where it was silent before,
causing errors like:

```
plat/imx/imx8m/hab.c: In function 'imx_hab_handler':
plat/imx/imx8m/hab.c:64:57: error: array subscript 0 is outside array bounds of 'uint32_t[0]' {aka 'unsigned int[]'} [-Werror=array-bounds]
   64 | #define HAB_RVT_CHECK_TARGET_ARM64      ((unsigned long)*(uint32_t *)(HAB_RVT_BASE + 0x18))
      |                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

It appears that these new cases will be silenced soon in gcc [ [... cut]
commit 5fe30678 (2022-06-16): imx-atf: Suppress array-bounds error

The array-bounds error is triggered now in cases where it was silent before,
causing errors like:

```
plat/imx/imx8m/hab.c: In function 'imx_hab_handler':
plat/imx/imx8m/hab.c:64:57: error: array subscript 0 is outside array bounds of 'uint32_t[0]' {aka 'unsigned int[]'} [-Werror=array-bounds]
   64 | #define HAB_RVT_CHECK_TARGET_ARM64      ((unsigned long)*(uint32_t *)(HAB_RVT_BASE + 0x18))
      |                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

It appears that these new cases will be silenced soon in gcc [ [... cut]
closed 2022-05-16 by otavio