Freescale/meta-freescale #1087: Support linux-yocto kmeta configuration

m5p3nc3r, 2022-05-13:
Hi community

Are there any plans for meta-freescale to support the standard linux-yocto kmeta config fragments system?  The reason I ask is that I am building a k3s based system for multiple targets and meta-virtualization contains the correct config fragment to make sure that the built kernel has the correct features to support k3s.  On platforms that support kmeta configuration, this 'just works'.

If there is no plan to do this, what is the preferred way to manage the kernel configuration based on the required workloads?

Cheers
Matt

angolini, 2022-05-16:
One of the reasons is that some BSP vendor kernels does not use that so far, but I don't see why not to include this support for kernels using mainline BSP, for example.

m5p3nc3r, 2022-05-17:
Hey @angolini, thanks for the feedback.

I have create a PoC that enables me to build the downstream meta-freescale kernel configuration for imx8mp.  It makes use of the tooling that Bruce Ashfield create (kgit-config) to try and automate the process of kmeta creation.  This mostly works, but I had to manually remove CONFIG_DRM due to a Cadence driver issue (I don't need HDMI output working, so this is fine for my use-case).  There are a lot of warnings, but these could be resolved with a little massaging, but I would rather that was adopted here rather than being done out of tree config.

The benefit of this approach now is that when I want to build with a userspace like k3s, the correct kernel configuration is automatically applied.  

You can see what I created here: https://github.com/m5p3nc3r/meta-ewaol-machine/tree/main/meta-ewaol-ext/recipes-imx/recipes-kernel/linux-yocto

angolini, 2022-05-18:
@m5p3nc3r , feel free to create a PR if you want to upstream your work on this topic.
Other than that, I think we can close this issue, what do you think?

closed 2022-06-03 by angolini