Freescale/meta-freescale #1107: imx53 + mesa + freedreno + wayland + video => color issue

woutervanh, 2022-06-13:
As I don't have really another place to keep track of this.
Is somebody else using freedreno + mesa + wayland + gstreamer video on imx53?
It it's using the gpu to composite, colors seems to be messed up. If running on kmssink or not-scaled (thus using hw plane, using following patch [](https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/966/diffs)), colors are correct
[image]

Also, I have the impression it's not that fluent. Although it's using hardware decoding and direct memory paths in between the elements. and only 360p

otavio, 2022-10-12:
Did you try on master?

otavio, 2022-11-29:
Closing due inactivity.

closed 2022-11-29 by otavio