Freescale/meta-freescale #1126: imx-boot fails to build on kirkstone when using imx-atf from meta-imx

Ondrej1, 2022-06-29:
Hi. I ran into an issue after upgrading to kirkstone. In this [commit](https://github.com/Freescale/meta-freescale/commit/546a884afd29b5e058ddf0f1a6964246db8ffb70) , the deploy location of bl31.bin is changed. This change is done in both imx-atf and imx-boot, so when imx-atf from meta-freescale is used, everything works fine. However, meta-imx contains imx-atf version 2.6 which is newer than in meta-freescale (2.4) and yocto chooses the newer version. ATF recipe in meta-imx deploys to the old location and imx-boot cannot find bl31.bin.

I worked around this by setting `PREFERRED_VERSION_imx-atf = "2.4%"`, but I assume this isn't the best way to fix this. Is there a better way to fix this on my end, or should I wait for atf recipe in meta-imx to change / use a bbappend in my layer?

zandrey, 2022-06-29:
Hello @Ondrej1 

Thanks for reporting this!

I was wondering how your build env setup looks like that you've combined the latest `kirkstone` branch from this layer together with _meta-imx_ one?

Usually, you do need _meta-imx_ layer to build NXP vendor BSP, and according to [instructions in _meta-imx_ layer](https://source.codeaurora.org/external/imx/meta-imx/tree/README?h=kirkstone-5.15.32-2.0.0) you shall use the manifest provided in [imx-manifest repository on a separate `imx-linux-kirkstone` branch](https://source.codeaurora.org/external/imx/imx-manifest/tree/imx-5.15.32-2.0.0.xml?h=imx-linux-kirkstone).

When I look into the [imx-5.15.32-2.0.0.xml](https://source.codeaurora.org/external/imx/imx-manifest/tree/imx-5.15.32-2.0.0.xml?h=imx-linux-kirkstone#n22) file though, I see that the _meta-freescale_ 2fb1ce365338126aad365012ebb913b3e4a9f1be commit listed there points to the version before _TF-A_ changes were introduced, which makes me believe there you have some kind of a "special" setup that you use to mix-and-match both NXP vendor BSP and upstream Community one.

The solution to this problem in this layer might be that someone submits the upgrade to NXP `rel_imx_5.15.32_2.0.0` tag (perhaps you can contribute here?) to align all NXP components to what _meta-freescale_ provides; or you start building either one of BSP flavors: NXP vendor one, or Community one.

-- andrey

Ondrej1, 2022-06-29:
Thanks for the quick response. Yes, I am using a custom manifest and I just used the newest commit from the kirkstone branch in each layer. I will update it to use the commits in imx-manifest.

closed 2022-06-29 by Ondrej1