Freescale/meta-freescale #1146: mkimage_uboot libssl failure

thepperson, 2022-07-26:
I ran into a failure building u-boot on commit 3cb29cff92568ea835ef070490f185349d712837 of this repository.
I found a mailing list message that details a solution that worked for me. https://www.mail-archive.com/meta-freescale@lists.yoctoproject.org/msg00297.html

otavio, 2022-07-26:
Which branch are you using?

thepperson, 2022-07-26:
It is the dunfell branch

otavio, 2022-07-26:
Could you check if master fails? So we'd first fix it in master and then do the backports.

thepperson, 2022-07-26:
I don't have a setup that works on the kirkstone release at the moment to do that test with.

otavio, 2022-07-28:
@thochstein, is this a known issue?

thochstein, 2022-07-28:
It's a mistake in the cherry-pick. In order to drop these lines from `do_compile`:

```
    # mkimage_uboot requires libssl.so.1.1 from ${STAGING_LIBDIR_NATIVE}
    export LD_LIBRARY_PATH=${STAGING_LIBDIR_NATIVE}:$LD_LIBRARY_PATH
```

you also need to cherry-pick this so the sysroot `mkimage` is used: https://github.com/Freescale/meta-freescale/commit/c59f817b0d1ee70ef224405ff8acd31e0743ed91.

Right, @zandrey?

zandrey, 2022-11-24:
Sorry, just saw the mention.

Yes, correct - c59f817b0d1ee70ef224405ff8acd31e0743ed91 is needed so that sysroot `mkimage` is used.

otavio, 2022-11-28:
I just checked this and it is already backported.

closed 2022-11-29 by otavio