Freescale/meta-freescale #1172: linux-imx fails to build when using ipk packages

johannesschrimpf, 2022-08-18:
In `linux-imx_5.15.bb`, `LOCALVERSION` is set to `"-5.15.32_2.0.0"`, including an underscore. This seems to end up in the package name, however, the underscore seems to be an illegal character for the package management system (I'm using `PACKAGE_CLASSES = "package_ipk"`).

https://github.com/Freescale/meta-freescale/blob/cdc56a683fe009ea2889ea6354d3a44b562454d7/recipes-kernel/linux/linux-imx_5.15.bb#L27

My build fails as following:
```
ERROR: linux-imx-5.15.32+gitAUTOINC+fa6c316859-r0 do_package_write_ipk: Fatal errors occurred in subprocesses:
Command 'PATH="/data/yocto-kirkstone/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/data/yocto-kirkstone/layers/poky/scripts:/data/yocto-kirkstone/build/tmp/work/imx6qdl_variscite_som-poky-linux-gnueabi/linux-imx/5.15.32+gitAUTOINC+fa6c316859-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi:/data/yocto-kirkstone/build/tmp/work/imx6qdl_variscite_som-poky-linux-gnueabi/linux-imx/5.15.32+gitAUTOINC+fa6c316859-r0/recipe-sysroot/usr/bin/crossscripts:/data/yocto-kirkstone/build/tmp/work/imx6qdl_variscite_som-poky-linux-gnueabi/linux-imx/5.15.32+gitAUTOINC+fa6c316859-r0/recipe-sysroot-native/usr/sbin:/data/yocto-kirkstone/build/tmp/work/imx6qdl_variscite_som-poky-linux-gnueabi/linux-imx/5.15.32+gitAUTOINC+fa6c316859-r0/recipe-sysroot-native/usr/bin:/data/yocto-kirkstone/build/tmp/work/imx6qdl_variscite_som-poky-linux-gnueabi/linux-imx/5.15.32+gitAUTOINC+fa6c316859-r0/recipe-sysroot-native/sbin:/data/yocto-kirkstone/build/tmp/work/imx6qdl_variscite_som-poky-linux-gnueabi/linux-imx/5.15.32+gitAUTOINC+fa6c316859-r0/recipe-sysroot-native/bin:/data/yocto-kirkstone/layers/poky/bitbake/bin:/data/yocto-kirkstone/build/tmp/hosttools" opkg-build -Z xz -a "--memlimit=50% --threads=28" kernel-module-snd-usb-audio-5.15.32-5.15.32_2.0.0+gfa6c3168595c /data/yocto-kirkstone/build/tmp/work/imx6qdl_variscite_som-poky-linux-gnueabi/linux-imx/5.15.32+gitAUTOINC+fa6c316859-r0/deploy-ipks/imx6qdl_variscite_som' returned non-zero exit status 1.
Subprocess output:kernel-module-snd-usb-audio-5.15.32-5.15.32_2.0.0+gfa6c3168595c
*** Error: Package name kernel-module-snd-usb-audio-5.15.32-5.15.32_2.0.0+gfa6c3168595c contains illegal characters, (other than [a-z0-9.+-])
```

pberginkonsult, 2022-08-18:
About an hour before I read this I also went in to this but with deb packaging. I put a PR with a proposed solution here: https://github.com/Freescale/meta-freescale/pull/1173

Some old reference that hepled me: https://www.yoctoproject.org/pipermail/meta-freescale/2019-January/023802.html

otavio, 2022-08-18:
Fixed in #1173.

closed 2022-08-18 by otavio