Freescale/meta-freescale #1178: imx-atf fails to build with binutils 2.39

pberginkonsult, 2022-08-23:
In current master of poky (081a3655fbb10b83793f102b975e153e75240682) where binutils is updated to 2.39 the package imx-atf fails to build for the imx8mp-lpddr4-evk machine (and most probably others to).

The failure is:
```
|   LD      <...>/imx-atf/2.6+gitAUTOINC+c6a19b1a35-r0/git/build/imx8mp/release/bl31/bl31.elf
| aarch64-poky-linux-ld: warning: <...>/imx-atf/2.6+gitAUTOINC+c6a19b1a35-r0/git/build/imx8mp/release/bl31/bl31.elf has a LOAD segment with RWX permissions
| make: *** [Makefile:1254: <...>/imx-atf/2.6+gitAUTOINC+c6a19b1a35-r0/git/build/imx8mp/release/bl31/bl31.elf] Error 1
```
A similar case in systemd with solution: https://github.com/systemd/systemd-stable/commit/8a6f966be404897b5333c218701965ac3b5a0806
Bug report for atf: https://developer.trustedfirmware.org/T996

Looking at the references above my guess is that a solution would be to add '-Wl,--no-warn-rwx-segments' but haven't been able to find where in the makefile structure to do it.

otavio, 2022-08-23:
Could you prepare a PR for it?

MaxKrummenacher, 2022-08-24:
I noticed the same error over the weekend. I have a fix nearly ready.

PR #1179 (2022-08-24): imx-atf: fix build error with binutils 2.39

Fixes #1178

I tried to build imx-atf with this patch in kirkstone and it builds fine. However, I assume that binutils will not be backported to kirkstone in openbedded-core and this patch will not be needed in kirkstone.
closed 2022-08-24 by otavio