Freescale/meta-freescale #1203: [kirkstone] chromium GPU acceleration doesn't work on imx6

bmx666, 2022-09-10:
On the latest kirkstone branch GPU acceleration doesn't work, if rollback to commit dd98526300937450d741c9b3403799632f417226 everything is fine.

otavio, 2022-09-10:
Please try to identify which commit breaks it?

bmx666, 2022-09-10:
Hi @otavio , sorry, my fault, looks like [meta-browser](https://github.com/OSSystems/meta-browser/) pushed [new version of Chromium](https://github.com/OSSystems/meta-browser/commit/b8e658b5a4da60a0d297fb7c2cdb7c7c251ec8bf) also. Our team will test various build combinations on Monday and I'll update the status of this issue:
- [old meta-browser](https://github.com/OSSystems/meta-browser/commit/6e7429cb084782954afe2bd8f8680370908c94ed) + [new meta-freescale](https://github.com/Freescale/meta-freescale/commit/f9efb4d22669cdade59fe78db524c37fbc67becf)
- [new meta-browser](https://github.com/OSSystems/meta-browser/commit/4bcc607678057d0600fa9290f1a7743d233646df) + [old meta-freescale](https://github.com/Freescale/meta-freescale/commit/dd98526300937450d741c9b3403799632f417226)

Current status of our builds:
- [new meta-browser](https://github.com/OSSystems/meta-browser/commit/4bcc607678057d0600fa9290f1a7743d233646df) + [new meta-freescale](https://github.com/Freescale/meta-freescale/commit/f9efb4d22669cdade59fe78db524c37fbc67becf) - GPU acceleration doesn't work
- [old meta-browser](https://github.com/OSSystems/meta-browser/commit/6e7429cb084782954afe2bd8f8680370908c94ed) + [old meta-freescale](https://github.com/Freescale/meta-freescale/commit/dd98526300937450d741c9b3403799632f417226) - GPU acceleration works fine

bmx666, 2022-09-12:
Hi @otavio , issue was in [new version of Chromium](https://github.com/OSSystems/meta-browser/commit/b8e658b5a4da60a0d297fb7c2cdb7c7c251ec8bf), sorry for report false issue, I will create issue on [meta-browser](https://github.com/OSSystems/meta-browser/)
 
Results of our builds:
[old meta-browser](https://github.com/OSSystems/meta-browser/commit/6e7429cb084782954afe2bd8f8680370908c94ed) + [new meta-freescale](https://github.com/Freescale/meta-freescale/commit/f9efb4d22669cdade59fe78db524c37fbc67becf)  - GPU acceleration works fine
[new meta-browser](https://github.com/OSSystems/meta-browser/commit/4bcc607678057d0600fa9290f1a7743d233646df) + [old meta-freescale](https://github.com/Freescale/meta-freescale/commit/dd98526300937450d741c9b3403799632f417226) - GPU acceleration doesn't work

closed 2022-09-12 by bmx666