Freescale/meta-freescale #1204: ls1012afrwy: Failure to fetch an existing patch file

angolini, 2022-09-12:
```
Loading cache: 100% |#######################################################################################################################################################################| Time: 0:00:00
Loaded 3078 entries from dependency cache.
ERROR: /media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq_5.4.bb: Unable to get checksum for linux-qoriq SRC_URI entry 0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch: file could not be found
The following paths were searched:
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq-5.4.47+gitAUTOINC+6bff40d413/fslc/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq/fslc/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/files/fslc/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq-5.4.47+gitAUTOINC+6bff40d413/ls1012afrwy/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq/ls1012afrwy/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/files/ls1012afrwy/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq-5.4.47+gitAUTOINC+6bff40d413/ls1012a/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq/ls1012a/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/files/ls1012a/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq-5.4.47+gitAUTOINC+6bff40d413/fsl-lsch2/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq/fsl-lsch2/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/files/fsl-lsch2/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq-5.4.47+gitAUTOINC+6bff40d413/qoriq-arm64/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq/qoriq-arm64/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/files/qoriq-arm64/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq-5.4.47+gitAUTOINC+6bff40d413/qoriq/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq/qoriq/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/files/qoriq/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq-5.4.47+gitAUTOINC+6bff40d413/use-nxp-bsp/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq/use-nxp-bsp/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/files/use-nxp-bsp/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq-5.4.47+gitAUTOINC+6bff40d413/aarch64/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq/aarch64/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/files/aarch64/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq-5.4.47+gitAUTOINC+6bff40d413/aarch64/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq/aarch64/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/files/aarch64/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq-5.4.47+gitAUTOINC+6bff40d413/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/linux-qoriq/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
/media/daiane/TRIFORCE/yocto/master/sources/meta-freescale-3rdparty/recipes-kernel/linux/files/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch
ERROR: Parsing halted due to errors, see error messages above
```

angolini, 2022-09-12:
@otavio can you, please, help assign this build issue?

thochstein, 2022-09-12:
@ting-liu and @junzhuimx, can you take a look? Looks like linux-qoriq 5.4 recipe was added without the patches.

junzhuimx, 2022-09-13:
@angolini 
The patch file exists as the https://github.com/Freescale/meta-freescale/blob/master/recipes-kernel/linux/linux-qoriq/0001-Makfefile-linux-5.4-add-warning-cflags-on-LSDK-20.04.patch 
And the log shows the build only search the paths in the layer meta-freescale-3rdparty only.

otavio, 2022-09-17:
@junzhuimx someone from QorIQ must keep an eye to the layer. Is this @ting-liu? The recipes are mostly abandoned.

otavio, 2022-09-17:
@angolini which branch did you use? I tried this on master and couldn't reproduce it.

otavio, 2022-09-17:
@angolini you can see the generated package below:
<details>
```
~/src/ossystems/oel/master/build/tmp/deploy/ipk/ls1012afrwy/kernel-image-5.10.52+ga11753a89ec6_5.10.52+git0+a11753a89e-r0_ls1012afrwy.ipk
[v] View
[i] Info
[s] Scripts
[x] Extract
Option (ENTER to cancel):
i
Package: kernel-image-5.10.52+ga11753a89ec6
Version: 5.10.52+git0+a11753a89e-r0
Description: Linux Kernel for NXP QorIQ platforms
 Linux Kernel for NXP QorIQ platforms.
Section: kernel
Priority: optional
Maintainer: O.S. Systems Software LTDA. <contato@ossystems.com.br>
License: GPL-2.0-only
Architecture: ls1012afrwy
OE: linux-qoriq
Depends: kernel-image-fitimage-5.10.52+ga11753a89ec6 (= 5.10.52+git0+a11753a89e-r0), kernel-image-image-5.10.52+ga11753a89ec6 (= 5.10.52+git0+a11753a89e-r0)
Provides: kernel-image
Source: linux-qoriq_5.10.bb
/home/otavio/src/ossystems/oel/master/build/tmp/deploy/ipk/ls1012afrwy/kernel-image-5.10.52+ga11753a89ec6_5.10.52+git0+a11753a89e-r0_ls1012afrwy.ipk
```
</details>

angolini, 2022-09-19:
I cannot explain, but cleaning everything and trying again (using exactly what I had before) does not reproduce the error again.

So I'm going to close this issue.

closed 2022-09-19 by angolini

thochstein, 2022-09-19:
Hmm, I'm confused. I thought the failure would be expected since there are no patch files in the recipe folder.