Freescale/meta-freescale #1211: upower package now requires the CONFIG_USER_NS kernel option to be included in config

bmx666, 2022-09-15:
Due to recent updates, upower.service no longer works if the kernel does not support CONFIG_USER_NS.
From a security point of view - this option is dangerous and should be disabled, but the upower maintainers decided to enable it... What do you think about that?

https://linuxfromscratch.org/blfs/view/systemd/general/upower.html

otavio, 2022-09-15:
Why is this a dangerous option to enable?

bmx666, 2022-09-15:
@otavio main reason - [user_namespaces - overview of Linux user namespaces](https://www.man7.org/linux/man-pages/man7/user_namespaces.7.html)

Links about **CONFIG_USER_NS**:
- https://snyk.io/blog/kernel-privilege-escalation/
- https://bugs.archlinux.org/task/36969
- https://github.com/netblue30/firejail/issues/1347
- https://bugzilla.redhat.com/show_bug.cgi?id=1170684
- https://jfrog.com/blog/the-impact-of-cve-2022-0185-linux-kernel-vulnerability-on-popular-kubernetes-engines/

many embedded devices doesn't include security audit or selinux and this kernel option opens a lot of opportunity for exploits year by year.

otavio, 2022-09-15:
I don't see those as a massive issue for the default setting, as I expect that someone developing a product should worry about it.

I believe we ought to enable it, as onboarding is easy when trying it. The security review is something the user should do.

closed 2022-09-18 by bmx666